gphoto/libgphoto2 #562: Movie capture failure with older Canon EOS

michaelsburns, 2020-09-16:
Hello,

I've been using gphoto2 with --capture-movie to capture the liveview from my Rebel T2i (EOS 550D), and recently tried libgphoto2 at HEAD to test new documented performance gains. Unfortunately the latest version does not work:

mbx@raspberrypi:~/software/libgphoto2 $ CAMLIBS="camlibs" LD_PRELOAD="libgphoto2/.libs/libgphoto2.so.6 libgphoto2_port/libgphoto2_port/.libs/libgphoto2_port.so.12" ../gphoto2/gphoto2/gphoto2 --capture-movie=1 --debug --debug-logfile=log.txt

Capturing 1 preview frames as movie to 'movie.mjpg'.
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)

I've narrowed this down to changes made in commit [e2ad1cb](https://github.com/gphoto/libgphoto2/commit/e2ad1cb). In the previous commit, 56fdba4, the capture works. My very naive, shot in the dark, wild guess here is that the repeated EOS_GetViewFinderData requests, which are all returning "Camera Not Ready," are actually preventing the camera from getting ready in a denial of service way. Maybe some sort of backoff for repeated "not readies" would help.

See attached logs:

[libgphoto2_56fdba4_success.log](https://github.com/gphoto/libgphoto2/files/5228881/libgphoto2_56fdba4_success.log)
[libgphoto2_e2ad1cb_fail.log](https://github.com/gphoto/libgphoto2/files/5228882/libgphoto2_e2ad1cb_fail.log)

msmeissn, 2020-09-18:
Hmm. the speed up hack works on newer eos, but the older are overloaded... i remember :/
The commit removed the backoffs to avoid slowing down the framerate ... I need to put it back, perhaps add some different strategy again.

michaelsburns, 2020-09-26:
Thanks, I can confirm that movie capture works @ HEAD on my 550D.

michaelsburns, 2020-09-26:
Also, as a datapoint, I'm getting ~25 FPS on the workhorse that is a Raspberry Pi 3 B+ when writing to /dev/null. Not bad!

AndreSteenbergen, 2020-12-17:
according to the supported list my eos 10d (I know ancient) should be able to liveview over gphoto2. It doesn't. it takes 1 image, not sending it to the pc and in the display of the camera the only thing shown is a blinking "busy" using version 2.5.26

hfiguiere, 2020-12-17:
I don't even see the 10D on the list, but it surely doesn't support liveview. Nor the 20D that came after.

AndreSteenbergen, 2020-12-18:
[image]

According to http://www.gphoto.org/proj/libgphoto2/support.php it is....

msmeissn, 2020-12-18:
The 10D is served by the "very old" canon driver. I took it from the abilities list in there. I might have done however copy and paste addition at this time :/

AndreSteenbergen, 2020-12-18:
no problem, I also own a 50D, which works. I just thought, maybe... just maybe ;)