gphoto/libgphoto2 #563: Ricoh Theta Z1: downloading the very first image is very slow if camera internal storage has many files

alik55, 2020-09-17:
**Describe the bug**

Ricoh Theta Z1 used in PTP mode.
Using the camera to capture/download/delete file. Do not care about other files already present on camera storage.
Downloading very the first image takes a long time if there are many files present on camera internal storage - about 17sec per each 100 files. If a program using libgphoto captures many files, all files after the first one get downloaded fast, as expected. The first download is fast only if the internal camera storage is empty.

It seems libgphoto parses the camera filesystem and gets info on every object found there to cache the whole tree.
This is not needed for my application. Ideally GP_EVENT_FILE_ADDED event should provide a file handle and the file can be downloaded and deleted using that handle only via MTP protocol, so there is no need for knowledge about other files.

Can this libgphoto be configured not to list all existing files and get their info?

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs

**libgphoto2 and gphoto2 version**
latest master

**To Reproduce**
Steps to reproduce the behavior:

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

alik55, 2020-09-18:
After more digging.

The first image slowdown occurs for two reasons (due to architecture of libgphoto):

1. On very first capture, gp_camera_trigger_capture takes 1.5sec longer, this delay being independent of how many files Theta has on internal storage.

2. On very first image download (actually inside gp_camera_wait_for_event when waiting of first GP_EVENT_FILE_ADDED) libgphoto calls ptp_getobjecthandles in camlibs/ptp.c which enables libgphoto cache all info on all images. The slowdown is about 3.2-3.5sec per each 150 images.

I recompiled libgphoto to have ptp_getobjecthandles return OK early and without querying camera to list all objects. That did remove the slowdown for reason #2, but added 0.5sec to every image download time afterwards, which makes this solution unacceptable. I guess libgphoto has a good reason to do the caching. 

As far as I understand, the MTP protocol that Theta uses should return new image handle on image add event and that handle can be used to fetch the image and delete it from camera and will avoid slowdown, unless I am missing something.  If libgphoto could support that workflow, that would be ideal.

msmeissn, 2020-09-18:
I try to handle the filesystem access in an on-demand fashion already, which you probably saw.

This on demand is per directory, so it should fetch only directories needed. 

Currently this "just return added objects" is only available by hacking the driver. Hmm. I could consider some option setting perhaps.