gphoto/libgphoto2 #566: Documentation incorrectly lists DC-GH5 camera support

VisionaryMind, 2020-10-04:
**Describe the bug**

At the support cameras [page](http://gphoto.org/proj/libgphoto2/support.php), you list Panasonic DC-GH5. Unfortunately, running gphoto2 --list-cameras does not show it in the list (nor any of the Panasonic DC Lumix cameras). Libgphoto2, further, does not recognize such cameras, either in PTP or tethered mode.

**Name the camera**
Camera name as shown by gphoto2 --auto-detect:
USB PTP Class Camera           usb:001,011

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2         2.5.15         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.16         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**version as shown via dpkg -l | grep *gphoto**
ii  libgphoto2-6:arm64                  2.5.16-2                                  arm64      gphoto2 digital camera library
ii  libgphoto2-l10n                        2.5.16-2                                  all             gphoto2 digital camera library - localized messages
ii  libgphoto2-port12:arm64         2.5.16-2                                  arm64       gphoto2 digital camera port library
ii  gphoto2                                    2.5.15-2                                  arm64       digital camera command-line client

**To Reproduce**
Steps to reproduce the behavior:
Attach a DC-GH5 to a USB 3.0 port and attempt to capture an image (or run any other command, in general). These cameras are not yet supported by your library.

hfiguiere, 2020-10-04:
libgphoto2 2.5.16 is three years old.

Current version is 2.5.25.

msmeissn, 2020-10-05:
as Hub writes, your version is outdated. 
GH5 support started in 2.5.19, so update to a newer version (2.5.25 is current)

closed 2020-10-05 by msmeissn