gphoto/libgphoto2 #567: --capture-movie Resolution?

Ben-Chapman, 2020-10-12:
Hi there,
Earlier this year I [wrote an article](https://www.crackedthecode.co/how-to-use-your-dslr-as-a-webcam-in-linux/), detailing how to setup a DSLR camera as  webcam in Linux, and specifically using `gphoto2 --capture-movie`. I've had a number of people reach out to me, asking about the output resolution they're getting from gphoto2 - which is far less than the specified capabilities of their camera. By way of example, I'm using a Canon EOS 500D (15 megapixel sensor), and get a 928x616 image from `--capture-preview` and `--capture-movie`.

From what I understand, `--capture-movie` is simply capturing a stream of preview images (as mpjeg) which are not of the full resolution of the camera's image sensor. 

The questions I have are:
- Is it possible to configure the output resolution for `--capture-movie` ?
- Assuming not, is there a reliable way to understand what the `--capture-movie` output resolution will be for a given camera (aside from just testing it)?
- Any other tips you might have for increasing the resolution of the video stream from these cameras?

msmeissn, 2020-10-14:
the new 2.5.26 release introduced liveviewsize option for
- newer Nikon DSLR
- Fuji XT
- Sony QX

still need to find out how / if tzhe Sony Alpha does
and/or if the Canon EOS do it.

I think its not a good reliable way to find out the resolution. hardcoding lists somewhere I would not prefer.

grssnbchr, 2020-10-20:
Hey @msmeissn 
Thanks for the update as well. Is the "liveviewsize" option somehow documented? How should we use it? 
Thanks,
Timo

msmeissn, 2020-10-20:
gphoto2 --get-config liveviewsize   will show the options available

it might e.g. report QVGA, VGA and XGA ... so you can set XGA:

gphoto2 --set-config liveviewsize=XGA --capture-movie

grssnbchr, 2020-10-27:
Thanks! Just for clarity: Is "liveview" generally something that's only available on DSLRs (when talking about Nikon)? The newer mirrorless Z models don't have it, right?

msmeissn, 2020-10-28:
It can be available on small form cameras, or only bigger ones.

e.g. very old canon powersjhot or sierras could do liveview too.

These days most cameras that support remote control also support liveview, but there is no clear rules on where it is available.

For liveviewsize the cameras classes are limited, as this is an option that was not much useful before

rodolpheh, 2020-11-22:
Hello, I'm also interested by this issue. I just bought a Canon EOS M50 and can't find a way to set up the resolution when capturing a movie. I did notice a few details though. First, the output resolution is different if the camera is set up in video capture mode or not. When in video capture mode, we can have an image of 512x288 (which is a 16:9 ratio so that seems to be normal). Otherwise, the image is 480x320 (ratio of 3:2, still coherent). Secondly, I tried the Canon EOS Webcam Utility on Windows and this one can output a 1280x720 image, much bigger than the resolutions I get with gphoto. One peculiar thing is that the screen stays on when using the Canon EOS Webcam Utility whereas it turns off when using gphoto. To me it seems Canon is making a secret communication thingy to allow a much better exploitation of the camera than what can be achieved with `--capture-movie` option. I would gladly try to sniff the USB if it could help investigating this.

sjoelund, 2020-11-30:
> One peculiar thing is that the screen stays on when using the Canon EOS Webcam Utility whereas it turns off when using gphoto.

The screen is 3:2 around 1040k dots, so around 1248x832 pixels if what I find in data sheets is correct. That's close to 1280x720. What if enabling the screen in some way allows them to get a larger preview? And yeah, 480x320 is a bit on the low side...
(Would be even nicer if the viewfinder could be captured since that's twice as big)

sjoelund, 2020-12-01:
So tinkering around a bit on my M50:

```bash
$ gphoto2 --get-config=/main/status/cameramodel | grep Current
Current: Canon EOS M50
$ gphoto2 --shell
gphoto2: {.../tmp} /> set-config /main/actions/viewfinder=1  
gphoto2: {.../tmp} /> capture-preview
Saving file as capture_preview.jpg
gphoto2: {.../tmp} /> 
$ file capture_preview.jpg 
capture_preview.jpg: JPEG image data, baseline, precision 8, 1024x576, components 3
$ rm capture_preview.jpg 
$ gphoto2 --set-config /main/actions/viewfinder=1 --capture-preview 
Saving file as capture_preview.jpg                                             
$ file capture_preview.jpg 
capture_preview.jpg: JPEG image data, baseline, precision 8, 512x288, components 3
```

You need to set /main/actions/viewfinder=1, but if you do the helper appears and you have to press the screen to select auto/manual/etc depending on the mode you are in. After that, the preview has a higher resolution (960x640 in picture mode).

But you cannot --capture-movie from the shell and if you do `gphoto2 --set-config /main/actions/viewfinder=1 --capture-movie`, the first few frames are lower resolution than the rest which messes up ffmpeg (it uses the resolution of the first jpg). Running the command again after `Ctrl+C` works though (since gphoto2 doesn't reset the viewfinder when you abort the command).

g-voelker, 2020-12-06:
Just so this is clear. @sjoelund tested while setting the M50 to video mode.

A couple more observations here:

- When in video mode and setting the viewfinder to 1 the camera screen stays on while it switches off in any other mode I tested with any setting for the viewfinder.
- When using the cam as webcam on Windows through the EOS webcam utility the camera functions stay available. I can shoot and process images as usual. That is the webcam utility most probably utilizes a similar liveview configuration.
- The resolution the EOS utility outputs (1280x720; not confirmed or checked) is exactly the lowest video resolution the camera records.That is, however a lot less pixels that the camera's screen technically has.

It seems to me that the windows utility does use a very similar liveview implementation which seems independent of the screen. However, the behavior of the camera using gphoto depends on how the viewfinder option is set.

g-voelker, 2020-12-06:
Another comment on this:
When using the option viewfinder=1 during image capture with gphoto2 and ffmpeg with the following command 

`gphoto2 --stdout --set-config viewfinder=1 --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video2`

it does generate a 1024x576 stream with rubbish output through ffmpeg. Only when forcing ffmpeg to the lower resolution (exactly quarter) it generates a clean stream:

`gphoto2 --stdout --set-config viewfinder=1 --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -s 512:288 -f v4l2 /dev/video2`

Now I have the viewfinder switched on during the stream, that is the camera screen is open and displays the liveview. I am however bound to the lower resolution.

For completeness: I am on Archlinux using

ffmpeg 2:4.3.1-2
gphoto2 2.5.26-1

msmeissn, 2020-12-06:
I need to dig into this. Currently a bit short on time, but i have a long christmas break upcoming

sjoelund, 2020-12-07:
> it does generate a 1024x576 stream with rubbish output through ffmpeg. Only when forcing ffmpeg to the lower resolution (exactly quarter) it generates a clean stream

I did manage to get a clean 1024x576 or 960x640 feed into v4l2loopback via ffmpeg. But it was a bit bugged with some restarts being necessary.
gphoto2 2.5.23, libgphoto2 2.5.26.1, ffmpeg 4.3 (Ubuntu 20.04 with some nonstandard packages installed)

g-voelker, 2020-12-07:
@sjoelund: that is interesting, I will see if more restarts help here. There should be some difference in the detailed logs maybe? How comes you are using gphoto2 2.5.23 with libgphoto2 2.5.26? Any particular reason?

@msmeissn Thanks for looking into it. If you need more data, maybe with this particular camera setup (EOS M50), just let us know here or in a DM. I will be happy to provide more info if needed.

sjoelund, 2020-12-07:
> How comes you are using gphoto2 2.5.23 with libgphoto2 2.5.26? Any particular reason?

The PPA to get a later libgphoto2 didn't include gphoto2.

g-voelker, 2020-12-08:
For some reason that I do not understand I get a 1080p feed now. Yesterday I have tried 1024x576 more than ten times and got nothing. The camera is on video mode and the following works like a charm:

`sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2 card_label="EOS M50"
gphoto2 --stdout --set-config viewfinder=1 --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -s 1920:1080 -f v4l2 /dev/video2`

I am observing this.

PS: The reason I wanted to try this is because the discord app seems to accept this resolution but not the lower one so I wanted to test the resolution and expected to see nothing useful but was surprised to get a good image :-D

rodolpheh, 2020-12-13:
Just reproduced this, webcamtests.com tells me I have 1920x1080. Brilliant !

sjoelund, 2020-12-13:
It's resizing from a smaller resolution though, so it's not really a 1920x1080 source.

g-voelker, 2020-12-13:
Indeed. Which leaves some off-topic questions on ffmpeg. For instance why it struggles to do the resizing for different target resolutions. Also the 1080 output image does not seem to be a mere interpolation of the input - it is subjectively too sharp for that. However, ffmpeg might just do a really good job there.
Finally the detection score of the mjpeg stream is supposedly low (warning is displayed) so that I am beginning to think that  we might be looking in the wrong direction at this point. I will see if I can take a closer look at the ffmpeg conversion during the Christmas break.

msmeissn, 2021-01-03:
So, the "output" variable actually also changes the resolution. I will see if I can make some sense of it.

I saw on my EOS 750D the resolutions:
960x640
192x128
480x320

msmeissn, 2021-01-03:
EOS 100d and EOS 1000d: only 960x640.

msmeissn, 2021-01-03:
ok, even weird:

these choices:

Choice: 6 PC + MOBILE

alternate the captured liveview between PC size and MOBILE size 

So using "PC" or "TFT + PC" is a safe bet.

shahms, 2021-01-06:
On my Canon EOS RP any of the options which include both PC and MOBILE (so 6,7,10,11) cause the camera to display an error that requires pulling the battery to reset.  When in movie mode, output={0,1,2,3} gives 1024x576 resolution, output={4,5,8,9} yields 512x288.

grssnbchr, 2021-01-06:
> > How comes you are using gphoto2 2.5.23 with libgphoto2 2.5.26? Any particular reason?
> 
> The PPA to get a later libgphoto2 didn't include gphoto2.

Slightly off-topic, but what PPA are you talking about? I'd like to update via `apt` on my system, I am currently on gphoto2 2.5.23 and libgphoto2 2.5.25, is there a way or do I have to compile from source?

EtherG, 2021-04-07:
Any news on changing capture-movie resolution? Setting up a dslr webcam has been my side project recently and I've only been getting 640x424 out of my camera, a Nikon D5000

msmeissn, 2021-04-25:
I checked the SDK, the D5000 does not support changing the liveview size. Some newer Nikons DSLR/Z allow that however.