gphoto/libgphoto2 #569: More on Nikon 1 V1 capture

simont77, 2020-10-19:
Hi again,
do you think is there any possibility to have at least Capture&Download working on the Nikon 1 V1? I though that since the V1 has and optical remote shutter, it may be more useful for long exposure than the J4, even if I could not set anything, just controlling an IR remote dongle through the PC (or using LIRC on Linux). The J4 is in any case limited to 30s.
Thanks

simont77, 2020-10-19:
Comparing capture log for the V1 and J4, it seems that V1 got stucks on ChangeCameraMode. There is already some code to ignore the failing of ChangeCameraMode on Nikon 1, maybe we should avoid the command at all?

```
/* Nilon V and J seem to like that */
	if (!params->controlmode && ptp_operation_issupported(params,PTP_OC_NIKON_ChangeCameraMode)) {
		ret = ptp_nikon_changecameramode (params, 1);
		/* FIXME: PTP_RC_NIKON_ChangeCameraModeFailed does not seem to be problematic */
		if (ret != PTP_RC_NIKON_ChangeCameraModeFailed)
			C_PTP_REP (ret);
		params->controlmode = 1;
	}
```

[capture_log_J4.txt](https://github.com/gphoto/libgphoto2/files/5400421/capture_log_J4.txt)
[capure_log_V1.txt](https://github.com/gphoto/libgphoto2/files/5400422/capure_log_V1.txt)

msmeissn, 2020-10-19:
I added code to blank out the ChangeCamera Mode on the V1, can you test git?

simont77, 2020-10-19:
uhm, now it got stuck on InitiateCaptureRecInSdram

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5401067/my-logfile.txt)

msmeissn, 2020-10-19:
i also filter

msmeissn, 2020-10-19:
i also filter a livestatus query now ... can you see if that helps?

simont77, 2020-10-19:
Still no luck. This is the log (seems nothing changed)

[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5401302/my-logfile.txt)

msmeissn, 2020-10-19:
can you try

gphoto2 --set-config opcode=0x100e --wait-event-and-download=5s  --debug --debug-logfile=100e.log

simont77, 2020-10-19:
The operation failed, BUT upon command the Camera switched on the AF assist light for a second.

```
*** Error ***              
Failed to set new configuration value 0x100e for configuration entry opcode.
Waiting for 5 seconds for events from camera. Press Ctrl-C to abort.

*** Error ***              
PTP I/O Error
*** Error (-7: 'I/O problem') ***       

^C
Cancelling...
^C
Aborting...
Aborted.
```

Attached the log. 

[100e.log](https://github.com/gphoto/libgphoto2/files/5401371/100e.log)

msmeissn, 2020-10-19:
0x100e is the generic ptp capture command ... it resulted in a "General Error":

5.142853 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x100e receiving resp failed: PTP General Error (0x2002)

but interesting that it at least tried to trigger a camera reaction

simont77, 2020-10-19:
Maybe I need to set the camera in some specific state?

msmeissn, 2020-10-19:
No, it should just work. :/ Hmm.

simont77, 2020-10-19:
Ok, now I got a shutter release, but still error on gphoto2 --set-config opcode=0x100e --wait-event-and-download=5s --debug --debug-logfile=100e.log

I though that maybe the camera did not fire the shutter because it cannot autofocus, so I place the camera in a position in which the autofocus can lock and now the shutter is released. Strange thing is that the camera is set to Manual Focus.

This is the new log: 

[100e.log](https://github.com/gphoto/libgphoto2/files/5402191/100e.log)

msmeissn, 2020-10-19:
it is still throwing errors ... but is there a new image on the card afterwards?

simont77, 2020-10-19:
Yes, the shutter is released (after focusing) and the image is saved into the card.

simont77, 2020-10-24:
...keep trying to understand if capture and download is supported on the V1, I found something that maybe is interesting.
I tried the application AstroDSLR which in the current release is based on Indigo driver (which seems a refactoring of libgphoto protocol) and it gives the same time of errors, BUT using an older release which claims to be based on Nikon SDK (which however does not officially support Nikon V1) I was able to capture and download an image. Don't know if this can help you....

simont77, 2020-10-25:
Looking back into gphoto history, you said that in 2.5.5 capure with V1 should work. I tried to look back at that code, but unfortunately you made a lot of modification to Nikon1 code. Maybe you remember something?

simont77, 2020-10-25:
https://github.com/gphoto/libgphoto2/commit/717802d87d022c40c6ee1b463b0069a3c791e5f0

Here you say that we should not use GetEvent but simply wait. If seems that this code has been removed. Do you remember something about that?

simont77, 2020-10-26:
ok, I added the NIKON_BROKEN_CAP flag for Nikon V1 and also filtered out any GetEvent call (property 90c7 is present, but any call to it stalls the communication) and now capture-image-and-download works. However the polling of the new image handle is very slow, and the command eventually throws an "PTP Invalid Object Handle" error, even if the file was properly downloaded and removed from the camera.

msmeissn, 2020-10-27:
To be frank, all this debugging mostly indicates that this camera is very unstable in capture handling.
I implemented your changes in master branch, (just did the 90c7 thing different, I overwrite it in the opcode list).

simont77, 2020-10-27:
See comment here https://github.com/gphoto/libgphoto2/commit/8fa367b9c8b1d7909c407b87d4e53a749f2fd07a#commitcomment-43612680

msmeissn, 2020-10-27:
I added code that just blanks out the 90c0 capture opcode from the opcode list, also removed broken_cap again.

so should be doing what you have in your bramnch also in master

simont77, 2020-10-27:
Ok, I will try

simont77, 2020-10-27:
Yes, it's working, thanks.

What about the rescan of all the card structure? The relevant section of the log is 

```
10.281971 get_folder_from_handle      (2): (10001,0,/store_00010001/)
10.281983 gp_filesystem_append        (2): Append /store_00010001/DCIM/100NC1V1/_DSC3237.JPG to filesystem
10.281992 lookup_folder               (2): Lookup folder '/store_00010001/DCIM/100NC1V1'...
10.281996 lookup_folder               (2): Folder / is dirty. Listing folders in there to make folder clean...
10.282000 gp_filesystem_list_folders  (2): Listing folders in /
10.282003 lookup_folder               (2): Lookup folder '/'...
10.282007 lookup_folder               (2): Found! / is 0x7f95a3d060e0
10.282010 gp_filesystem_list_folders  (2): ... is dirty, getting from camera
10.282015 folder_list_func            (2): folder_list_func(/)
10.282030 delete_all_folders          (2): Internally deleting all folders from '/'...
10.282034 lookup_folder               (2): Lookup folder '/'...
10.282037 lookup_folder               (2): Found! / is 0x7f95a3d060e0
10.282042 recurse_delete_folder       (2): Recurse delete folder 0x7f95a3d060e0//
10.282046 append_folder_one           (2): Append one folder store_00010001
10.282062 gp_filesystem_list_folders  (2): Folder / contains 1 subfolders.
10.282065 lookup_folder               (2): Done making folder / clean...
10.282072 lookup_folder               (2): Folder /store_00010001/ is dirty. Listing folders in there to make folder clean...
10.282076 gp_filesystem_list_folders  (2): Listing folders in /store_00010001/
10.282079 lookup_folder               (2): Lookup folder '/store_00010001/'...
10.282081 lookup_folder               (2): Found! /store_00010001/ is 0x7f95a3d09590
10.282084 gp_filesystem_list_folders  (2): ... is dirty, getting from camera
10.282087 folder_list_func            (2): folder_list_func(/store_00010001/)
10.282098 ptp                         (2): (storage=0x00010001, handle=0x00000000)
10.282102 folder_list_func            (2): after list folder (storage=0x00010001, handler=0x080)
10.282105 folder_list_func            (2): adding 0x11004000 / ob=0x7f95a3d0b570 to folder
10.282109 delete_all_folders          (2): Internally deleting all folders from '/store_00010001/'...
10.282112 lookup_folder               (2): Lookup folder '/store_00010001/'...
10.282115 lookup_folder               (2): Found! /store_00010001/ is 0x7f95a3d09590
10.282118 recurse_delete_folder       (2): Recurse delete folder 0x7f95a3d09590/store_00010001
10.282132 append_folder_one           (2): Append one folder DCIM
10.282136 gp_filesystem_list_folders  (2): Folder /store_00010001/ contains 1 subfolders.
10.282139 lookup_folder               (2): Done making folder /store_00010001/ clean...
10.282142 lookup_folder               (2): Folder /store_00010001/DCIM/ is dirty. Listing folders in there to make folder clean...
10.282145 gp_filesystem_list_folders  (2): Listing folders in /store_00010001/DCIM/
10.282148 lookup_folder               (2): Lookup folder '/store_00010001/DCIM/'...
10.282150 lookup_folder               (2): Found! /store_00010001/DCIM/ is 0x7f95a3d069e0
10.282153 gp_filesystem_list_folders  (2): ... is dirty, getting from camera
10.282156 folder_list_func            (2): folder_list_func(/store_00010001/DCIM/)
10.282159 ptp                         (2): (storage=0x00010001, handle=0x00000000)
10.282162 ptp                         (2): (storage=0x00010001, handle=0x11004000)
10.282165 ptp                         (2): Listing ... 
10.282171 ptp_usb_sendreq             (2): Sending PTP_OC 0x1007 (Get object handles) (0x10001,0x0,0x11004000) request...
10.282175 gp_port_write               (3): Writing 24 = 0x18 bytes to port...
```

and after that a lot of GetObjectHandle/GetObjectInfo are issued. All are without errors, but they take a lot of time if the card has many photos on it.

msmeissn, 2020-10-27:
This will need a bit longer work I think.

I general we carefully update the filesystem, not sure what triggered a full reread here.

We need to read it at least once though.

simont77, 2020-10-27:
So so the first time I call gphoto you create a sort of snapshot of the filesystem of the card? Where it is stored? At the end of the log there is this part where apparently everything is cleared 

```
9.060946 gp_filesystem_reset         (2): resetting filesystem
9.060954 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
9.060957 gp_filesystem_lru_clear     (2): fscache LRU list already empty
9.060973 delete_all_folders          (2): Internally deleting all folders from '/'...
9.060976 lookup_folder               (2): Lookup folder '/'...
9.060980 lookup_folder               (2): Found! / is 0x7fc572e060b0
9.060983 recurse_delete_folder       (2): Recurse delete folder 0x7fc572e060b0//
9.060986 recurse_delete_folder       (2): Recurse delete folder 0x7fc572d06b30/store_00010001
9.060989 recurse_delete_folder       (2): Recurse delete folder 0x7fc572d06910/DCIM
9.060992 recurse_delete_folder       (2): Recurse delete folder 0x7fc572d06940/100NC1V1
9.060995 delete_folder               (2): Delete one folder 0x7fc572d06940/100NC1V1
9.060998 delete_all_files            (2): Delete all files in folder 0x7fc572d06940/100NC1V1
9.061003 delete_folder               (2): Delete one folder 0x7fc572d06910/DCIM
9.061006 delete_all_files            (2): Delete all files in folder 0x7fc572d06910/DCIM
9.061009 delete_folder               (2): Delete one folder 0x7fc572d06b30/store_00010001
9.061012 delete_all_files            (2): Delete all files in folder 0x7fc572d06b30/store_00010001
9.061015 delete_all_files            (2): Delete all files in folder 0x7fc572e060b0//
9.061018 gp_port_free                (2): Freeing port...
9.061021 gp_port_close               (2): Closing port...
9.061382 gp_filesystem_reset         (2): resetting filesystem
9.061395 gp_filesystem_lru_clear     (2): Clearing fscache LRU list...
9.061400 gp_filesystem_lru_clear     (2): fscache LRU list already empty
9.061403 delete_all_folders          (2): Internally deleting all folders from '/'...
9.061406 lookup_folder               (2): Lookup folder '/'...
9.061410 lookup_folder               (2): Found! / is 0x7fc572e060b0
9.061414 recurse_delete_folder       (2): Recurse delete folder 0x7fc572e060b0//
9.061417 delete_all_files            (2): Delete all files in folder 0x7fc572e060b0//
```

is it intended?

simont77, 2020-10-27:
uhm...in shell mode only the first capture is long, the followings are fast, so maybe it is the normal way of operation I guess

simont77, 2020-10-27:
also in timelapse mode, from the second shoot there is no additional delay due to filesystem scanning

msmeissn, 2020-10-28:
libgphoto2 provides a virtual filesystem. it is filled on demand...

The ptp2 driver which drives the Nikon camera also caches all filesystem information from the camera and feeds the libgphoto2 virtual filesystem.

Due to the nature of PTP filesystem, to find out the correct path and name of a newly captured file, the whole directory containing the file must be read, and also the lower level ones in the chain.

SO if there is a lot of stuff in DCIM/ 100NC1V1/ it needs to be queried first.

simont77, 2020-10-28:
Thanks for the explanation, no it's clear that this is the intended behavior. 

So now capture is working, but it seems weird to me that V1 exposes a lot of params but apparently you cannot even ask for their values. Maybe there is some hidden param to activate remote control. Without bothering you further, I would like to do some experiment on my own. Do you have any suggestion?

simont77, 2021-01-19:
Hi, I've been using libgphoto in indi with the Nikon V1 for some time, and it seems quite reliable, however I found a problem with long exposure (>10s) similar to other timeout issues with other cameras.

After some investigation, I found that in the fallback capture path used by V1, the timeout is set to CAPTURE_VALUE only after the call to ptp_initiatecapture See here: https://github.com/gphoto/libgphoto2/blob/a6e324e068e72ac6e680968a3d6fd9ed1384cc9c/camlibs/ptp2/library.c#L5507. I moved the timeout setting before capture and I got rid of the timeout, however I still see a weird behavior when testing capture with gphoto. I got a ptp_general_error right after the long exposure is completed, BUT then the file is correctly downloaded to the PC. For exposure up to 10s I got no error at all. It seems there is still some timeout to be increased.
This is the relevant part of the 

```
log:0.195794 gp_port_set_timeout         (2): Setting port timeout to 100000 milliseconds.
0.195829 camera_capture [library.c:5515](0): 'gp_port_set_timeout (camera->port, capture_timeout)' OK: 'No error' (0)
0.195848 ptp_usb_sendreq             (2): Sending PTP_OC 0x100e (Initiate capture) (0x0,0x0) request...
0.195861 gp_port_write               (3): Writing 20 = 0x14 bytes to port...
0.195950 gp_port_write               (3): Wrote   20 = 0x14 bytes to port: (hexdump of 20 bytes)
0000  14 00 00 00 01 00 0e 10-05 00 00 00 00 00 00 00  ................
0010  00 00 00 00            -                         ....            

0.195990 ptp_usb_getresp             (2): Reading PTP_OC 0x100e (Initiate capture) response...
0.196032 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
25.460562 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 02 20-05 00 00 00              ....... ....    

25.460673 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x100e receiving resp failed: PTP General Error (0x2002)
25.460722 camera_capture [library.c:5520](0): 'ptp_initiatecapture(params, 0x00000000, 0x00000000)' failed: 'PTP General Error' (0x2002)
25.460794 gp_context_error            (0): PTP General Error
25.460960 gp_camera_capture [gphoto2-camera.c:1340](0): 'camera->functions->capture (camera, type, path, context)' failed: -1
25.461176 camera_wait_for_event       (2): waiting for events timeout 3000 ms
25.461231 gp_port_get_timeout         (2): Current port timeout is 100000 milliseconds.
25.461269 gp_port_set_timeout         (2): Setting port timeout to 3000 milliseconds.
25.461311 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
26.542302 _cb_irq                     (2): 0xd099b8 with status 0
26.542389 _cb_irq                     (3): interrupt (hexdump of 16 bytes)
0000  10 00 00 00 04 00 02 40-ff ff ff ff 7e 1a 19 09  .......@....~...

26.542426 _cb_irq                     (2): Requeuing completed transfer 0xd099b8
26.542516 gp_port_check_int           (3): Read    16 = 0x10 out of 24 bytes from interrupt endpoint: (hexdump of 16 bytes)
0000  10 00 00 00 04 00 02 40-ff ff ff ff 7e 1a 19 09  .......@....~...

26.542559 ptp                         (2): event: nparams=0x1, code=0x4002, trans_id=0xFFFFFFFF, p1=0x9191A7E, p2=0x0, p3=0x0
26.542593 gp_port_set_timeout         (2): Setting port timeout to 100000 milliseconds.
26.542655 ptp_usb_sendreq             (2): Sending PTP_OC 0x1008 (Get object info) (0x9191a7e) request...
26.542689 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
26.542877 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 08 10-06 00 00 00 7e 1a 19 09  ............~...

26.542922 ptp_usb_getdata             (2): Reading PTP_OC 0x1008 (Get object info) data...
26.542958 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
26.698029 gp_port_read                (3): Read    158 = 0x9e out of 1024 bytes from port: (hexdump of 158 bytes)
0000  9e 00 00 00 02 00 08 10-06 00 00 00 01 00 01 00  ................
0010  00 30 00 00 5f 88 6d 00-08 38 ff 11 00 00 a0 00  .0.._.m..8......
0020  00 00 78 00 00 00 20 0f-00 00 20 0a 00 00 00 00  ..x... ... .....
0030  00 00 00 00 19 11 00 00-00 00 00 00 00 00 00 00  ................
0040  0d 5f 00 44 00 53 00 43-00 36 00 37 00 38 00 32  ._.D.S.C.6.7.8.2
0050  00 2e 00 4e 00 45 00 46-00 00 00 10 32 00 30 00  ...N.E.F....2.0.
0060  32 00 31 00 30 00 31 00-31 00 39 00 54 00 31 00  2.1.0.1.1.9.T.1.
0070  32 00 32 00 37 00 35 00-34 00 00 00 10 32 00 30  2.2.7.5.4....2.0
0080  00 32 00 31 00 30 00 31-00 31 00 39 00 54 00 31  .2.1.0.1.1.9.T.1
0090  00 32 00 32 00 37 00 35-00 34 00 00 00 00        .2.2.7.5.4....  

26.698127 ptp_usb_getresp             (2): Reading PTP_OC 0x1008 (Get object info) response...
26.698167 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
26.698321 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-06 00 00 00              ....... ....    

26.698621 get_folder_from_handle      (2): (10001,11190000,/store_00010001/)
26.698683 ptp_usb_sendreq             (2): Sending PTP_OC 0x1008 (Get object info) (0x11190000) request...
26.698723 gp_port_write               (3): Writing 16 = 0x10 bytes to port...
26.698868 gp_port_write               (3): Wrote   16 = 0x10 bytes to port: (hexdump of 16 bytes)
0000  10 00 00 00 01 00 08 10-07 00 00 00 00 00 19 11  ................
```

Do you have any idea?

simont77, 2021-01-31:
Hi, any idea on this? At the moment I filtered the error and got 30s exposure and download with no problem, but I think this is not a proper way to solve the issue.

msmeissn, 2021-02-01:
the 0x2002 general error comes from the camera.

I filtered that out now too, also moved the timeout.

simont77, 2021-02-01:
Thanks. I will try out your latest commit.

msmeissn, 2021-02-01:
> Thanks for the explanation, now it's clear that this is the intended behavior.
> 
> So now capture is working, but it seems weird to me that V1 exposes a lot of params but apparently you cannot even ask for their values. Maybe there is some hidden param to activate remote control. Without bothering you further, I would like to do some experiment on my own. Do you have any suggestion?

to be frank, i have concerns regarding the stability of the V1 firmware, i would largely consider any issues we see due to hot-needled firmware for the first nikon 1 series camera.

simont77, 2021-02-01:
well, in general the camera does not show any issue. There are some limitations, but not bugs. On the other hand, the ptp part seems to be only developed in half. Probably it is there only for test purposes and not meant for real use. But, again, the few that works is reliable. On one night, I shot one image every 2 seconds for more then 2 hours (checking telescope drifting) without any error.

msmeissn, 2021-02-01:
yes, that might fit it better. we can still go and look for workarounds / enhancements ;)