gphoto/libgphoto2 #570: Nikon equivalent to canon eos immediate release

Patrick-Lascombe, 2020-10-22:
Hello,
For a project we are migration from Canon EOS RP to Nikon Z7. We need to take pictures at a high rate so we used the config eosremoterelease="Immediate", but I couldn't find a similar config with the Nikon Z7.

Am I missing something here ?

Thanks for your help and sorry for the trivial question

msmeissn, 2020-10-22:
There is no direct equivalent to the EOS direct shutter manipulation on Nikons.

I have a autofocus=no  setting which might help, or you can try setting it to manual focus.

and use --trigger-capture   to trigger capturing wthout download

Patrick-Lascombe, 2020-10-22:
The problem is that I'm trying to activate the shutter inside a cpp code, I don't have access to the CLI version of gphoto2, I'll look up how the trigger-capture use the libgphoto2 to try to mimic it.
Thanks

closed 2020-10-26 by Patrick-Lascombe

msmeissn, 2020-10-27:
its an API call gp_camera_trigger_capture ;)