gphoto/libgphoto2 #571: Compile issue on Raspberry Pi OS

starlord1980, 2020-10-22:
So running 'sudo apt install gphoto2' on PiOS currently pulls this version:
```
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

I am using a Sony a6300 which stops capturing after taking one shot so I noticed theres been some improvements for Sony on v2.5.26 so want to try that but it seems i need to compile it manually.

I've installed all the dependencies its asked for along all the trial compiles..

```
   61  sudo apt-get install libgphoto2-dev
   64  sudo apt install popt
   67  sudo apt install libpopt-dev
   74  sudo apt-get install autoconf
   76  sudo apt-get install autopoint
   79  sudo apt-get install libtool libtool-bin
  108  autoreconf -is
  109  ./configure 
  110  sudo make
```

I now get a fairly successful compile up to here:

```
make[2]: Leaving directory '/home/pi/gphoto2-2.5.26/gphoto2'
Making all in packaging
make[2]: Entering directory '/home/pi/gphoto2-2.5.26/packaging'
Making all in rpm
make[3]: Entering directory '/home/pi/gphoto2-2.5.26/packaging/rpm'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/home/pi/gphoto2-2.5.26/packaging/rpm'
make[3]: Entering directory '/home/pi/gphoto2-2.5.26/packaging'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/home/pi/gphoto2-2.5.26/packaging'
make[2]: Leaving directory '/home/pi/gphoto2-2.5.26/packaging'
Making all in po
make[2]: Entering directory '/home/pi/gphoto2-2.5.26/po'
test -z "az.gmo cs.gmo da.gmo de.gmo en_GB.gmo es.gmo eu.gmo fi.gmo fr.gmo hu.gmo id.gmo is.gmo it.gmo ja.gmo nl.gmo pa.gmo pl.gmo pt_BR.gmo ro.gmo ru.gmo rw.gmo sk.gmo sr.gmo sv.gmo uk.gmo vi.gmo zh_CN.gmo zh_TW.gmo" || make az.gmo cs.gmo da.gmo de.gmo en_GB.gmo es.gmo eu.gmo fi.gmo fr.gmo hu.gmo id.gmo is.gmo it.gmo ja.gmo nl.gmo pa.gmo pl.gmo pt_BR.gmo ro.gmo ru.gmo rw.gmo sk.gmo sr.gmo sv.gmo uk.gmo vi.gmo zh_CN.gmo zh_TW.gmo
make[3]: Entering directory '/home/pi/gphoto2-2.5.26/po'
: --update az.po gphoto2.pot
rm -f az.gmo && : -c --statistics -o az.gmo az.po
mv: cannot stat 't-az.gmo': No such file or directory
make[3]: *** [Makefile:116: az.gmo] Error 1
make[3]: Leaving directory '/home/pi/gphoto2-2.5.26/po'
make[2]: *** [Makefile:140: stamp-po] Error 2
make[2]: Leaving directory '/home/pi/gphoto2-2.5.26/po'
make[1]: *** [Makefile:465: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/gphoto2-2.5.26'
make: *** [Makefile:397: all] Error 2
```

This is where I am now stuck.

msmeissn, 2020-10-22:
you need to install "gettext" too I think. try with itinstalled

starlord1980, 2020-10-22:
Thanks for the quick reply, yep that worked.
Presumably it would be beneficial up compile the latest libgphoto2 as well and then give it a test?

pi@octopi:~/gphoto2-2.5.26 $ gphoto2 -v
gphoto2 2.5.26

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

starlord1980, 2020-10-22:
Not sure why libgphoto2 is compiling to v2.5.22 when I downloaded libgphoto2-2.5.26.tar.gz?

I followed these instructions and started again compiling libgphoto2-2.5.26 first, _then_ gphoto2-2.5.26
http://cholla.mmto.org/photography/software/gphoto2/build1.html

```
gphoto2 -v
gphoto2 2.5.26
....
gphoto2         2.5.26         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
```
Does it matter that these are mismatched?

ndim, 2020-10-22:
If you are going through the effort of compiling yourself, that will make a significant difference in feature set and bug set. (Debian and thus Raspbian are note famous for their quick update cycle.)

After installing `libgphoto2` and before running `configure` for the gphoto2 build, set `PKG_CONFIG_PATH` or `PKG_CONFIG_LIBDIR` to point to the directory where the `libgphoto2.pc` file has been installed (`/home/usr/.local/lib/pkgconfig/libgphoto2.pc` or `/usr/local/lib64/pkgconfig/libgphoto2.pc`).

Then the `configure` run for `gphoto2` should pick up the `libgphoto2` installation from that location.

starlord1980, 2020-10-28:
Weol tried that and I got this:

```
pi@octopi:~/gphoto2-2.5.26 $ pkg-config --path libgphoto2
Unknown option --path

```
Googled around and found that I should install pkgconf, which removed pkg-config
Did the exports:

```
export PKG_CONFIG_PATH=/usr/local/lib/pkgconfig
export PKG_CONFIG_LIBDIR=/usr/local/lib/pkgconfig
```

Ran the command:
```
pi@octopi:~/gphoto2-2.5.26 $ pkgconf --path libgphoto2
/usr/local/lib/pkgconfig/libgphoto2.pc
```

`./configure && sudo make && sudo make install
`
```
pi@octopi:~/gphoto2-2.5.26 $ which gphoto2
/usr/local/bin/gphoto2
pi@octopi:~/gphoto2-2.5.26 $ gphoto2 -v
gphoto2 2.5.26

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26         gcc, popt(m), no exif, no cdk, no aa, jpeg, no readline
libgphoto2      **2.5.22**         all camlibs, gcc, ltdl, EXIF

```

Still not working for me

ndim, 2020-10-29:
I have just tried to reproduce this on my Raspberry Pi 2B+ with `/etc/debian_version` being `10.6`. `$PATH` contains `/usr/local/bin`, no `PKG_*` variables set.

git cloned libgphoto2, then ran `./configure --with-camlibs=ptp2,canon,directory && make && sudo make install` without issues (limiting the camlib set to accelerate the build).

Then I confirmed that `pkg-config` picks up `libgphoto2.pc` from the correct location in `/usr/local` (as the default implementation of `pkg-config` does not support `--path`):

```
[user@host ~]$ pkg-config --libs libgphoto2
-L/usr/local/lib -lgphoto2 -lm -lgphoto2_port -lm
[user@host ~]$ 
```

OK, so far, so good.

git cloned gphoto2, then ran `./configure && make && sudo make install`. `configure` failed because popt had not been installed, but one `sudo apt-get install libpopt-dev` later, `./configure && make && sudo make install` ran through successfully.

Hooray! Now let's run `gphoto2 --version`:

```
[user@host ~]$ gphoto2 --version
gphoto2: error while loading shared libraries: libgphoto2.so.6: cannot open shared object file: No such file or directory
[user@host ~]$ ldd /usr/local/bin/gphoto2
	/usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v6l.so (0xb6f5a000)
	libgphoto2.so.6 => not found
	libgphoto2_port.so.12 => not found
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0xb6f30000)
	libpopt.so.0 => /usr/lib/arm-linux-gnueabihf/libpopt.so.0 (0xb6f15000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0xb6e93000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0xb6d45000)
	/lib/ld-linux-armhf.so.3 (0xb6f6d000)
[user@host ~]$ 
```

Weird. Checking `/usr/local/lib/`, there *is* the symlink `libgphoto2.so.6 -> libgphoto2.so.6.1.0` and the file `libgphoto2.so.6.1.0`. `/etc/ld.so.conf.d/libc.conf` contains `/usr/local/lib`... so the library should be picked up from there.

Let's see what `ldconfig` says:

```
[user@host ~]$ sudo ldconfig --verbose
[...]
/usr/local/lib:
	libgphoto2.so.6 -> libgphoto2.so.6.1.0
	libgphoto2_port.so.12 -> libgphoto2_port.so.12.0.0
[...]
[user@host ~]$ 
```

Hmm. Might this have been the issue?

```
[user@host ~]$ gphoto2 --version
gphoto2 2.5.26.1
[...]
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26.1       gcc, popt(m), no exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.26.1       NON-STANDARD CAMLIB SET (canon directory ptp2), gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial, gcc, ltdl, no EXIF, no USB, serial without locking
[user@host ~]$ 
```

Apparently so.

Now... why some Raspbian package installs a file to `/etc/ld.so.conf.d/` containing `/usr/local/lib` without `ldconfig` picking that up, I do not understand. Might be a Raspbian problem. Might be a Debian problem. Might be intended behaviour. No idea.

So... just to test out what would happen I installed the distro provided `libgphoto2-6` package:

```
[user@host ~]$ sudo apt-get install libgphoto2-6
[...]
[user@host ~]$ gphoto2 --version
gphoto2 2.5.26.1
[...]
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26.1       gcc, popt(m), no exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
[user@host ~]$ 
```

Re-running `sudo ldconfig --verbose` did not change that behaviour.

OK... so apparently the `gphoto2` executable does not hardcode the location of `libgphoto2.so.6` it links against, and the distro provided `ld.so` dynamic linker configuration prioritizes the distro-provided libraries from `/usr/lib/arm-linux-gnueabihf` those installed locally to `/usr/local/lib`.

As you probably want to use the self-compiled `libgphoto2.so.6` with other distro provided packages linking against `libgphoto2.so.6`, removing the `libgphoto2-6` distro package is not an option because that would remove all depending packages as well. This means the only option left is to re-configure ld.so to prefer the self-compiled `libgphoto2.so.6` from `/usr/local/lib` over the distro-provied `libgphoto2.so.6` from `/usr/lib/arm-linux-gnueabihf`.

Setting `LD_LIBRARY_PATH` would be the quick and dirty option (but might require a reboot until all process have picked up the changed environment), while properly setting up the config files in `/etc/ld.so.conf.d` should work instantly after running `ldconfig`.

TL;DR: You probably do not have a compile and build-time link issue, but a run-time dynamic linking issue.

ndim, 2020-10-29:
As it turns out, instead of the first `ldconfig --verbose`, I should have run this instead:

```
[user@host ~]$ ldconfig -p | grep gphoto2
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so.12
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so.12
	libgphoto2_port.so (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so
	libgphoto2.so.6 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2.so.6
	libgphoto2.so.6 (libc6,hard-float) => /usr/local/lib/libgphoto2.so.6
	libgphoto2.so (libc6,hard-float) => /usr/local/lib/libgphoto2.so
[user@host ~]$ 
```

Let's try how to put `/usr/local/lib` more to the front of the list of directories where ld.so looks for libraries, starting with a manual attempt:

```
[user@host ~]$ ldconfig --verbose /usr/local/lib
[...]
[user@host ~]$ ldconfig -p | grep gphoto2
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so.12
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so.12
	libgphoto2_port.so (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so
	libgphoto2.so.6 (libc6,hard-float) => /usr/local/lib/libgphoto2.so.6
	libgphoto2.so.6 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2.so.6
	libgphoto2.so (libc6,hard-float) => /usr/local/lib/libgphoto2.so
[user@host ~]$ 
```

However, I doubt this would last long, as I would expect many system packages to run ldconfig on install or uninstall, which will then run `ldconfig` without the `/usr/local/lib` command line argument.

So, let's add `/usr/local/lib` to the ldconfig a bit earlier:

```
[user@host ~]$ sudo sh -c 'echo /usr/local/lib > 00-usr-local-lib.conf'
[user@host ~]$ sudo ldconfig
[user@host ~]$ ldconfig -p | grep gphoto2
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so.12
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so.12
	libgphoto2_port.so (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so
	libgphoto2.so.6 (libc6,hard-float) => /usr/local/lib/libgphoto2.so.6
	libgphoto2.so.6 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2.so.6
	libgphoto2.so (libc6,hard-float) => /usr/local/lib/libgphoto2.so
[user@host ~]$ 
```

There we go. The issue is the sequence in which the directories are loaded from `/etc/ld.so.conf.d/` from:

```
[user@host ~]$ ls -1 /etc/ld.so.conf.d/
00-usr-local-lib.conf
00-vmcs.conf
arm-linux-gnueabihf.conf
fakeroot-arm-linux-gnueabihf.conf
libc.conf
[user@host ~]$ 
```

I have no idea why not all of those files are prefixed with numbers to define a sequence of them being loaded. I also have no idea why the `libc.conf` file which adds `/usr/local/lib` is not named something else to put it in front.

I think this should fix your immediate issue. No idea whether this issue should be reported up to Raspbian or Debian.

bezineb5, 2020-11-04:
Did you try https://github.com/gonzalo/gphoto2-updater ? It works fine for me, I have no issue on Raspbian or on debian containers on Raspberry Pi with it.

starlord1980, 2020-11-04:
I havent but that sounds right up my street!
Ran it and this is the version I got:

> pi@octopi:~ $ gphoto2 -v
gphoto2 2.5.23.1
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF

Hmmm.. I actually got a downgrade of gphoto2 and no change on libgphoto2.
Theres also ndims post above I've yet to have a chance to look into.

Where does it come from when you install it from apt? Does the author have to update it somewhere for it to become available via apt install?

bezineb5, 2020-11-04:
> EDIT: although that script specifically states its installing the latest 2.5.26... maybe I'll test it on a fresh raspian install.

Yes, I think it doesn't overwrite the raspbian-installed version, indeed.

ndim, 2020-11-04:
The `gphoto2-updater.sh` script mentioned above runs `./configure` without any parameters for both `libgphoto2` and `gphoto2`. This means that it installs to the default install prefix of `/usr/local`. Which in turn means that you still need to take care of configuring `ld.so` as I have described above.

starlord1980, 2020-11-04:
> > EDIT: although that script specifically states its installing the latest 2.5.26... maybe I'll test it on a fresh raspian install.
> 
> Yes, I think it doesn't overwrite the raspbian-installed version, indeed.

Well tried that in fresh raspian and got the same result. Maybe that means it not me doing something wrong after all - just me stumbling around but learning how to compile as I go along!

But it looks like lockdown 2.0 has started so ndim I'm going to follow your post now I have some time free!

starlord1980, 2020-11-04:
It worked!!

> pi@octopi:~ $ gphoto2 -v
gphoto2 2.5.26.1
gphoto2         2.5.26.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.26.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

Ok, so its clicked now how that was working, I mistakenly thought it was compiling a single binary with libgphoto2 included, but I guess its just referencing them.

I had a few more of these links than you had, so if theres an order to it I guess these are my actual issue:

> pi@octopi:/usr/lib/arm-linux-gnueabihf $ ldconfig -p | grep gphoto2
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so.12
	libgphoto2_port.so.12 (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so.12
	**libgphoto2_port.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2_port.so**
	libgphoto2_port.so (libc6,hard-float) => /usr/local/lib/libgphoto2_port.so
	**libgphoto2.so.6 (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2.so.6**
	libgphoto2.so.6 (libc6,hard-float) => /usr/local/lib/libgphoto2.so.6
	**libgphoto2.so (libc6,hard-float) => /usr/lib/arm-linux-gnueabihf/libgphoto2.so**
	libgphoto2.so (libc6,hard-float) => /usr/local/lib/libgphoto2.so

I wasnt quite understanding what you did there to fix the order, but then it occurred to me that maybe the issue I've been having then is because I'd previously installed libgphoto2 from apt causing it to pick up from the /usr/lib/arm-linux-gnueabihf/ directory so...

> pi@octopi:/ $ sudo apt remove libgphoto2-6
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libaa1 libcdk5nc6 libgphoto2-l10n libgphoto2-port12 libllvm8 rpi-eeprom-images
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  libgphoto2-6 libgphoto2-dev
0 upgraded, 0 newly installed, 2 to remove and 6 not upgraded.
After this operation, 6,781 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 53746 files and directories currently installed.)
Removing libgphoto2-dev:armhf (2.5.22-3) ...
Removing libgphoto2-6:armhf (2.5.22-3) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for udev (241-7~deb10u4+rpi1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...

> pi@octopi:/ $ gphoto2 -v
[...]
gphoto2         2.5.26         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.26.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

Ahaaa!

So then I just used bezineb5's gphoto2-updater to bring it back to 2.5.26.1 and....

> pi@octopi:~ $ gphoto2 -v
gphoto2 2.5.26.1
[...]
gphoto2         2.5.26.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.26.1       standard camlibs, gcc, ltdl, EXIF

Now I just hope this build works with my a6300 and that this was all worth it and I dont have to buy a new camera!

Thanks all!

ndim, 2020-11-05:
I thought you might want to use software packages from Raspberry Pi OS linking to `libgphoto2.so.6` with your self-compiled OS, that is why I suggested you change the configuration of the `ld.so` dynamic linker to load the self-compiled library from `/usr/local/lib` before the ones supplied by the OS software packages. Then OS-supplied, say, digikam will run with your self-compiled `libgphoto2.so.6`.

If you just want to use the `gphoto2` command line client, you do not even need to do a system wide install into `/usr/local` at all. You can just install somewhere in `$HOME`. I will need to check whether those things still work the way they used to 10 to 20 years ago when I did most of my work on gphoto.