gphoto/libgphoto2 #575: Panasonic G9 list-config null/error

rmonat, 2020-10-26:
**Describe the bug**

I was trying to setup the Panasonic G9 as a webcam (using `--capture-movie`, and then ffmpeg+v4l2). The camera was connected by USB and I chose the tethering mode. Liveview works in 640x360, but since the [Lumix Tether Tool seems to support setting a higher live view resolution](https://www.youtube.com/watch?v=EH4orJaPPmI&t=123), I wanted to try that. 
When calling `--list-config` on my Panasonic G9 (a camera close to the GH5), I get the following:
```
                                                                               
*** Error ***              
(null)

*** Error ***              
(null)

*** Error ***              
(null)

*** Error ***              
(null)

*** Error ***              
(null)

*** Error ***              
(null)
/main/actions/viewfinder
/main/actions/opcode
/main/settings/capturetarget
/main/status/serialnumber
/main/status/manufacturer
/main/status/cameramodel
/main/status/deviceversion
/main/status/vendorextension
/main/status/batterylevel
/main/other/5001
```
I guess there are some options missing and hidden by the errors?

**Name the camera**

```
$ gphoto2 --auto-detect
Model                          Port                                            
----------------------------------------------------------
Panasonic DC-GH5               usb:001,011
$ lsusb
Bus 001 Device 011: ID 04da:2382 Panasonic (Matsushita) DC-G9
```

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version: `2.5.26.1` (compiled from git just now).

**To Reproduce**

Attached is the debug file for the `--list-config` call.

rmonat, 2021-01-24:
Also, Panasonic now provides an SDK for the G9: https://av.jpn.support.panasonic.com/support/global/cs/soft/tool/sdk.html, if that may help.

msmeissn, 2021-01-31:
this was something i had not yet reverse engineered. the SDK seems helpful, I need to test this on a panasonic though , lets see when I find time to rent one again

msmeissn, 2021-01-31:
also the null ptr derefs seem to be bugs :/

msmeissn, 2021-01-31:
if you want to try something already, can you run:

gphoto2 --set-config opcode=0x9414,0x0d800012  --debug --debug-logfile=allformats.log

gphoto2 --set-config opcode=0x9414,0x0d800011  --debug --debug-logfile=currentformats.log

and attach both .log files here

totoantibes, 2021-02-10:
here is my view of the logs. note that i connected the GH5S (not GH5). 
[currentformats.log](https://github.com/gphoto/libgphoto2/files/5958304/currentformats.log)
[allformats.log](https://github.com/gphoto/libgphoto2/files/5958305/allformats.log)

I also have a USB wireshark dump avail upon demand if that can help.

I can also share the C++ sdk that is published by pana.

msmeissn, 2021-02-11:
Thanks! I will review :)

I will also rent a panasonic over the weekend and try to improve these parts.

msmeissn, 2021-02-13:
I fixed the (null) errors already

totoantibes, 2021-02-13:
do you need the Pana SDK?

msmeissn, 2021-02-13:
i have downloaded one already.

msmeissn, 2021-02-13:
i can extract it, but setting it , while reporting sucecss, currently does not change the preview size :/

msmeissn, 2021-02-14:
gphoto2 --get-config liveviewsize
select the 1280x... string ... 

then running:
gphoto2 --set-config liveviewsize="1280x960 700 30HZ" --capture-movie=10s
now brings me a 1280x960 video stream.

totoantibes, 2021-02-14:
hmmm. trying to 'make' and it hangs at ptp2/config.c

```
ptp2/config.c:10570:24: warning: format ‘%ld’ expects argument of type ‘long int’, but argument 3 has type ‘uint64_t’ {aka ‘long long unsigned int’} [-Wformat=]
     sprintf (buf, fmt, dpd.CurrentValue.val); \
                        ~~~~~~~~~~~~~~~~~~~~~~~~
     gp_widget_set_value (widget, buf); \
     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    }\
    ~~
    break;
    ~~~~~~

   X(PTP_DTC_INT8,i8,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_UINT8,u8,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_INT16,i16,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_UINT16,u16,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_INT32,i32,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_UINT32,u32,"%d")
   ~~~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_INT64,i64,"%ld")
   ~~~~~~~~~~~~~~~~~~~~~~~~~~
   X(PTP_DTC_UINT64,u64,"%ld")
   ~~~~~~~~~~~~~~~~~~~~ ^~~~~
ptp2/config.c:10558:19: note: in definition of macro ‘X’
     sprintf (buf, fmt, dpd.CurrentValue.val); \
```

msmeissn, 2021-02-14:
it should not hang during compilation.... how long did you wait?

totoantibes, 2021-02-14:
> it should not hang during compilation.... how long did you wait?

Close to 1 hour!

totoantibes, 2021-02-14:
When in doubt ReBouT. Managed to "make". But I see there are new commits now. will recompile tomorrow and test then.

totoantibes, 2021-02-14:
either i was not successful in compiling the new version  or the problem persists.
i still errors in the launch gphoto2 --list config
`
*** Error ***
(null)

*** Error ***
(null)

*** Error ***
(null)

*** Error ***
(null)

*** Error ***
(null)

*** Error ***
(null)
/main/actions/viewfinder
/main/actions/opcode
/main/settings/capturetarget
/main/status/serialnumber
/main/status/manufacturer
/main/status/cameramodel
/main/status/deviceversion
/main/status/vendorextension
/main/status/batterylevel
/main/other/5001`

attached is the debug log of gphoto2 --get-config liveviewsize 
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5978690/my-logfile.txt)

msmeissn, 2021-02-15:
its not the current git state, did you configure with the correct --prefix and --libdir to overwrite the exisitng 2.5.26.1 build?

totoantibes, 2021-02-15:
i just re-cloned and reconfigure the full repo. rebuilt. checked the lib versions and dates (all in /usr/lib) and i still get  the null output.
[my-logfile (1).txt](https://github.com/gphoto/libgphoto2/files/5982230/my-logfile.1.txt)

issue may be that i am connecting a GH5**S** not a GH5. **S** has a smaller sensor/resolution

msmeissn, 2021-02-15:
can you checjk timestamp of 

ls -l /usr/lib/arm-linux-gnueabihf/libgphoto2/2.5.26.1/ptp2.so

msmeissn, 2021-02-15:
(it should be from your buildtime ... if not, the new ptp2.so was installed somewhere else  ;)

totoantibes, 2021-02-17:
./configure --libdir=/usr/lib/arm-linux-gnueabihf/   helped in putting the SO in the right directory. now gphoto2 --get-config liveviewsize gives a full output! 

Thanks for your patience. 
Robert

msmeissn, 2021-02-17:
very good to hear :) thanks for feedback, enjoy :)