gphoto/libgphoto2 #576: Olympus E-M1.2 bulb multiplication

coldfire22x, 2020-10-31:
**Describe the bug**

When using bulb, the time passed is multiplied by 3.

`gphoto2 --bulb=5 --capture-image`

Results in a 15s exposure.

**Name the camera**

Olympus OM-D E-M1 Mark II (firmware 3.3)

```
$ gphoto2 --auto-detect
Model                          Port
----------------------------------------------------------
Olympus E-M1                   usb:002,006

$ lsusb -s 2:6
Bus 002 Device 006: ID 07b4:0130 Olympus Optical Co., Ltd E-M1MarkII
```

**libgphoto2 and gphoto2 version**

```
$ gphoto2 --version
gphoto2 2.5.26

Copyright (c) 2000-2020 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.26         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

**To Reproduce**

Camera set to M, ISO 200, BULB, f/4.

```
$ gphoto2 --bulb=3 --capture-image --debug --debug-logfile=debug.log
Bulb mode enabled (exposure time: 3s).
New file is in location /store_00010001/DCIM/100OLYMP/EM120013.ORF on the camera
```

This results in a 15s exposure (the attached debug.log is from this command).

Now interestingly, when run with more complex parameters, the exposure times are inconsistent.

```
$ gphoto2 --frames=5 --interval=1 --bulb=5 --capture-image
Time-lapse mode enabled (interval: 1s).
Bulb mode enabled (exposure time: 5s).
Capturing frame #1/5...
not sleeping (16 seconds behind schedule)
Capturing frame #2/5...
New file is in location /store_00010001/DCIM/100OLYMP/EM120003.ORF on the camera
not sleeping (12 seconds behind schedule)
Capturing frame #3/5...
New file is in location /store_00010001/DCIM/100OLYMP/EM120004.ORF on the camera
not sleeping (12 seconds behind schedule)
[debug.log](https://github.com/gphoto/libgphoto2/files/5468612/debug.log)

Capturing frame #4/5...
New file is in location /store_00010001/DCIM/100OLYMP/EM120005.ORF on the camera
not sleeping (12 seconds behind schedule)
Capturing frame #5/5...
New file is in location /store_00010001/DCIM/100OLYMP/EM120006.ORF on the camera
New file is in location /store_00010001/DCIM/100OLYMP/EM120007.ORF on the camera
```

Results in 15.02", 12.25", 12.56", 13.23", and 13.18" exposures.  I suspect the interval is somehow taking precedence over the bulb in this case, but the single bulb exposure times 3 is consistent.
[debug.log](https://github.com/gphoto/libgphoto2/files/5468614/debug.log)

msmeissn, 2021-02-27:
there are some longer hangs in the waiting time.
does it work better with

gphoto --set-config bulb=1 --wait-event=5s --set-config bulb=0 --wait-event-and-download=10s

msmeissn, 2021-02-27:
actuallly there was a bug, we were waiting 2x the time as you saw due to a missing return edge on "no events" handled.

closed 2021-02-27 by msmeissn