gphoto/libgphoto2 #580: GP_ERROR_FILE_EXISTS after get info

dungcules, 2020-11-12:
**Describe the bug**

I am trying to get info out of an image inside a folder, then remove that folder. The command I use is:

1. gp_filesystem_get_info
2. gp_filesystem_remove_dir

These two command is placed next to each other. I feed the folder and filename manually, so they should have the correct input. To activate the first command, I have to create a "CameraFileInfo" as following:

CameraFileInfo* file_info = new CameraFileInfo()

If the second command is place first, I got the error -107 (GP_ERROR_DIRECTORY_NOT_FOUND). This is an expected behavior when the folder was removed before getting the information of the image.

If I put the first command before the second one, the return error is -103 (GP_ERROR_FILE_EXISTS ). I wonder if the getting info command creates any file or not. I also tried to free "CameraFileInfo" but the error is still the same. 

Please let me know if I did something wrong or if you need more information. 

**libgphoto2 and gphoto2 version**
gphoto2         2.5.20         
libgphoto2      2.5.22

dungcules, 2020-11-12:
I repeat my process using CLI and it giving me the same error -103. Therefore, I tried the debug command as following: 

gphoto2 --port=usb:001,004 --show-info='/store_00010001/DCIM/100_GEOC/GEOC0003.JPG' --rmdir='/store_00010001/DCIM/100_GEOC' --debug --debug-logfile=/DATA/show_remove.txt

[show_remove.txt](https://github.com/gphoto/libgphoto2/files/5532281/show_remove.txt)

The result is attached. The log said that there is still files in the folder that I want to delete (line 27884). This means I have to remove all the files inside this folder and then remove the folder. Then, I tried these commands:

1. gp_filesystem_get_info
2. gp_filesystem_delete_all
3. gp_filesystem_remove_dir

These combination worked and it removed my folder. However, I believe that removing directory command (without get info) take much less time to remove my folder than using this method. If you have any suggestion to improve my process, I really appreciate.

Regardless, 

Dung

dungcules, 2020-11-17:
I found the solution on this problem. After using gp_filesystem_get_info, we should reset the filesystem before remove the directory by using gp_filesytem_reset.

closed 2020-11-17 by dungcules

msmeissn, 2020-11-18:
i think one issue is also that you first need to remove all files from the folder before removing it.
10.410027 gp_context_error            (0): There are still files in folder '/store_00010001/DCIM/100_GEOC' that you are trying to remove.

as you noticed.

Big question if you want to remove the folder with files on the camera too? Or just in memory?

dungcules, 2020-11-18:
Hi Marcus,

My goal is to remove the data on camera SD Card using timestamp on an image without unplugging the memory stick . I know the manual process is cleaning up all files before removing that folder but I saw libgphoto2 support gp_filesystem_remove_dir and I tried it.

At very beginning, I only issue gp_filesystem_remove_dir to remove a specific folder and it works fine even when I dont remove files inside that folder. In addition, this command takes much shorter time comparing with the manual process. However, when I start using gp_filesystem_get_info to get the timestamp out of an image, the error -103 happened. I cannot figure out where the problem is until I read the log from debug process.

 Luckily, I used the API documentation of libgphoto2 and it gave the magical command gp_filesytem_reset which solve my problem with long delete process. 

For your question, I believe that I want to remove the folder with file in memory card because my camera does not have internal storage.

Thank you for your responding,

Dung