gphoto/libgphoto2 #581: Regression: File size truncated to 4 GiB on Samsung Galaxy S7 via MTP

nh2, 2020-11-14:
Commit https://github.com/gphoto/libgphoto2/commit/0cc43cc7b5f509dea38a463814a4eb2467d1ac36 `imported from libmtp` (between 2.5.22 and 2.5.23) introduced a regression that makes files > 4 GiB show up as 4 GiB.

`gphoto2 --get-file` consequently silently corrupts the file when copying, truncating it to 4 GiB. Note that before the regression, `--get-file` was also broken, truncating the large file to `file_size % (2**32)` Bytes:

```
$ du --apparent-size --bytes bigfile.jpg
1172299776	bigfile.jpg
```

when the file was shown as `9533432 KB` by `--list-files` before the regresssion:

```
% gphoto2 --list-files | grep bigfile
#186   bigfile.jpg                rd 9533432 KB image/jpeg
```
```
>>> 9533432 * 1024 % (2**32)
1172299776
```

This happens at least on our Samsung Galaxy S7, possibly other other Galaxy devices.

Previous related commits from `libmtp`:

* https://github.com/libmtp/libmtp/commit/9f7867e0ca1d0928a5cc2aad1253a9aad6290c05#diff-ce8b79decb68f2053f576057da29ce731530ddbf38a74945e1c758b9c6ab5b3aR384
* https://github.com/libmtp/libmtp/commit/7c5ccc1c330e06f8c6a9f3c91185820887d3ee93
* https://github.com/libmtp/libmtp/commit/a81377bf990ad201ccd623ae76d18b23494d6075

So `libmtp` might have this issue as well.

CC @msmeissn.

---

I found this as part of my investigation in https://github.com/whoozle/android-file-transfer-linux/issues/250.

nh2, 2020-11-14:
`libmtp version 1.1.17`'s `mtp-files` shows the file size correctly:

```
File ID: 6401
   Filename: bigfile.jpg
   File size 9762234368 (0x0000000245DFE000) bytes
```

and `mtp-getfile` transfers it correctly.

So this issue seems to exist only in `libgphoto2`, not `libmtp`.

msmeissn, 2020-11-24:
list files might be fixed, i only tested it with a Nikon camera though.

Next step is file download larger than 4GB.

msmeissn, 2021-02-27:
reading files larger than 4GB with getobject should work, tested on Nikon too.

msmeissn, 2021-10-03:
i think fixed, although i have not personally tested it.

closed 2021-10-03 by msmeissn