gphoto/libgphoto2 #582: ptp2 driver does not support file downloads over 4GB

msmeissn, 2020-11-17:
**Describe the bug**

Currently we do not support file download over 4GB in the ptp driver.

Largely because these are special handled by ptp, and Marcus did not yet find a device for testing. :(

**Name the camera**

e.g. Ricoh Theta supposedly allow such files, also Android phones , also Fuji XT?

msmeissn, 2020-11-28:
nikon dslr ... now can do > 4gb files.

nh2, 2020-11-28:
@msmeissn Should I already try and test this with my Ricoh cams?

For clarification, do I understand correctly that https://github.com/gphoto/libgphoto2/commit/c94b36cd6e0389793f78020539097a21be5368cc# should result in _listing_ the file sizes correctly also on my Ricoh, but _transferring_ shouldn't work yet and only works on Nikon as part of https://github.com/gphoto/libgphoto2/commit/19449d03a5912e6a44b53e433fbe7f311506a4e7?

Thanks!

msmeissn, 2020-11-28:
you can try if the filresizes are ok.  downloading will not yet work.

But for the Ricoh .. can you do
gpoto2 --summary --debug --debug-logfile=summary-debug.txt > summary.txt
and send me both .txt files?  (for both cameras if you have them)
So I can see which 64bit download method  they use

nh2, 2020-11-28:
[`ricoh-theta-v-summary.txt`](https://github.com/gphoto/libgphoto2/files/5610623/ricoh-theta-v-summary.txt)
[`ricoh-theta-z1-summary.txt`](https://github.com/gphoto/libgphoto2/files/5610624/ricoh-theta-z1-summary.txt)

msmeissn, 2020-11-28:
can you also add the summary-debug.txt ?

nh2, 2020-11-28:
[`ricoh-theta-v-summary-debug.txt`](https://github.com/gphoto/libgphoto2/files/5610653/ricoh-theta-v-summary-debug.txt)
[`ricoh-theta-z1-summary-debug.txt`](https://github.com/gphoto/libgphoto2/files/5610654/ricoh-theta-z1-summary-debug.txt)

nh2, 2020-11-28:
> you can try if the filresizes are ok.

They aren't' OK yet for me:

```diff
-#18    R0012112.MP4               rd 4194304 KB 3840x1920 video/mp4
+#18    R0012112.MP4               rd 4194304 KB 3840x1920 video/mp4
```

Should the results depend on whether I have https://github.com/libmtp/libmtp/pull/68/files in my udev rules (currently those are in for my machine)?

msmeissn, 2020-11-28:
no, its entirely gphoto2 side... can you attach debug output of list files

gphoto2 -L --debug --debug-logfile=list-files.txt

and attach list-files.txt

nh2, 2020-11-28:
[ricoh-theta-v-list-files.txt](https://github.com/gphoto/libgphoto2/files/5610849/ricoh-theta-v-list-files.txt)
[ricoh-theta-z1-list-files.txt](https://github.com/gphoto/libgphoto2/files/5610850/ricoh-theta-z1-list-files.txt)

The `v` currently doesn't have a large file on it, the `z1` does.

msmeissn, 2020-11-28:
thanks!
0.000133 main                        (2): libgphoto2 2.5.26

one thing, you need to update libgphoto2 to current GIT ... the fix is not in 2.5.26 release

msmeissn, 2020-11-28:
(version should report as 2.5.26.1)

nh2, 2020-11-29:
> one thing, you need to update libgphoto2 to current GIT

Ah, I do have multiple versions available as compile outputs.

Here are the `gphoto2 -L --debug --debug-logfile=list-files.txt` outputs with commit 37f4841abbddd7cc1e0a19880f639b4a67297ca7:

[ricoh-theta-v-list-files.txt](https://github.com/gphoto/libgphoto2/files/5612142/ricoh-theta-v-list-files.txt)
[ricoh-theta-z1-list-files.txt](https://github.com/gphoto/libgphoto2/files/5612143/ricoh-theta-z1-list-files.txt)

msmeissn, 2020-11-29:
i added another smnall commit that might fix the fileisze if you want to try.

msmeissn, 2020-11-29:
Some more hardware tested:
Canon EOS: the 100D and 750D do not want to record movies larger thn 4GB , they split them ... even on a exfat formated SD Card.
Android Samsung Galaxy S9: also exfat formatted micro sdcard ... standard camera app splits movies at 4GB border.
GoPro Hero 7 Silver: exfat formatted microsd: also splits at 4GB
Nikon KeyMission 170: also splits at 4GB on exfat formatted micro sd card.

nh2, 2020-11-29:
> i added another smnall commit that might fix the fileisze if you want to try.

That seems to list correctly on commit 93f2b140fc8f46a05a591c2b11241f54bb7ae4a2:

```
$ /nix/store/abfrf3m44l47c0sh4vzcks0fb1sxp4zw-gphoto2-2.5.23/bin/gphoto2 --list-files
#18    R0012112.MP4               rd 9925760 KB 3840x1920 video/mp4
```

So I guess now only the download part is missing?

nh2, 2020-11-29:
> Android Samsung Galaxy S9: also exfat formatted micro sdcard

To test on my Galaxy S7, I just used `adb shell` and created a big file with `dd`, in case that is helpful.

msmeissn, 2020-11-29:
sounds like a workable idea, will try :)

msmeissn, 2020-12-19:
actually the ptp_getobject code in libgphoto2 was already capable of downloading > 4GB files... 
(so only the size was reported incorrectly before)

I reenable that code...

closed 2022-01-04 by msmeissn