gphoto/libgphoto2 #585: movie capture not working on ubuntu 20.04 - Canon 60D

milixov, 2020-11-19:
**Describe the bug**

movie capture is not working on Ubuntu 20.04, nothing readable from /dev/video0 or /dev/vdieo1. Also Entanle app is not working on Ubuntu 20.04

**Name the camera**
`Canon EOS 60D`

**libgphoto2 and gphoto2 version**
`gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.24         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking`

**To Reproduce**
each movie capture command

When reproducing with the commandline tool, attach debugout using  --debug --debug-logfile=debug.log   if considered useful.

msmeissn, 2020-11-24:
- entangle should work ... what kind of error does it report?

- for movie capture, you need to setup a ffmpeg pipe and v4l2loopback, see http://gphoto.org/doc/remote/ "Using as webcam"

milixov, 2020-12-02:
Entangle just take a picture without preview mode. and the movie record button is disabled

milixov, 2020-12-13:
Hi, I reinstall `v4l2loopback` and problem is solved. but figured out video is very laggy because of low fps:
`frame= 2195 **fps=8.9** q=-0.0 size=N/A time=00:01:27.80 bitrate=N/A speed=0.356x`

msmeissn, 2020-12-20:
so one issue is that the framerate is a bit low ... (20 - 25 frames might be reachable) ... might be an issue of libgphoto2 and tioming.

ffmpeg buffers some frames, this brigns delay :/

msmeissn, 2021-06-19:
not much we can do regarding delays

closed 2021-06-19 by msmeissn