gphoto/libgphoto2 #588: Nikon Z50 bulb mode fails

wintermute101, 2020-11-26:
When issuing command:

gphoto2 --set-config bulb=1 --wait-event=1s --set-config bulb=0

*** Error ***              
Failed to set new configuration value 0 for configuration entry bulb.
*** Error (-1: 'Unspecified error') *** 

Camera version: "Nikon Z50"

All verstions up to latest from github.
libgphoto2      2.5.26.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

The problem is Nikon camera returns Bulb Release Busy to DeviceReady command:
1.149419 ptp_usb_getresp             (2): Reading PTP_OC 0x90c8 (DeviceReady) response...
...
1.152013 ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x90c8 receiving resp failed: Bulb Release Busy (0xa200)
1.152019 _put_Nikon_Bulb [config.c:7163](0): 'nikon_wait_busy(params, 100, 5000)' failed: Bulb Release Busy (0xa200)

Which is considered an error but according to Nikon specification:

"The host issues the DeviceReady command and confirms that the response code is
Bulb_Release_Busy."

This is expected behaviour.

With this ugly fix to latest git version I was able to fix issue and make it work:

```
diff --git a/camlibs/ptp2/config.c b/camlibs/ptp2/config.c
index 2e5bf2cd9..b0982d8fc 100644
--- a/camlibs/ptp2/config.c
+++ b/camlibs/ptp2/config.c
@@ -643,6 +643,10 @@ nikon_wait_busy(PTPParams *params, int waitms, int timeout) {
                tries=50;
        do {
                res = ptp_nikon_device_ready(params);
+               if (res == PTP_RC_NIKON_Bulb_Release_Busy) {
+                       GP_LOG_D("nikon ugly fix for bulb");
+                       return PTP_RC_OK;
+               }
                if (res != PTP_RC_DeviceBusy)
                        return res;
                if (waitms) usleep(waitms*1000)/*wait a bit*/;
diff --git a/camlibs/ptp2/usb.c b/camlibs/ptp2/usb.c
index 0d15d9743..9a66e6453 100644
--- a/camlibs/ptp2/usb.c
+++ b/camlibs/ptp2/usb.c
@@ -511,6 +511,10 @@ ptp_usb_getresp (PTPParams* params, PTPContainer* resp)
        if (dtoh16(usbresp.code)!=resp->Code) {
                ret = dtoh16(usbresp.code);
        }
+       if (ret == 0xa200) {
+               GP_LOG_D("nikon ugly fix for bulb");
+               ret = PTP_RC_OK;
+       }
        if (ret!=PTP_RC_OK) {
                GP_LOG_E ("PTP_OC 0x%04x receiving resp failed: %s (0x%04x)", resp->Code, ptp_strerror(ret, params->deviceinfo.VendorExtensionID), ret);
                return ret;
```

msmeissn, 2020-11-29:
hmm, i added some code to nikon_wait_busy, that continues to wait when it sees that error code instead.

in theory i could also chabnge it to return OK from wait_busy when it sees that, whatever works better

wintermute101, 2020-11-30:
I can confirm it did the trick. (git commit 93f2b140fc8f46a05a591c2b11241f54bb7ae4a2)

```
2.397992 ptp_usb_getresp             (2): Reading PTP_OC 0x920c (TerminateCapture) response...
2.397995 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.400075 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-1e 00 00 00              ....... ....

2.400080 ptp_usb_sendreq             (2): Sending PTP_OC 0x90c8 (DeviceReady) request...
2.400082 gp_port_write               (3): Writing 12 = 0xc bytes to port...
2.400859 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 c8 90-1f 00 00 00              ............

2.400862 ptp_usb_getresp             (2): Reading PTP_OC 0x90c8 (DeviceReady) response...
2.400865 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.402030 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 00 a2-1f 00 00 00              ............

2.402038 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x90c8 receiving resp failed: Bulb Release Busy (0xa200)
2.502428 ptp_usb_sendreq             (2): Sending PTP_OC 0x90c8 (DeviceReady) request...
2.502442 gp_port_write               (3): Writing 12 = 0xc bytes to port...
2.502488 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 c8 90-20 00 00 00              ........ ...

2.502493 ptp_usb_getresp             (2): Reading PTP_OC 0x90c8 (DeviceReady) response...
2.502496 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
2.505628 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-20 00 00 00              .......  ...

2.506093 gp_camera_free              (2): Freeing camera...
2.506098 gp_camera_exit              (2): Exiting camera ('Nikon Z50')...
2.506103 ptp_usb_sendreq             (2): Sending PTP_OC 0x9202 (EndLiveView) request...
2.506105 gp_port_write               (3): Writing 12 = 0xc bytes to port...
2.506413 gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
0000  0c 00 00 00 01 00 02 92-21 00 00 00              ........!...
```

msmeissn, 2020-12-01:
I think this is fixed now?= :)

closed 2020-12-01 by msmeissn