gphoto/libgphoto2 #589: Nikon Z6 II video recording error

tomsch1, 2020-11-27:
**Describe the bug**

Tried to record a video on a Nikon Z6 II on the camera's storage card. The camera can be set to ApplicationMode/0x90c2 (Debug log line 1104) but when the video recording should start I get the error message "Movie recording cannot start: Camera is not in application mode." (debug log line 1230). And therefore the camera fails to enter movie recording mode.

**Name the camera**
Nikon Z6_2 usb:001,051

**libgphoto2 and gphoto2 version**
gphoto2 2.5.23.1
libgphoto2 2.5.26.1

**To Reproduce**
`gphoto2 --set-config controlmode=1 --set-config d1a6=1 --set-config movie=1 --wait-event=1s --set-config movie=0`

[z6_2_videoDebug.txt](https://github.com/gphoto/libgphoto2/files/5607017/z6_2_videoDebug.txt)

msmeissn, 2020-11-29:
the camera does not seem to report 0xd1f0 (application mode) as available ... 
can you try set it manually:
 gphoto2 --set-config controlmode=1 --set-config d1a6=1 --set-config d1f0=1 --set-config movie=1 --wait-event=1s --set-config movie=0

tomsch1, 2020-12-01:
It turns out that there is no ApplicationMode or at least not at address _d1a6_. A newer Nikon SDK for th Z5 II is not available yet. So it is not possible to manually enable it and the error message is correct.

msmeissn, 2020-12-01:
yes, we will likely need to wait for the Z6 II SDK here.

msmeissn, 2020-12-19:
it is now available... they turned the application mode into a opcode...

I added support for that to current git, please test.

msmeissn, 2020-12-27:
should be fixed.
i did some further changes to fix "ending movie".

please reopen if not working

closed 2020-12-27 by msmeissn