gphoto/libgphoto2 #590: Nikon Z50 video recording error

tomsch1, 2020-11-27:
**Describe the bug**

Tried to record a video on a Nikon Z50 on the camera's storage card. The camera can be set to ApplicationMode/0x90c2 (Debug log line 792) but when the video recording should start I get the error message "PTP_OC 0x1016 receiving resp failed: PTP Access Denied" (debug log line 884). And therefore the camera fails to enter movie recording mode.

The solution will be probably the same as with #589, but created a new Bug report because of the diffrent error message.

**Name the camera**
Nikon Z50                      usb:001,005   

**libgphoto2 and gphoto2 version**
gphoto2         2.5.26.1
libgphoto2      2.5.26.1
libgphoto2_port 0.12.0  

**To Reproduce**
`gphoto2 --set-config controlmode=1 --set-config d1a6=1 --set-config movie=1 --wait-event=1s --set-config movie=0`

[z50_videoRecord.log](https://github.com/gphoto/libgphoto2/files/5607343/z50_videoRecord.log)

msmeissn, 2020-11-29:
log: Setting the 0xd1f0 (ApplicationMode) results with access denied.

Can you perhaps try leaving the set-config controlmode and d1a6 out if that helps?
Or can you perhaps inject some --wait-event=1s inbetween the --set-config  calls to see if we perhaps just need to drain events?

Cannot repoduce on my Z6, there all combinations seem to result in movie capture.

tomsch1, 2020-12-01:
Tried both command chains (with waiting for event and skip controlmode and d1a6).
Same error is returned "PTP access denied".

msmeissn, 2020-12-27:
can you try this also with a --get-config 0xd0a4     

can you try this with controlmode=0   (the SDK doc is hard to read, but it might be that in remote mode we cannot change the application mode)

gphoto2 --set-config controlmode=0  --set-config movie=1 --wait-event=1s --set-config movie=0

tomsch1, 2021-01-05:
This is the output of --get-config d0a4 and it does not matter if controlmode=1 or 0.

`Label: MovRecProhibitCondition
Readonly: 1
Type: TEXT
Current: 24576
END`

The access to enter movie record mode is denied with controlmode =0 and 1.

When I try to set 0xd1a6 too with controlmode=0, I get the error returned that 0xd1a6 is read only.

wintermute101, 2021-02-03:
I tried above command to reproduce but video is recorded. I get a lot of messages in the process but video is in sd card.

```
gphoto2 2.5.23

Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.25         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

```

```
*** Error ***              
Invalid Status

*** Error ***              
Failed to set new configuration value 1 for configuration entry movie.
Waiting for 1 seconds for events from camera. Press Ctrl-C to abort.
UNKNOWN PTP Event 400c, Param1 00010001
UNKNOWN PTP Event c10a, Param1 00000000
UNKNOWN PTP Property 5003 changed
UNKNOWN PTP Property 5004 changed
UNKNOWN PTP Property 500d changed
UNKNOWN PTP Property 500e changed
UNKNOWN PTP Property 501c changed
UNKNOWN PTP Property d030 changed
UNKNOWN PTP Property d0a0 changed
UNKNOWN PTP Property d0a4 changed
UNKNOWN PTP Property d0b5 changed
UNKNOWN PTP Property d0bc changed
UNKNOWN PTP Property d0c1 changed
UNKNOWN PTP Property d0c2 changed
UNKNOWN PTP Property d0f8 changed
UNKNOWN PTP Property d100 changed
UNKNOWN PTP Property d109 changed
UNKNOWN PTP Property d126 changed
UNKNOWN PTP Property d17a changed
UNKNOWN PTP Property d1a2 changed
UNKNOWN PTP Property d1a9 changed
UNKNOWN PTP Property d1ab changed
UNKNOWN PTP Property d1af changed
UNKNOWN PTP Property d1b3 changed
UNKNOWN PTP Property d1b4 changed
UNKNOWN PTP Property d1b7 changed
UNKNOWN PTP Property d1bb changed
UNKNOWN PTP Property d1f1 changed
UNKNOWN PTP Property d20d changed
UNKNOWN PTP Property 500c changed
UNKNOWN PTP Property 5013 changed

```

wintermute101, 2021-02-03:
On current git version (70b9fe4c132df2f37fcf05929d27ed820003312e) it does not work.

gphoto2 2.5.23
```
Copyright (c) 2000-2019 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.26.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

msmeissn, 2021-02-03:
i did changes to support actually movie recording better on the Z series. BUt I would say it was not working before either?

f you are comfortable with git bisect, you might try bisecting?

msmeissn, 2021-02-03:
I think movierrecprohibit  24576 is 0x6000, means these 2 prohibit conditions trigger:

if (value.u32 & (1<<14)) { gp_context_error (context, _("Movie recording cannot start: Camera is not in application mode.")); return GP_ERROR; }
if (value.u32 & (1<<13)) { gp_context_error (context, _("Movie recording cannot start: Set liveview selector is enabled.")); return GP_ERROR; }

msmeissn, 2021-02-03:
i added decoding of movierecprohibit condition too now , not just reporting as error

wintermute101, 2021-02-03:
Hmm I reverted in git to version 2.5.25 but no luck still does not work.
Then ok It worked on my system installed version (but in mean time installed git version to my system) so I tried going back to my system version.
Reinstalled libgphoto2-6 (mint version) no luck still does not wiork.
Ok this is really weird.
Deleted all libgphoto2 manually and reinstalled.
Missing libgphoto2_port so installed it.
And viola it works...

log from working state
[debug.log.gz](https://github.com/gphoto/libgphoto2/files/5919919/debug.log.gz)

how can I bisect that? 

When I test git build version I test like this:
LD_LIBRARY_PATH=./build/lib/ gphoto2

msmeissn, 2021-02-03:
I am trying to spot differences ..  there is a slightly difference in event handling.

(the camera is in the same setup and state right?)

can you throw in --wait-event=1s before all other ops?

wintermute101, 2021-02-03:
`gphoto2 --wait-event=1s --set-config controlmode=1 --set-config d1a6=1 --set-config movie=1 --wait-event=1s --set-config movie=0 --debug --debug-logfile=debug.log`

[debug.log.gz](https://github.com/gphoto/libgphoto2/files/5920260/debug.log.gz)

wintermute101, 2021-02-03:
Yes camera should be in same state, I tried testing with camera switch on or switched to photo mode but if it works it works.

msmeissn, 2021-02-04:
can you test the same as above on a 2.5.26.1 libgphoto2 ?

wintermute101, 2021-02-04:
Sure tested on ddfc8f8f549599ddfbdb8e33db93be6f2786fb64.

[debug.log.gz](https://github.com/gphoto/libgphoto2/files/5927253/debug.log.gz)

with:

`LD_LIBRARY_PATH=./build/lib/ gphoto2 --wait-event=1s --set-config controlmode=1 --set-config d1a6=1 --set-config movie=1 --wait-event=1s --set-config movie=0 --debug --debug-logfile=debug.log`

and:

```
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.26.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

msmeissn, 2021-02-06:
ok, major diff i now see is
2.5.26.1 is vbuilt with libusb0 not with libusb1.

can you build it with libusb1 development headers and libs instead of libusb 0 ? 

(I think this causes an USB PTP interrupt not to be delivered , and in turn results in this error.)

wintermute101, 2021-02-06:
Ok, I did not notice this. Compiled with usb1 and it behaves differently but still no movie recorded on sd.

```
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.26.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

[debug.log.gz](https://github.com/gphoto/libgphoto2/files/5936932/debug.log.gz)

wintermute101, 2021-02-06:
Oh sorry last test was with zoom lens not extended preventing camera to record.
[debug.log.gz](https://github.com/gphoto/libgphoto2/files/5936952/debug.log.gz)

Now movie is recorded :)

msmeissn, 2021-02-06:
Very good to hear! (you can remove the additional wait-events, as the issue was the libusb0 usage ;)

closed 2021-02-06 by msmeissn