gphoto/libgphoto2 #591: Zoom control reported as "RADIO" instead of "RANGE"

Ocirne94, 2020-11-30:
**Describe the bug**
Zoom setting as obtained from `gphoto2 --list-all-config` is reported as
```
/main/capturesettings/zoom
Label: Zoom
Readonly: 0
Type: RADIO
Current: 0
Choice: 0 101
END
```

instead the type should be a range (from 0 to 101). Setting zoom with (for example) `gphoto2 --set-config="/main/capturesettings/zoom=50"`works as expected (zoom in to half-way).

**Name the camera**
Canon Powershot G1 X Mark III (built-in zoom lens)

**libgphoto2 and gphoto2 version**
gphoto2         2.5.23
libgphoto2      2.5.25
libgphoto2_port 0.12.0

msmeissn, 2020-11-30:
you can shorten above with 
gphoto2 --set-config zoom=50  
btw.

Can you give me debug output of gphoto2 --list-all-config --debug --debug-logfile=list-config-debug.txt 
(attach list-config-debug.txt here)

Ocirne94, 2020-11-30:
Sure, here it is (and thanks for the tip)
[list-config-debug.txt](https://github.com/gphoto/libgphoto2/files/5615179/list-config-debug.txt)