gphoto/libgphoto2 #593: Stupid user question: how to start/stop recording video (Canon 60D)

arikrupnik, 2020-12-02:
This feels like such a simple thing, but I can't find the answer.

I want a command line that would be the equivalent of pressing the "record video" button. I want to record to the SD card. I don't need live preview, and I don't need to download the captured file to the host immediately. The example on http://www.gphoto.org/doc/remote/ is very helpful, but it captures a specific-duration video and does not save it to the SD card.

Thank you, and apologies again for asking such an obvious question.

msmeissn, 2020-12-02:
from http://gphoto.org/doc/remote 

Newer Canon EOS DSLRs (around 7D and later) : 
Switch the camera to movie record mode on the camera.
Then run 
   gphoto2 --set-config viewfinder=1 --set-config movierecordtarget=Card --wait-event=10s --set-config movierecordtarget=None --wait-event-and-download=2s

 (before 2.5.6, replace viewfinder by eosviewfinder). 
If this does not work, check and switch the capturetarget to SD Card (gphoto2 --set-config capturetarget=1) 

can you try that?

arikrupnik, 2020-12-02:
Thank you for the quick response. This command line is the example I was referring to. It works exactly as you say. My main question is how to specify unlimited duration (i.e. until the user stops recording or the camera reaches a limit or runs out of space on SD card).

msmeissn, 2020-12-02:
You can just specify a very very long --wait-event
- a user can press ctrl-c to stop the waitevent  
- with gphoto2 2.5.26 you can end the --wait-event  by sending a "SIGUSR2" signal to the gphoto2 process.
- if the memory runs full, it will error out I think ... have to test that.

arikrupnik, 2020-12-02:
Thank you @msmeissn. I guess my basic misconception was that `gphoto2` can exit and the camera would still be rolling. I guess that's impossible and I need to keep the process alive and the connection going. I was going to ask how to specify a very long --wait-event, but the man page says

> If no argument is given, the wait time is 1 million events (basically forever).

Thank you again for your help.

closed 2020-12-02 by arikrupnik

r-a-i, 2023-05-19:
I'm trying a Canon M50 and I can:
`gphoto2 --set-config movierecordtarget=0 # STARTS recording a movie and exits`
`gphoto2 --set-config movierecordtarget=1 # STOPS recording a movie and exits`

BTW:  `movierecordtarget=Card` or `movierecordtarget=None` throw back errors.

r-a-i, 2023-06-03:
Dear @msmeissn , `gphoto2 --set-config movierecordtarget=0` works on some systems but not others. 

I have two linux systems, a Dell Tower and a Kamrui Mini PC, both running Ubuntu 20.04.6, gphoto2 2.5.23 and libgphoto2 2.5.25, but gPhoto2 only works with the Dell Tower.  This is consistent for a Canon M50 and R50.  

`gphoto2 --set-config movierecordtarget=0` on Dell Tower (Lightning) starts a movie capture and results in a movie file on the SDCard.
`gphoto2 --set-config movierecordtarget=0` on MiniPC (Kami) does NOT starts a movie capture, and instead returns The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).

Another Raspberry Pi system running Ubuntu 22.04 also behaves like MiniPC. The MiniPC with Ubuntu 22.04 also fails. 

Below is the detailed respose from a debug log, and attached are both log files (without timestamps for easy diff), and the diff.
[my-logfile_kami_20.04_notimes.txt](https://github.com/gphoto/libgphoto2/files/11643590/my-logfile_kami_20.04_notimes.txt)
[my-logfile_lightning_0_notimes.txt](https://github.com/gphoto/libgphoto2/files/11643592/my-logfile_lightning_0_notimes.txt)
[diff_file.txt](https://github.com/gphoto/libgphoto2/files/11643640/diff_file.txt)

Please help.

The following block is from the debug log file when the system reports failure. 
```
  _set_config                 (2): Setting property 'Recording Destination' / 0xd1b8
  _put_Genericu16Table        (2): FOUND right value for Card in the enumeration at val 4
  ptp_usb_sendreq             (2): Sending PTP_OC 0x9110 (EOS_SetDevicePropValueEx) request...
  gp_port_write               (3): Writing 12 = 0xc bytes to port...
  gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
  0000  0c 00 00 00 01 00 10 91-25 00 00 00              ........%...    
  
  ptp_usb_senddata            (2): Sending PTP_OC 0x9110 (EOS_SetDevicePropValueEx) data...
  gp_port_write               (3): Writing 24 = 0x18 bytes to port...
  gp_port_write               (3): Wrote   24 = 0x18 bytes to port: (hexdump of 24 bytes)
  0000  18 00 00 00 02 00 10 91-25 00 00 00 0c 00 00 00  ........%.......
  0010  b8 d1 00 00 04 00 00 00-                         ........        
  
  ptp_usb_getresp             (2): Reading PTP_OC 0x9110 (EOS_SetDevicePropValueEx) response...
  gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
  gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
  0000  0c 00 00 00 03 00 19 20-25 00 00 00              ....... %...    
  
  ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
  _set_config [config.c:9680] (0): 'ptp_canon_eos_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Busy (0x2019)
  gp_context_error            (0): The property 'Recording Destination' / 0xd1b8 was not set (0x2019: PTP Device Busy).
  gp_camera_set_single_config [gphoto2-camera.c:1144](0): 'camera->functions->set_single_config ( camera, name, widget, context)' failed: -110
  gp_context_error            (0): Failed to set new configuration value 0 for configuration entry movierecordtarget.
  gp_camera_free              (2): Freeing camera...
  gp_camera_exit              (2): Exiting camera ('Canon EOS M50')...
  ptp_usb_sendreq             (2): Sending PTP_OC 0x9116 (EOS_GetEvent) request...
  gp_port_write               (3): Writing 12 = 0xc bytes to port...
  gp_port_write               (3): Wrote   12 = 0xc bytes to port: (hexdump of 12 bytes)
  0000  0c 00 00 00 01 00 16 91-26 00 00 00              ........&...    
```

r-a-i, 2023-06-05:
I found a workaround to start/stop recording on a M50 by following instructions from another issue at https://github.com/gphoto/libgphoto2/issues/135

The solution was to run:
gphoto2 --set-config capturetarget=1 --set-config movierecordtarget=0 # Start recording
gphoto2 --set-config movierecordtarget=1 # Stop recording. 

The 'stop recording' command still throws a PTP Device Busy (0x2019) error, but does stop the recording.

msmeissn, 2023-06-18:
yes, one of the idff was that the capturetarget was set already. 
hm, weird. usually means we either need to wait for something to happen / poll the camera