gphoto/libgphoto2 #597: Support capture from RX100M3

Esokrates, 2020-12-05:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Sony RX100 M3

**USB IDs**
Bus 001 Device 005: ID 054c:07ca Sony Corp. DSC-RX100M3

**camera summary output**
```
*** Error ***              
Could not detect any camera
*** Error (-105: 'Unknown model') ***       
```

Here http://www.gphoto.org/proj/libgphoto2/support.php M4-7 are all listed as supporting capture. Can we do something about the M3 too? I am willing to try things out and supply all the info you may need.

msmeissn, 2020-12-05:
In which USB access mode is the Sony ? 

You need to switch it to PC Control mode (on camera).

Esokrates, 2020-12-05:
Sadly there is not such option, only mass storage and MTP. Ok I guess that means it can't be supported?

msmeissn, 2020-12-05:
can you check in MTP mode?

msmeissn, 2021-01-03:
ping?

duvrai, 2021-02-02:
I put mine in MTP mode. I did: `env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt --summary`

Here is the logfile: [my-logfile.txt](https://github.com/gphoto/libgphoto2/files/5913643/my-logfile.txt)
and stdout produced
```
Camera summary:                                                                
Manufacturer: Sony Corporation
Model: DSC-RX100M3
  Version: 2.0
  Serial Number: 00000000000000008081615002890539
Vendor Extension ID: 0x11 (1.0)

Capture Formats: 
Display Formats: Association/Directory, MPEG, JPEG, Unknown(b982), Unknown(b301), Undefined Type, ARW, Unknown(b310)

Device Capabilities:
	File Download, No File Deletion, No File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_00010001:
	StorageDescription: Storage Media
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Generic Hierarchical
	Access Capability: Read-Only
	Maximum Capability: 0 (0 MB)
	Free Space (Bytes): 0 (0 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Enumeration [0,1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40,41,42,43,44,45,46,47,48,49,50,51,52,53,54,55,56,57,58,59,60,100] value: 100% (100)
Property 0xd402:(read only) (type=0xffff) 'DSC-RX100M3'
Property 0xd406:(readwrite) (type=0xffff) '(null)'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd303:(read only) (type=0x2) 1
```

msmeissn, 2021-02-15:
I looked over some google hits, and no one seem to have made remote control over USB possible for the RX100M3. :/

So it seems a camera side limitation.

also not listed here: https://support.d-imaging.sony.co.jp/app/webcam/en/download/

sorry, not much we can do

closed 2021-02-15 by msmeissn