gphoto/libgphoto2 #598: config remotemode and uilock on Canon EOS M6 II issue

marczh, 2020-12-09:
**Cannot change config-entry for uilock and remotemode**

**Canon EOS M6 II**

gphoto2 2.5.26
libghoto2 2.5.26

- trying to use EOS M6 Mark II with --wait-event-and-download, but camera is locked and shows no preview
- gphoto2 --get-config uilock # current : 2
- gphoto2 --get-config remotemode # current: 0
- gphoto2 --set-config-value uilock=0 # command digests but does not change the config, still current: 2
- gphoto2 --set-config-value remotemode=1 # command digests but does not change the config, still current: 0

The camera's display shows battery and Computer Icon when connected. It turns black when gphoto2-command is performed. Stays black afterwards. No PTPCamera Process running.

msmeissn, 2020-12-29:
uilock is something we cannot query back from the camera. :/ 
remotemode only on some models, but no the EOS M6 Mark 2.

I spent half an hour with my M10 again to try to have it accept any form of key input, so far no luck :(
I will also see if I can get windows to do anything

marczh, 2021-01-03:
Issue resolved with commit 8d9d9efaa719542abfcc8bc3637efd1f95b8e0f5
Thanks, Marcus.

closed 2021-01-03 by marczh