gphoto/libgphoto2 #599: gphoto2 get ptpip camera's model

huangmingyou, 2020-12-10:
when a camera connect with usb, use gphoto2 -a can get the camera's model. like 
gphoto2 -a
Abilities for camera             : Canon EOS 100D

,but when connect with ptpip ,
the gphoto2 -a --port ptpip:192.168.1.1 only return this message

Abilities for camera             : PTP/IP Camera

;
I write a demo.c init the ptpip camera , then want get device info from struct  camera->pl->params->deviceinfo.Model ; 

but will get error like this

t1.c:38:26: error: dereferencing pointer to incomplete type ‘CameraPrivateLibrary {aka struct _CameraPrivateLibrary}’

so , how can I get a ptpip connected camera is model ?

msmeissn, 2020-12-10:
Currently either via summary output, or via the config api  "cameramodel" and "manufacturer" values.

Reason is that the camera abilities is a static functionality, which does not a camera to be present

huangmingyou, 2020-12-10:
thanks , via the config api is a good idea.

msmeissn, 2020-12-27:
seems solved

closed 2020-12-27 by msmeissn