gphoto/libgphoto2 #601: Disconnect camera blocking until timeout

dungcules, 2020-12-12:
Hello Marcus,

I need your help with the problem that I have on my USB cameras. The cameras are connected to a hub before plugging my station. The hub can feed the power to cameras. I make a thread to capture image from these cameras. 

The problem I have when I disconnect the hub and reconnect it, I found that camera thread prevented the OS to recognize the connected camera. The OS saw the connected device but no camera port shows up when I use lsusb. I think that there is still data transfer on the old port cause this problem. I try to close port but it does not help. Do you have any idea that I can cancel all data transfer on this old port?

Let me know if you need some more information.

Thank you

msmeissn, 2020-12-13:
Not sure I 100% understand...

If you unplug/replug the camera directly , or via its hub, the cameras will be completely "new" to the system, and need to be initialized fresh.

So old connections will show IO errors and are not able to be used.

You will need to redetect and reinit them.

If lsusb is not showing them after their hub is plugged back in, they might have went to powersave or simiklar? It is likely a problem on a lower level than libgphoto2.

See if there are errors in dmesg.

dungcules, 2020-12-20:
Thanks you for your response, Marcus.

I connected 5 devices including my two cameras into this hub. The cameras are connected on port 2 and 3. I made following tests with my code or using gphoto2 CLI but the results are the same:

1. Dont run anything, just unplug the hub and re-plug. All devices showed up on lsusb and dmesg immediately.
2. Make a loop to capture images every 2 seconds. After an amount of time, unplug and replug the hub. The device on port 1 and the hub showed up immediately on lsusb and dmesg. However, cameras and other two devices did not show up. It blocked my hub in about 60s seconds. After the camera on port 2 showed up, all devices on other ports are recognized immediately. 

I dont know if the camera used synchronous communication or not. But it looks like the camera is blocked when its response packages (after capturing) are not picked up due to disconnection. If this is the case, I dont know how to deal with this situation except unplug the camera (This is not the solution that I want!!!)

As you mentioned, camera might switch into powersave mode. I will send out a message to the manufacturer to confirm this.

Thanks you,

Dung