gphoto/libgphoto2 #602: Support for Pentax K-r

jpcuzz, 2020-12-16:
I'd love to get gphoto2 working with my Pentax K-r.
I'm willing to do testing and some debugging.

**Name of the camera*
Pentax K-r
Use the Marketing name.

**USB IDs**
0a17:0100 Pentax Corp.

**camera summary output**
run

  gphoto2 --summary > summary.txt

attach summary.txt here
In "MSC" mode, I get:

> *** Error ***              
> 
> This camera does not support summaries.
> *** Error (-6: 'Unsupported operation') ***

In PTP mode, see [summary.txt](https://github.com/gphoto/libgphoto2/files/5702354/summary.txt)

gphoto2 --list-all-config > list-all-config.txt

In MSC mode, I get the following error. 

> *** Error ***              
> This camera does not provide any configuration options.
> *** Error (-6: 'Unsupported operation') ***       

In PTP mode, please see [list-all-config.txt](https://github.com/gphoto/libgphoto2/files/5702353/list-all-config.txt)

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
gives the following error:

> *** Error ***              
> Sorry, your camera does not support generic capture
> ERROR: Could not capture image.
> ERROR: Could not capture.
> *** Error (-6: 'Unsupported operation') ***       

 Similarly
  gphoto2 --capture-preview
  gives the following error:
 

>  *** Error (-6: 'Unsupported operation') ***

msmeissn, 2020-12-16:
what are the USB ids in Mass Storage mode?

The Pentax uses massstorage scsi tunneling of commands.

Run 
gphoto2 --auto-detect

in MSC mode, and try selecting the Pentax entry.

jpcuzz, 2020-12-17:
Sorry, I didn't realize the USB IDs were different between MSC and PTP.
0a17:0100 MSC
0a17:0101 PTP

In Mass Storage mode the output of gphoto2 --auto-detect is:

> Model                          Port                                            
> ----------------------------------------------------------