gphoto/libgphoto2 #603: Support for Fujifilm X-S10

dbrgn, 2020-12-22:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera**

Fujifilm X-S10 (released late 2020)

**USB IDs**

    Bus 002 Device 011: ID 04cb:02ea Fuji Photo Film Co., Ltd USB PTP Camera

**camera summary output**

[summary.txt](https://github.com/gphoto/libgphoto2/files/5729306/summary.txt) (serial number is censored)

**camera configuration output**

[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/5729313/list-all-config.txt) (seiral number is censored)

**test capture**

Capturing does not work at all:

```
$ gphoto2 --capture-image-and-download

*** Error ***
Sorry, your camera does not support generic capture
ERROR: Could not capture image.
ERROR: Could not capture.
*** Error (-6: 'Unsupported operation') ***
```

```
$ gphoto2 --capture-preview
*** Error (-1: 'Unspecified error') ***
```

May be closely related to #505 since the camera shares the same sensor (and thus, maybe also capturing logic).

Let me know if I can help in any way!

msmeissn, 2020-12-22:
In general Fuji offers remote support apparently only for a limited list, e.g. the XT-<single digit> variant cameras, X-Pro2 and GFX, (perhaps X-H1).

Please check if Fuji itself claims to support it in their software..

For the X-S10 it does not look like it supports remote control.

dbrgn, 2020-12-22:
@msmeissn thanks for the fast reaction! According to the list at http://fujifilm-x.com/global/support/compatibility/software/x-webcam/, the following cams should support live capture:

GFX100, GFX 50S, GFX 50R, X-H1, X-Pro3, X-Pro2, X-T4, X-T3, X-T2, X-S10

msmeissn, 2020-12-22:
which gphoto2 and libgphoto2 version is this? if its not the last release, can you update libgphoto2?

Also, can you capture a debug trace of 

gphoto2 --capture-preview --debug --debug-logfile=capturepreview.log

dbrgn, 2020-12-22:
Wow, you're fast @msmeissn :slightly_smiling_face: 

I use the following versions:

```
- libgphoto 2.5.26
- gphoto2 2.5.26
- libgphoto2_port 0.12.0
```

I believe this is the latest stable release.

By the way, I noticed that `--capture-preview` works now that I switched the camera to "X Webcam" in "Set Up > Connection Settings > Connection Mode". However, `--capture-image-and-download` still does not work.

Debug trace: 

    gphoto2 --capture-preview --debug --debug-logfile=capturepreview.log

[capturepreview.log](https://github.com/gphoto/libgphoto2/files/5729677/capturepreview.log)

    gphoto2 --capture-image-and-download --debug --debug-logfile=captureimage.log

[captureimage.log](https://github.com/gphoto/libgphoto2/files/5729681/captureimage.log)

dbrgn, 2020-12-22:
I also noticed that in "X Webcam" mode, the `--capture-move` option seems to work as well, however ffmpeg reports roughly 3 fps and the video output looks more like 0.5-1 fps.

Command used (with v4l2loopback):
    
    gphoto2 --stdout --capture-movie | ffmpeg -i - -vcodec rawvideo -pix_fmt yuv420p -threads 0 -f v4l2 /dev/video0

msmeissn, 2020-12-23:
--capture-movie basically just takes repeated preview images, thats how it works ;)

Might they have just added liveview support, but not capture support? weird

msmeissn, 2020-12-23:
can you capture another --summary and --list-all-config output in this x webcam mode?

dbrgn, 2020-12-23:
Sure! This is with libgphoto2 development branch, at commit 118903d57192ec048b3b3dd4101aba5f0a264bf7, and with the X-S10 connected in "X Webcam" connection mode:

[summary.txt](https://github.com/gphoto/libgphoto2/files/5736195/summary.txt)
[list-all-config.txt](https://github.com/gphoto/libgphoto2/files/5736193/list-all-config.txt)

> --capture-movie basically just takes repeated preview images, thats how it works ;)

Is such a low frame rate (often less than 1 frame per second) to be expected?

I wonder what the "X Webcam" software does. I'll have to try it out on a Windows computer.

msmeissn, 2020-12-23:
All the big  DSLR vendors have released drivers to "hook" the DSLR liveview mode into Zoom/Teams/Skype/whatever video conferencing software, this has been all the rage over the year ;)
Fuji X Webcam software is such a plugin.

dbrgn, 2020-12-23:
Yes, I know, but it runs over USB so there _is_ some way to grab frames in a sufficient resolution at a decent framerate :slightly_smiling_face: That's also my main interest (I only rarely shoot tethered). However, I don't know if this is also via PTP, or some other proprietary protocol.

msmeissn, 2020-12-23:
it should be  higher framerate ... i had rented a XT4 some weeks ago and it seems to perform several frames / second at least.

can you try:

gphoto2 --capture-movie=1s --debug --debug-logfile=1second.log

then I can inspect that log for delays.

dbrgn, 2020-12-23:
Sure, here: [1second.log](https://github.com/gphoto/libgphoto2/files/5736252/1second.log)

msmeissn, 2020-12-23:
can you in libgphoto2, camlibs/ptp2/library.c  in 
3592                 int             tries = 10;

change that to tries = 20;   or 30 

and see if that makes it faster?

dbrgn, 2021-01-02:
Oh, yes, I changed it to 30 and it's quite smooth now! When piping it into v4l2loopback, I get smooth playback (with a few hundred ms of latency).

Do you need any logfiles?

msmeissn, 2021-01-03:
Thanks for the feedback! 
No , no further logfiles required currently.

closed 2021-01-03 by msmeissn

molusk, 2021-04-08:
Just for information, I just tested with Fuji X-E4 in X Webcam mode and it works great !
Thank's for your work 🥇

dbrgn, 2021-04-08:
(As a sidenote, if you want full resolution with the Fuji as a webcam, there are some very cheap (around 5-15$) HDMI-USB capture cards on Ali Express. I've had good experience with one that looks similar to this: https://www.amtel.co.ke/wp-content/uploads/2020/08/61kaDUNKcL._AC_SL1000_.jpg)