gphoto/libgphoto2 #604: Regression: gphoto2 no longer shuts down Sony Alpha-A6000

yorickvP, 2020-12-30:
**Describe the bug**

When downloading photo's over wifi, my camera no longer automatically shuts down after finishing.

**Name the camera**
```
Sony Alpha-A6000 (Control)     usb:005,012
```

**libgphoto2 and gphoto2 version**
```
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.23         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking
```

**To Reproduce**
Steps to reproduce the behavior:
1. put camera in "download to pc" mode
2. `gphoto2  -P --port ptpip:192.168.178.67 --skip-existing`
3. When finished, camera does not automatically shut down (UI says it will)

I've bisected this issue to https://github.com/gphoto/libgphoto2/commit/f1e3b5ae8f9d8258db9a19861749b2bb6ad57b1e, but it seems not doing this is desirable in many other cases. Maybe this behavior should be a flag?

Other discussion: https://github.com/falk0069/sony-pm-alt/issues/14, https://github.com/gphoto/libgphoto2/issues/323

msmeissn, 2021-01-01:
yeah, the only way to do it currently would be to have a flag. :/

peturdainn, 2021-07-01:
FYI, it also prevents my Sony RX10 mark IV from shutting down