gphoto/libgphoto2 #607: Can you provide the installation package on the windows10 system?

xiaqiuchu, 2021-01-11:
I tried to run gphoto2 on windows10, but I did not find a reliable solution on the Internet, and the installation packages provided in the tutorials I found were all a long time ago (newer cameras do not support it).

Can you provide the installation package on windows? I have seen cases where others have successfully compiled .exe, but many people (including me) don't know how to do it. If you can provide the installation package on windos, it will be very successful.

If I can't provide the installation package under windows, can I provide a tutorial running on windows? For example, windows10 and windows7.

msmeissn, 2021-01-12:
(from the gphoto project itself .. no ... we have however had some users that build it for windows)

xiaqiuchu, 2021-01-12:
I don't quite understand what you mean. can you describe it in more detail?

If Windows installation packages are not available, can you provide a tutorial on how to compile and run on Windows?

xiaqiuchu, 2021-01-12:
The Windows installation package in the tutorial found on the Internet was long ago and did not support new cameras.