gphoto/libgphoto2 #608: Canon EOS R support for HDR, set shutter speed range and Fv mode

bogardeblaksteen, 2021-01-11:
We can successfully control some features on a Canon EOS R with libgphoto2 2.5.26. but there is a lot of features/settings we can not seem to access. Is it because we are doing it wrong, or because there is only a select few settings controllable with libgphoto2?

If possible, we would like to control all parameters remotely, as if we had the camera in our hands, specifically HDR mode, "set shutter speed range" and Fv mode.

For example, shouldn't there be an Fv mode here:
 },
                    "autoexposuremode": {
                        "label": "Canon Auto Exposure Mode",
                        "type": "choice",
                        "value": "AV",
                        "choices": [
                            "P",
                            "TV",
                            "AV",
                            "Manual",
                            "Bulb",
                            "A_DEP",
                            "DEP",
                            "Custom",
                            "Lock",
                            "Green",
                            "Night Portrait",
                            "Sports",
                            "Portrait",
                            "Landscape",
                            "Closeup",
                            "Flash Off"
                        ]
                    },

Thank you very much.

When we run gphoto2 --set-config capture=on, this is the full readout:
    {
    "main": {
        "type": "section",
        "label": "Camera and Driver Configuration",
        "children": {
            "actions": {
                "type": "section",
                "label": "Camera Actions",
                "children": {
                    "autofocusdrive": {
                        "label": "Drive Canon DSLR Autofocus",
                        "type": "toggle",
                        "value": 0
                    },
                    "cancelautofocus": {
                        "label": "Cancel Canon DSLR Autofocus",
                        "type": "toggle",
                        "value": 0
                    },
                    "eosremoterelease": {
                        "label": "Canon EOS Remote Release",
                        "type": "choice",
                        "value": "None",
                        "choices": [
                            "None",
                            "Press Half",
                            "Press Full",
                            "Release Half",
                            "Release Full",
                            "Immediate",
                            "Press 1",
                            "Press 2",
                            "Press 3",
                            "Release 1",
                            "Release 2",
                            "Release 3"
                        ]
[... 768 lines cut ...]
                    "manufacturer": {
                        "label": "Camera Manufacturer",
                        "type": "string",
                        "value": "Canon.Inc"
                    },
                    "model": {
                        "label": "Camera Model",
                        "type": "string",
                        "value": "2147484708"
                    },
                    "serialnumber": {
                        "label": "Serial Number",
                        "type": "string",
                        "value": "f522ed3711684f8490d0d352b146eef1"
                    },
                    "vendorextension": {
                        "label": "Vendor Extension",
                        "type": "string",
                        "value": "None"
                    }
                }
            }
        }
    }
}

msmeissn, 2021-01-31:
I need to implement decoders for those usually, not all are covered.
i added autoexposuremodes from EOS M6 Mark II, which should likely cover a lot of the EOS R ones.

If you want to supply more:
- switch to this mode on camera,
- gphoto2 --get-config autoexposuremode
- report the Unknown value XXXX <-> which mode was selected  as list

I am also trying to see if I can implement other things you miss. shutter speed range seems not there on eos m6, or hidden in another menu

sorenkrabbe, 2021-02-06:
With a Canon EOS R, when I set mode to FV, --get-config returns:
```
...
Label: Canon Auto Exposure Mode
Readonly: 0
Type: RADIO
Current: Unknown value 0037
...
```
Hope this could helps

msmeissn, 2021-02-07:
FV i have already added in current libgphoto2 GIT.
aperture I need to check