gphoto/libgphoto2 #613: gp_camera_trigger_capture() stops generating EVENT_FILE_ADDED after a few captures

notnot, 2021-01-19:
gp_camera_trigger_capture() stops generating EVENT_FILE_ADDED after a few captures. Usually a handful of captures only. Most i have seen is 11. 
I am not sure if calling this function should also generate CAPTURE_COMPLETE? I see none of those. 

**Name the camera**
Sony Alpha-A6300 (Control)     usb:001,046 

**libgphoto2 and gphoto2 version**
gphoto2         2.5.20         gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.20         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

**To Reproduce**
Run gp_camera_trigger_capture() in a loop while handling camera events.

notnot, 2021-01-19:
Adding a pause (of 1 second) between captures helps a bit. I can get sequences of several hundred successful captures this way. It seems best to put the pause after gp_camera_file_get().

msmeissn, 2021-01-29:
As I did improvements to Sony capture in newer versions of libgphoto2, like e.g. 2.5.23, can you update libgphoto2 to the current 2.5.26 release and see if this makes it better.

notnot, 2021-01-29:
I will install the latest libgphoto2 soon. For the time being i can't test with the Sony a6300 because it is working in a machine 24/7 where i am testing if capturing has become more reliable after adding event queue draining to my capture code. This test can take weeks...

msmeissn, 2021-06-19:
please reopen if not fixed...

closed 2021-06-19 by msmeissn