gphoto/libgphoto2 #614: Add Canon EOS M200

yursan9, 2021-01-24:
Fill in fields below. If you don't know how, leave it free.

**Name of the camera*

Canon EOS M200

**USB IDs**
Bus 001 Device 005: ID 04a9:32ef Canon, Inc. Canon Digital Camera

**camera summary output**
x-special/nautilus-clipboard
copy
file:///home/yurizal/summary.txt

**camera configuration output**
x-special/nautilus-clipboard
copy
file:///home/yurizal/config.txt

**test capture**

test if capture perhaps already works:

  gphoto2 --capture-image-and-download
```
New file is in location /capt0000.cr3 on the camera
Saving file as capt0000.cr3
Deleting file /capt0000.cr3 on the camera
```
  gphoto2 --capture-preview
```
Saving file as capture_preview.jpg
```

closed 2021-01-24 by msmeissn

yursan9, 2021-01-24:
That was fast! :+1: Thank you @msmeissn

msmeissn, 2021-01-24:
No worries :) camera seems well supported already, just the id was missing in the list.