gphoto/libgphoto2 #616: version

papalamg, 2021-01-29:
installed on a new raspberry pi 3B the version shows
gphoto2.            2.5.26
libgphoto2.        2.5.22
libgphoto2port  0.12.0
although I installed version 2.5.26? 
With this version I recognize a long delay when using --capture-image with an Olympus OMD E-M1. In the version description is written in V2.5.26 there where some advantages for Olympus E-Mxx cameras.
It takes 3 s for the shutter to work, 6 more seconds for displaying the file, and more 4 s until finished.

msmeissn, 2021-01-30:
did you install libgphoto2 2.5.26? note gphoto2 is the commandline tool, libgphoto2 is the library with the drivers

papalamg, 2021-01-30:
Hallo,
Yes, I installed both. 
First try was with the tool gphoto2-updater.sh from here: https://github.com/gonzalo/gphoto2-updater <https://github.com/gonzalo/gphoto2-updater>

It says it will install libgphoto 2.5.26 and gphoto 2.5.26. At the first start I saw with "gphoto2 - v" Gphoto2 version 2.5.24 and libgphoto 2.5.22
After this I downloaded libgphoto2-2.5.26.tar.bz2 and gphoto2-2.5.26.tar.gz and installed gphot2o and the lib again with 
autoreconf --install —symlink
./configure
make -j "$cores"
make install

Thanks for your answer
Regards Peter

ndim, 2021-02-01:
This keeps coming up. Should we try to revive gphoto-suite for a git world, or otherwise offer a bundled package of libgphoto2 with gphoto2?