gphoto/libgphoto2 #617: Could not claim usb

ebolisa, 2021-01-30:
Hi,

Not sure if this' the right place to expose my issue so, I apologize in advance.

I installed the latest version on a Pi 4 (with no errors) and when I search for my Nikon D70 camera, I get the following response:
**_Nikon DSC D70 (PTP mode)       usb:001,011_**

However, when I try to take a picture I get the following error:
**_An error occurred in the io-library ('Could not claim the USB device')_**

Under FAQ I read up on Why do I get the error message "Could not claim the USB device"? but, I don't quite understand where I should look to sove the problem.

I appreciate any help I can get.
Regards.

Pi4 running OS from USB. No SD installed.

msmeissn, 2021-01-31:
gphoto2 is blocked by another tool, on the PI likely gvfs that makes the camera aavailable to Nautilus and other browsers.

try:

gio mount -s gphoto2

and then replugging the camera and see if that helps

ebolisa, 2021-01-31:
Thank you, made progress.

closed 2021-01-31 by ebolisa