gphoto/libgphoto2 #618: Failure to close Sessions with X-H1? (and other issues)

entanglemint, 2021-02-02:
Bug 1:  After executing a capture-image-and-download (either with gphoto2 or through python-gphoto) the camera remains in "writing" mode (orange/green blinking led) indefinitely.  While I can continue to capture images once in this state, I can no longer capture previews.   I haven't exhaustively tested functionality.
python 2.7 (sorry!)
```
f = camera.capture_preview() #works
f = camera.capture(gp.GP_CAPTURE_IMAGE) #works
f = camera.capture_preview() #doesn't work
---------------------------------------------------------------------------
GPhoto2Error                              Traceback (most recent call last)
<ipython-input-33-a3730bd7c37c> in <module>()
----> 1 f = camera.capture_preview()

GPhoto2Error: [-1] Unspecified error
-----------------------------------------------------------------------------
f = camera.capture(gp.GP_CAPTURE_IMAGE) #works
```

Bug 2: Sometimes I can adjust focus position (d171) with a write and sometimes I can't.  As far as I can tell, there is no feedback as to why/why not.  I am setting by writing
config = camera.config()
 fl = config.get_child_by_name('other').get_child_by_name('d171')
fl.set_value('xxx') 
camera.set_config(config)
As I mentioned sometimes it works and sometimes it doesn't.  This would be a really nice astrophotography feature using focusable lenses if I could get it to work reliably!

Model                          Port                                            
----------------------------------------------------------
Fuji Fujifilm X-H1             usb:001,009  

Mac (intel) os 10.15.7
output of: gphoto2 --version

gphoto2         2.5.26         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.26         standard camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1, clang, ltdl, no EXIF, USB, serial without locking

Happy to assist as much as my limited ability allows for this camera!

Thanks and appreciate all your hard work!

entanglemint, 2021-02-07:
I've been spending more time looking at this bug (and looking at your documentation on how to report) and  I have found that the first "bug" (the camera remaining active) was the user. It was also an erroneous bug report which I have corrected above.  

The key step to getting the camera "quiescent" is deleting the file after a capture event:
`gphoto2 --capture-image-and-download`
or 
```p = camera.capture_preview()
f = camera.capture(gp.GP_CAPTURE_IMAGE) 
cf = camera.file_get(f.folder, f.name, gp.GP_FILE_TYPE_NORMAL)
camera.file_delete(f.folder, f.name)
 p = camera.capture_preview()# now works properly```
Both work as intended and restore the camera to "quiescent".  The python-gphoto example patterns did not include the delete step above.  

I still reproduce the focusing error, with the same frustrating irregularity.  It appears that it may the more general "device busy" problem that shows up on Fujis.
'>gphoto2 --set-config FocusPosition=00 --debug'

> 0.000011 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
> 0.000025 main                        (2): gphoto2 2.5.26
> 0.000027 main                        (2): gphoto2 has been compiled with the following options:
> 0.000029 main                        (2):  + clang (C compiler used)
> 0.000030 main                        (2):  + popt (mandatory, for handling command-line parameters)
> 0.000031 main                        (2):  + no exif (for displaying EXIF information)
> 0.000033 main                        (2):  + no cdk (for accessing configuration options)
> 0.000034 main                        (2):  + no aa (for displaying live previews)
> 0.000035 main                        (2):  + jpeg (for displaying live previews in JPEG format)
> 0.000036 main                        (2):  + readline (for easy navigation in the shell)
> 0.000040 main                        (2): libgphoto2 2.5.26
> 0.000041 main                        (2): libgphoto2 has been compiled with the following options:
> 0.000114 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
> 0.000118 main                        (2):  + clang (C compiler used)
> 0.000120 main                        (2):  + ltdl (for portable loading of camlibs)
> 0.000121 main                        (2):  + no EXIF (for special handling of EXIF files)
> 0.000123 main                        (2): libgphoto2_port 0.12.0
> 0.000125 main                        (2): libgphoto2_port has been compiled with the following options:
> 0.000127 main                        (2):  + iolibs: disk ptpip serial usb1
> 0.000128 main                        (2):  + clang (C compiler used)
> 0.000129 main                        (2):  + ltdl (for portable loading of iolibs)
> 0.000131 main                        (2):  + no EXIF (for vusb)
> 0.000132 main                        (2):  + USB (libusb1, for USB cameras)
> 0.000133 main                        (2):  + serial (for serial cameras)
> 0.000135 main                        (2):  + no resmgr (serial port access and locking)
> 0.000136 main                        (2):  + no ttylock (serial port locking)
> 0.000138 main                        (2):  + no lockdev (serial port locking)
> 0.000139 main                        (2): CAMLIBS env var not set, using compile-time default instead
> 0.000141 main                        (2): IOLIBS env var not set, using compile-time default instead
> 0.000142 main                        (2): invoked with following arguments:
> 0.000143 main                        (2):   --set-config
> 0.000145 main                        (2):   d171=00
> 0.000146 main                        (2):   --debug
> ........
> 0.590954 gp_widget_set_value         (2): Setting value of widget 'FocusPosition' to '600'...
> 
> *** Error ***              
> The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).
>  

Even when this command is successful there is an failed operation but I don't know if is of any consequence:

> 0.648432 gp_camera_get_single_config [gphoto2-camera.c:903](0): 'camera->functions->get_single_config ( camera, name, widget, context)' failed: -2
> 1.033244 gp_widget_set_value         (2): Setting value of widget 'FocusPosition' to '600'...

Once it starts working it seems to keep on working...

And finally, on some infrequent occasion when I first power up/connect the camera I get the following error:
`gphoto2 --capture-image-and-download`
> *** Error ***              
> PTP Device Busy
> ERROR: Could not capture image.
> ERROR: Could not capture.
> *** Error (-110: 'I/O in progress') ***     

Which goes away if I run the command again.

And on a side note for anyone else who is out there, I run into a conflict with the default mac os x libiconv let to some strange system behavior.  I have to run gphoto2 as  
 `>DYLD_LIBRARY_PATH=/usr/local/Cellar/libiconv/1.16/lib/:$DYLD_LIBRARY_PATH; gphoto2`
If I export environment variable chrome and itunes stop working, which honestly doesn't make any sense to me, but has has been repeated.

msmeissn, 2021-02-07:
fuji capture is still a bit fragile... good to hear it is working a bit better.
can you capture debuglogfiles for those failures? perhaps we can enhance the code a bit.
(add --debug --debug-logfile=debug.log  to the gphoto2 commandline)

i have not tried focusposition myself when I rented the camera.

entanglemint, 2021-02-08:
I'll attach the log here.  I have yet to find guaranteed steps to reproduce.  What I can say for sure is that I have _NEVER_ received an error when executing `gphoto2 --set-config FocusPosition=xxx`  I only seem to be seeing  the error when I run `gphoto2 --set-config d171=xxx`  It seems that I am more likely to see the error if the camera has been sitting idle for a little while (but less than the auto-off time), but not always.

[debug_ad.log](https://github.com/gphoto/libgphoto2/files/5941284/debug_ad.log) 

- Successful `gphoto2 --set-config FocusPosition=00 --debug-logfile=debug_ad.log`
- Successful `gphoto2 --set-config d171=00 --debug-logfile=debug_ad.log`
- ~30 second delay
- Failed `gphoto2 --set-config d171=00 --debug-logfile=debug_ad.log`

I don't see a sign of the error in the debug file, but I still see the same error when there is a failure.
`The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).`

entanglemint, 2021-02-08:
Tomorrow I'll repeat a comparison of access by address vs access by name for other settings and see if this is more general.

entanglemint, 2021-02-09:
Note, First off I did a complete rebuild of the libgphoto2/gphoto2 from the _same_ sources, but this time correcting the linking path to libiconv, so I no longer have to do specify the DYLD path.  This seems to have changed my debug files, they now have much more information.

I have also, seen, for the first time in perhaps 200 runs, the error occur when using "FocusPosition" instead of d171.  (I note that in the debug file it looks like the failure occurs at a slightly different step)  I'm attaching the updated log files along with the terminal output:
[debug_fp.log](https://github.com/gphoto/libgphoto2/files/5948999/debug_fp.log)

> % gphoto2 --set-config d171=200 --debug --debug-logfile=debug_fp.log
> % gphoto2 --set-config d171=200 --debug --debug-logfile=debug_fp.log
>                                                                                
> *** Error ***              
> The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).
> % gphoto2 --set-config FocusPosition=200 --debug --debug-logfile=debug_fp.log
>                                                                                
> *** Error ***              
> The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).
> % gphoto2 --set-config FocusPosition=200 --debug --debug-logfile=debug_fp.log
> % gphoto2 --set-config FocusPosition=200 --debug --debug-logfile=debug_fp.log
> 

I have seen the same issue with LiveViewImageSize/d174.  I never saw the bug when using the name, only when using the address.

Reviewing the call-stack in the debug, it appears that when setting by address (d171) there is an extra read-back step `tp_usb_sendreq             (2): Sending PTP_OC 0x1014 (Get device property description) (0xd171) request...` which is absent when accessing by widget e.g. "FocusPosition" .  

This situation changes when I use members of the more traditional widgets.

I have done the following with shutterspeed.  I'm attaching both the terminal output and the log file.

> % gphoto2 --set-config 500d=35 --debug --debug-logfile=debug_ss.log
> % gphoto2 --set-config 500d=35 --debug --debug-logfile=debug_ss.log
> % gphoto2 --set-config shutterspeed=35 --debug --debug-logfile=debug_ss.log
> % gphoto2 --set-config shutterspeed=35 --debug --debug-logfile=debug_ss.log
>                                                                                
> *** Error ***              
> The property 'Shutter Speed' / 0x500d was not set (0x2019: PTP Device Busy)
> 
> *** Error ***              
> The property 'Shutter Speed' / 0x500d was not set (0x2019: PTP Device Busy)
> 
> *** Error ***              
> Failed to set new configuration value 35 for configuration entry shutterspeed.
> *** Error (-1: 'Unspecified error') ***       
> 
> % gphoto2 --set-config shutterspeed=1/10 --debug --debug-logfile=debug_ss.log
>                                                                                
> *** Error ***              
> The property 'Shutter Speed' / 0x500d was not set (0x2019: PTP Device Busy)
> 
> *** Error ***              
> The property 'Shutter Speed' / 0x500d was not set (0x2019: PTP Device Busy)
> 
> *** Error ***              
> Failed to set new configuration value 1/10 for configuration entry shutterspeed.
> *** Error (-1: 'Unspecified error') ***       
> 
> % gphoto2 --set-config shutterspeed=1/10 --debug --debug-logfile=debug_ss.log
> %

[debug_ss.log](https://github.com/gphoto/libgphoto2/files/5948979/debug_ss.log)

Is there a way to meter the communication delays?  I am wondering if there is a timing issue i.e. a minimum dead time between commands?  (This is coming for zero familiarity with PTP)  Given the way this is cropping up I have a suspicion that this is why indi_gphoto_ccd is so problematic for the my fuji camera at least.

entanglemint, 2021-02-12:
I've continued to spend time exploring this issue.  I've found that I'm able to get a clearer handle through python-gphoto2.  I have found how to very consistently generate the bug and a few things that have an impact on the behavior.  There seem to be some kind of timeout issues

First, I can 100% get the bug to happen with the following code:
```
>>> fp = camera.get_single_config('d171');time.sleep(13);gp.gp_camera_set_single_config(camera, 'd171',fp)
WARNING: gphoto2: (ptp_usb_getresp [usb.c:515]) PTP_OC 0x1016 receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (_set_config [config.c:10450]) 'ptp_generic_setdevicepropvalue (params, propid, &propval, dpd.DataType)' failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (gp_context_error) The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).
```
But: The following seems to work correctly sometimes:
```
>>> fp = camera.get_single_config('d171');time.sleep(12);gp.gp_camera_set_single_config(camera, 'd171',fp)
```
and the following seems to work 100% of the time.
```
>>> fp = camera.get_single_config('d171');time.sleep(11);gp.gp_camera_set_single_config(camera, 'd171',fp)
```

On the shorter end, The following fails 100% of the time:
```
time.sleep(12);camera.get_single_config('d171');time.sleep(0);gp.gp_camera_set_single_config(camera, 'd171',fp);
```
but the following works 100% of the time:
```
time.sleep(12);camera.get_single_config('d171');time.sleep(.1);gp.gp_camera_set_single_config(camera, 'd171',fp);
```
and finally, 
```
time.sleep(12);camera.get_single_config('d171');time.sleep(.01);gp.gp_camera_set_single_config(camera, 'd171',fp);
```
also fails 100% of the time.

For me this at least provides some clarity and a clear workaround for my own programs, but now sure if there would be a good way to propagate this to e.g. indi_gphoto.  Also,  I can only test on my x-h1.

entanglemint, 2021-02-13:
I have an ugly ugly fix for the problem.   Adding a pause ptp_usb_sendreq fixes the problem. 
I made the following change

```
usb.c:107:  usleep(100000);
```

This isn't a good fix because everything becomes slow.  Perhaps there would be a way to add in this pause if a request has not gone out in the past 10 seconds time.  Honestly I have trouble navigating the code so I don't know how I to properly implement something like this.    I should have the chance to test this more exhaustively tomorrow.

entanglemint, 2021-02-14:
I've tried to push forward with a slightly more comprehensive solution and this is making it clear that the underlying issues is a little more subtle than the above fix would indicate.  

First, it only actually works _most_ of the time. As far as I can tell it always works from the command line (`gphoto2 --set-config d171=2000`) but it only works sometimes from python (see below).

The python code is even stranger.  The fix seems to always work for sleep times of 10 or 13 seconds.  (what I tested when I made the previous post) But for a sleep of 12 seconds, e.g. 
```
fp = camera.get_single_config('d171'); time.sleep(13); gp.gp_camera_set_single_config(camera, 'd171',fp)
```
The write fails every _second_  attempt.  This latter behavior was tested with the following code, with a random sleep introduced to make sure that there wasn't another time artifact.
```
ts = 12;
while 1:
     fp = camera.get_single_config('d171');time.sleep(ts);print('\nsleeping for {} seconds\n\n'.format(ts)),gp.gp_camera_set_single_config(camera, 'd171',fp)
     time.sleep(random.randint(0,15))
```

I next proceeded to add in a global `last_access_time` in `usb.c` and then to check when the last access was.  Replacing line 107 in usb.c with the following
```
        time(&current_time);
        dt = difftime(current_time, last_access_time);
	if (dt > 10)
	{
		GP_LOG_D ("Excessive time since last access, %f, sleeping...", dt);
		usleep(100000);
	}
       last_access_time = current_time;
```
This now seems to allow me to set the property from the python environment most but not all of the time, i.e. a command that failed 100% previously now succeeds 99% of the time:
```
fp = camera.get_single_config('d171'); time.sleep(13); gp.gp_camera_set_single_config(camera, 'd171',fp)
```
However, `gphoto2--set-config`  fails quite frequently.  

Again, in all cases, "fail" means `(ptp_usb_getresp [usb.c:533]) PTP_OC 0x1016 receiving resp failed: PTP Device Busy (0x2019)`

Finally, reverting to the default library:
I have tried an alternate to restarting the clock using `get_single_config`, using two `set_config` commands:
```
fp = camera.get_single_config('d171');time.sleep(ts);gp.gp_camera_set_single_config(camera, 'd171',fp);time.sleep(.1);gp.gp_camera_set_single_config(camera, 'd171',fp)
```
This shows the following behavior:

- `ts=10`: no error is ever thrown
- `ts=12`: every other attempt _both_ `gp_camera_set_single_config`s fail
- `ts=13`: the first set always fails, and the second set always succeeds
Finally, I note that I need the same inter-command sleep of 0.1s or both sets will always fail.

Are there any directions to look at?  I am really scratching my head here!  

DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1014 (Get device property description) (0xd171) request...
DEBUG: gphoto2: (ptp_usb_sendreq) Excessive time since last access, 13.000000, sleeping...
DEBUG: gphoto2: (ptp_usb_getdata) Reading PTP_OC 0x1014 (Get device property description) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1014 (Get device property description) response...
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0xd171) request...
DEBUG: gphoto2: (ptp_usb_sendreq) Last ptp_usb_sendreq access: 1613259194.000000
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...
WARNING: gphoto2: (ptp_usb_getresp [usb.c:535]) PTP_OC 0x1016 receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (_set_config [config.c:10450]) 'ptp_generic_setdevicepropvalue (params, propid, &propval, dpd.DataType)' failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (gp_context_error) The property 'FocusPosition' / 0xd171 was not set (0x2019: PTP Device Busy).
```

msmeissn, 2021-02-14:
It is hard to help from the distance. I would try out things like
- adding wait events in between startup and set (via gp_camera_wait_for_event()) to see if the camera need some event polling
- alternatively , also try to hack the  ptp_generic_setdevicepropvalue to have a wait-and-retry logic in config.c (the last one around line 10850)
- or create a own put function for d171 via the config tree mechanisms

entanglemint, 2021-02-14:
I'll continue to look in to these suggestions.  

In the mean time I've been playing around to see how specific this issue is to Focus position.  I'm seeing the problem with some pretty normal patterns as well, although the syndrome is slightly different. 

First, it seems like I can pretty much always set shutterspeed or iso at will by themselves.  BUT if I try to write more than one parameter at a time, I start to see the problem again.  This problem never shows up in ```gphoto2``` because it seems to always to the initial readback of config.
```
ts = 15 # or 10
config = camera.get_config()
print '*'*50  
print '\n\n\n' 
# I'll only reproduce from here on down
config.get_child_by_name('shutterspeed').set_value('1/60')
config.get_child_by_name('iso').set_value('400')
config.get_child_by_name('d174').set_value('2')
camera.set_config(config)
print '\nsleeping {}\n'.format(ts)
time.sleep(ts)
config.get_child_by_name('shutterspeed').set_value('1/10')
config.get_child_by_name('iso').set_value('100')
config.get_child_by_name('d174').set_value('1')
camera.set_config(config)
```
When `ts=10` I never get an error.  
When `ts=15` I always see a failure on the second and subsequent parameter sets.

```
DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'Shutter Speed' to '1/50'...
DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'ISO Speed' to '200'...
DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'LiveViewImageSize' to '2'...
DEBUG: gphoto2: (_set_config) Setting property 'ISO Speed' / 0x500f
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0x500f) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...
DEBUG: gphoto2: (_set_config) Setting property 'Shutter Speed' / 0x500d
DEBUG: gphoto2: (_put_Genericu32Table) FOUND right value for 1/50 in the enumeration at val 19686
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0x500d) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0xd174) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...

sleeping 15

DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'Shutter Speed' to '1/10'...
DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'ISO Speed' to '100'...
DEBUG: gphoto2: (gp_widget_set_value) Setting value of widget 'LiveViewImageSize' to '1'...
DEBUG: gphoto2: (_set_config) Setting property 'ISO Speed' / 0x500f
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1014 (Get device property description) (0x500f) request...
DEBUG: gphoto2: (ptp_usb_getdata) Reading PTP_OC 0x1014 (Get device property description) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1014 (Get device property description) response...
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0x500f) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...
DEBUG: gphoto2: (_set_config) Setting property 'Shutter Speed' / 0x500d
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1014 (Get device property description) (0x500d) request...
DEBUG: gphoto2: (ptp_usb_getdata) Reading PTP_OC 0x1014 (Get device property description) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1014 (Get device property description) response...
DEBUG: gphoto2: (_put_Genericu32Table) FOUND right value for 1/10 in the enumeration at val 99212
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0x500d) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1016 (Set device property value) response...
WARNING: gphoto2: (ptp_usb_getresp [usb.c:535]) PTP_OC 0x1016 receiving resp failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (_set_config [config.c:10317]) 'ptp_generic_setdevicepropvalue (params, cursub->propid, &propval, cursub->type)' failed: PTP Device Busy (0x2019)
WARNING: gphoto2: (gp_context_error) The property 'Shutter Speed' / 0x500d was not set (0x2019: PTP Device Busy)
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1014 (Get device property description) (0xd174) request...
DEBUG: gphoto2: (ptp_usb_getdata) Reading PTP_OC 0x1014 (Get device property description) data...
DEBUG: gphoto2: (ptp_usb_getresp) Reading PTP_OC 0x1014 (Get device property description) response...
DEBUG: gphoto2: (ptp_usb_sendreq) Sending PTP_OC 0x1016 (Set device property value) (0xd174) request...
DEBUG: gphoto2: (ptp_usb_senddata) Sending PTP_OC 0x1016 (Set device property value) data...
[... 4 lines cut ...]
```

Regardless of the focus position issue, this I think is a major issue.  Any way to check with other fuji users?

[... cut ...]
This fix seems to have fixed everything _but_ my specific 12 second bug with focus position. 

So unless I happen to have a 12 second delay, I can now always run the following:
```
config.get_child_by_name('shutterspeed').set_value('1/60')
config.get_child_by_name('iso').set_value('400')
config.get_child_by_name('d174').set_value('2')
config.get_child_by_name('d171').set_value('1000')
camera.set_config(config)
```

Finally, I have checked for events after the get_config and I don't see any events generated.

entanglemint, 2021-02-14:
I will also add that this has now seems to have cleared up my control issues from INDI.  I think that INDO was writing multiple parameters in a single transaction as above causing the write timeouts and having captures fail repeatedly.    This seems to have solved a majority if issues for me in INDI related to captures.  (Still live view, jpeg instead of raw, FITS conversion etc, but the basic camera operation seems to work great now!)