gphoto/libgphoto2 #620: X-T3 error 200a on query for all Device Property Summary

jasonhr13, 2021-02-02:
`gphoto2 --summary`

gives

```
Camera summary:                                                                
Manufacturer: FUJIFILM
Model: X-T3
  Version: 4.00
  Serial Number: 593534373732190520A39320212AC5
Vendor Extension ID: 0xe (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0; 

Capture Formats: 
Display Formats: JPEG, JFIF

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
	StorageDescription: External Memory
	VolumeLabel: 000000000000000000000000000000000000000000000000
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 63864569856 (60906 MB)
	Free Space (Bytes): 45372669952 (43270 MB)
	Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 10, step 1] value: 10% (10)
Property 0xd303:(read only) (type=0x2) 1
Property 0xd406:(readwrite) (type=0xffff) ''
Property 0xd407:(read only) (type=0x6) 1
Exposure Time(0x500d): error 200a on query.
F-Number(0x5007): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
FocusPosition(0xd171): error 200a on query.
Property 0xd21c: error 200a on query.
Focus Point(0xd347): error 200a on query.
LensZoomPos(0xd170): error 200a on query.
BatteryLevel(0xd242): error 200a on query.
LiveViewImageSize(0xd174): error 200a on query.
VideoOutOnOff(0xd168): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
ForceMode(0xd230): error 200a on query.
Property 0xd16e: error 200a on query.
FocusArea1(0xd372): error 200a on query.
FaceDetectionMode(0xd020): error 200a on query.
RawCompression(0xd022): error 200a on query.
GrainEffect(0xd023): error 200a on query.
SetEyeAFMode(0xd024): error 200a on query.
FocusPoints(0xd025): error 200a on query.
MFAssistMode(0xd026): error 200a on query.
InterlockAEAFArea(0xd027): error 200a on query.
Shadowing(0xd029): error 200a on query.
CropMode(0xd16f): error 200a on query.
TNumber(0xd02f): error 200a on query.
FocusArea4(0xd395): error 200a on query.
HighLightTone(0xd320): error 200a on query.
ShadowTone(0xd321): error 200a on query.
LongExposureNR(0xd322): error 200a on query.
FullTimeManualFocus(0xd323): error 200a on query.
LensISSwitch(0xd346): error 200a on query.
InstantAFMode(0xd34a): error 200a on query.
PreAFMode(0xd34b): error 200a on query.
LMOMode(0xd34d): error 200a on query.
ISMode(0xd351): error 200a on query.
FocusCheckMode(0xd35e): error 200a on query.
LiveViewImageQuality(0xd173): error 200a on query.
FileNamePrefix1(0xd365): error 200a on query.
FileNamePrefix2(0xd366): error 200a on query.
FocusArea3(0xd374): error 200a on query.
TotalShotCount(0xd310): error 200a on query.
ExposurePreview(0xd359): error 200a on query.
FrameGuideGridInfo1(0xd375): error 200a on query.
FrameGuideGridInfo2(0xd376): error 200a on query.
CustomDispInfo(0xd36e): error 200a on query.
ViewMode1(0xd33f): error 200a on query.
CustomAutoPowerOff(0xd364): error 200a on query.
LockButtonMode(0xd34e): error 200a on query.
WideDynamicRange(0xd02e): error 200a on query.
LensNameAndSerial(0xd36d): error 200a on query.
LensUnknownData(0xd38a): error 200a on query.
BatteryInfo1(0xd36a): error 200a on query.
BatteryInfo2(0xd36b): error 200a on query.
FunctionLockCategory1(0xd36f): error 200a on query.
FunctionLockCategory2(0xd370): error 200a on query.
SensitivityFineTune1(0xd222): error 200a on query.
SensitivityFineTune2(0xd223): error 200a on query.
LensZoomPosCaps(0xd38c): error 200a on query.
LensFNumberList(0xd38d): error 200a on query.
LensFocalLengthList(0xd38e): error 200a on query.
Property 0xd17b: error 200a on query.
```

Wondering if it is possible to set things like HighlightTone from gphoto2.

msmeissn, 2021-02-03:
is the camera in the right mode?

We have a static list of props, the error 0x200a means the camera reports it as not present, but it does not break things.

Prototypers-lair, 2021-05-09:
I am having the same issue, also with an X-T3. It seems like the camera is "busy", but it is set up (on the camera) to USB tether auto. As far as I've been able to tell, that's what it's supposed to be set to. Below are the first few lines from summary.txt, everything below this is also error 200a

Camera summary:
Manufacturer: FUJIFILM
Model: X-T3
  Version: 4.10
  Serial Number: *removed*
Vendor Extension ID: 0x6 (1.0)
Vendor Extension Description: fujifilm.co.jp: 1.0;

Capture Formats: Defined Type JPEG Unknown(b103)
Display Formats: Defined Type, JPEG, Unknown(b103)
Supported MTP Object Properties:
        Defined Type/3800: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        JPEG/3801: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth
        Unknown(b103)/b103: dc01/StorageID dc02/ObjectFormat dc03/ProtectionStatus dc04/ObjectSize dc07/ObjectFileName dc08/DateCreated dc0b/ParentObject dc41/PersistantUniqueObjectIdentifier dc44/Name dc87/Width dc88/Height dcd3/ImageBitDepth

Device Capabilities:
        File Download, File Deletion, No File Upload
        Generic Image Capture, Open Capture, No vendor specific capture

Storage Devices Summary:
store_10000001:
        StorageDescription: Still
        VolumeLabel: 15935343737341901249C533021CC4E
        Storage Type: Builtin RAM
        Filesystemtype: Generic Hierarchical
        Access Capability: Read Only with Object deletion
        Maximum Capability: 18446744073709551615 (4294967295 MB)
        Free Space (Bytes): 18446744073709551615 (4294967295 MB)
        Free Space (Images): -1
store_10000002:
        StorageDescription: Live
        VolumeLabel: 25935343737341901249C533021CC4E
        Storage Type: Builtin RAM
        Filesystemtype: Generic Hierarchical
        Access Capability: Read Only with Object deletion
        Maximum Capability: 18446744073709551615 (4294967295 MB)
        Free Space (Bytes): 18446744073709551615 (4294967295 MB)
        Free Space (Images): -1

Device Property Summary:
Battery Level(0x5001): error 200a on query.
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
        '3120x2080',
        '3120x1760',
        '2080x2080',
        '4416x2944',
        '4416x2488',
        '2944x2944',
        '6240x4160',
        '6240x3512',
        '4160x4160',
        '4992x3328',
        '4992x2808',
        '3328x3328'
        ] value: '6240x4160'
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,6,8,32769,32770,32771,32774,32775,32776,32777,32778] value: Automatic (2)
Focus Mode(0x500a):(readwrite) (type=0x4) Enumeration [1,32769,32770] value: 32769
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [1,2,3,4] value: Multi-spot (3)
Flash Mode(0x500c): error 200a on query.
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [1,3,4,6] value: 6
Exposure Index (film speed ISO)(0x500f):(readwrite) (type=0x5) Enumeration [80,100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,25600,51200,-1,-2,-3] value: ISO -3 (-3)
Date & Time(0x5011):(readwrite) (type=0xffff) '20210509T201500'
Pre-Capture Delay(0x5012):(readwrite) (type=0x4) Enumeration [10000,2000,0] value: 0.0s (0)
Sharpness(0x5015):(readwrite) (type=0x3) Range [-40 - 40, step 10] value: 0
Burst Number(0x5018): error 200a on query.
Burst Interval(0x5019): error 200a on query.
Focus Metering Mode(0x501c):(readwrite) (type=0x4) Enumeration [32769,32770,32771] value: 32769
Property 0xd001:(readwrite) (type=0x4) Enumeration [1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16] value: 1
Property 0xd002: error 200a on query.