gphoto/libgphoto2 #622: Can trigger photos on Canon 600D but cannot download to raspberry pi

kevinlowen1, 2021-02-07:
**Describe the bug**

I am able to trigger photos on Canon 600D but cannot download to raspberry pi.
I run the commands listed below and while they seem to be creating photos on the canon 600D.  The photos are not downloading onto my raspberry pi that is connected to the camera.
I have checked that mirror-lock is off on the camera.  And have tried reinstalling libgphoto2 manually because i read somewhere that the version that comes with the raspberry pi is not up to date.

I am trying to use applications called kstars and ekos to be able to take photos on my dslr through my raspberry pi.
After troubleshooting that application it appears that the issue is with the photo tool on the raspberry pi (gphoto2).
I have tried taking photos directly on the raspberry pi with the commands below and seem to be coming up with issues so i think i am on the right track thankfully.

**Name the camera**
`lsusb`

>Bus 001 Device 009: ID 04a9:3218 Canon, Inc. EOS 600D / Rebel T3i (ptp) 

`gphoto2 --auto-detect`

>Model                          Port                                            
>----------------------------------------------------------
>Canon EOS 600D                 usb:001,009 

**libgphoto2 and gphoto2 version**

`gphoto2 --version`

>gphoto2 2.5.23.1

>Copyright (c) 2000-2020 Marcus Meissner and others

>gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
>redistribute copies of gphoto2 under the terms of the GNU General Public
>License. For more information about these matters, see the files named COPYING.

>This version of gphoto2 is using the following software versions and options:
>gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
>libgphoto2      2.5.26.1       standard camlibs, gcc, ltdl, EXIF
>libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking<<

**Debug Logs**
query 1:
`gphoto2 --capture-image --debug 2>> debug.txt`
[debug.txt](https://github.com/kevinlowen1/bugs/blob/main/debug1.txt)

query 2:
`gphoto2 --camera « Canon EOS 600D » --port usb: --capture-image --debug 2> debug2.txt`
[debug2.txt](https://github.com/kevinlowen1/bugs/blob/main/debug2.txt)

query 3:
`gphoto2 --camera « Canon EOS 600D » --port usb: --capture-image-and-download --debug 2> debug3.txt`
[debug3.txt](https://github.com/kevinlowen1/bugs/blob/main/debug3.txt)

after rebooting the canon i run the below command and get this result
`gphoto2 --capture-image`
>New file is in location /store_00020001/DCIM/100CANON/IMG_7042.CR2 on the camera

then i run this command and get the below result
`gphoto2 --capture-image-and-download`
>*** Error ***              
>error 0x02ff
>New file is in location /store_00020001/DCIM/100CANON/IMG_7043.CR2 on the camera

>*** Error ***              
>PTP I/O Error
>ERROR: Could not get image.

Finally, running the original command results in this error.
`gphoto2 --capture-image`
>*** Error ***              
>PTP Timeout

>*** Error ***              
>An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available
>ERROR: Could not capture image.
>ERROR: Could not capture.

>*** Error ***              
>PTP Timeout

>*** Error ***              
>An error occurred in the io-library ('Timeout reading from or writing to the port'): No error description available

kevinlowen1, 2021-02-07:
maybe it should be noted that the queries are all being run as the pi user on the raspberrypi computer.

msmeissn, 2021-02-08:
fwiw , you can leave out the --camera and --port specifiers if you have just 1 cameras.

debug3.txt shows that libgphoto2 tries to transfer the image, but it seems to crash the camera.

- is regular filetransfer using gphoto2 -P  working at all?
-  are there hardware issues ? 
- can you try a different SD Card?

kevinlowen1, 2021-02-08:
1)The 3 commands were all run consecutively and i believe something happens after running the second which causes any further commands to have an issue.
1A)Running the third command before any other commands returns this debug log:
`gphoto2 --camera « Canon EOS 600D » --port usb: --capture-image-and-download --debug 2> debug4.txt`
[debug4.txt](https://github.com/kevinlowen1/bugs/blob/main/debug4.txt)

2)I have also tried with a microsd card and an sd adapter (i do not have a second sd card) and I have the same results
as in the original post

3)I cannot rule out hardware issues entirely but the raspberry pi seems to work with a separate camera and the canon dslr works in all other aspects except this.  I don't really have a second canon to test with though.

4)running `gphoto2 -P`  before any other commands returns:
>*** Error ***              
>error 0x02ff

>*** Error ***              
>PTP General Error
>*** Error (-1: 'Unspecified error') ***       

>For debugging messages, please use the --debug option.
>Debugging messages may help finding a solution to your problem.
>If you intend to send any error or debug messages to the gphoto
>developer mailing list <gphoto-devel@lists.sourceforge.net>, please run
>gphoto2 as follows:

>    env LANG=C gphoto2 --debug --debug-logfile=my-logfile.txt -P

>Please make sure there is sufficient quoting around the arguments.

4A)Also running `gphoto2 -P` makes it so the next time i run `gphoto2 --capture-image` i get an error as well.

kevinlowen1, 2021-02-08:
I can't believe this but i actually just figured out the issue.  I was using a usb hub since the raspberry pi only has a single micro-usb port.  Connecting without the hub makes it so that everything works almost perfectly.

Your "hardware issues" suggestion got me thinking about that so thank you!

closed 2021-02-08 by kevinlowen1

msmeissn, 2021-02-09:
good to hear! :)