gphoto/libgphoto2 #623: Canon EOS 1500D and 2000D duplicate USB IDs

knro, 2021-02-09:
**Describe the bug**

It appears that both 1500D and 2000D have identical USB VID & PID identifiers.

camlibs/ptp2/library.c: {"Canon:EOS 2000D",                     0x04a9, 0x32e1, PTP_CAP|PTP_CAP_PREVIEW|PTPBUG_DELETE_SENDS_EVENT},
camlibs/ptp2/library.c: {"Canon:EOS 1500D",                     0x04a9, 0x32e1, PTP_CAP|PTP_CAP_PREVIEW|PTP_DONT_CLOSE_SESSION},

So the 2000D is now getting detected as 1500D. Is there a mistake somewhere in 1500D IDs? since the 2000D is indeed correct. Or this could be a case of Canon somehow using the same USB IDs for both?

msmeissn, 2021-02-09:
I actually have logfiles for both cameras, and they seem to report the same USB ID.
Perhaps someone can cross check that.

knro, 2021-02-09:
Thank you. Is there a way then to tell me apart? from the product name perhaps?

msmeissn, 2021-02-09:
Yes... 
There is a "cameramodel" configuration value to query for instance

knro, 2021-02-10:
So is this something that can be fixed in libgphoto2 to correctly identify the camera in case of duplicate USB IDs?

hfiguiere, 2021-02-10:
> So is this something that can be fixed in libgphoto2 to correctly identify the camera in case of duplicate USB IDs?

The both are the Rebel T7. (ref: https://sg.canon/en/support/0400542202/1 )
Also known as the Kiss X90.

Canon marketing....

msmeissn, 2021-02-13:
the autodetection is only using the UBS IDs, so we cannot diferentiate between 1500 , 2000 or the Rebel flavor. 
I can add additional names to the same USB ID, but only the first in the list will be printed with --auto-detect

msmeissn, 2021-02-13:
i could also write in the name Canon EOS 1500D/2000D(Rebel T7  or similar. but its a but ugly to my taste.

manhha1402, 2021-11-09:
I have the same issue. I plug canon eos 2000d. The command `gphoto2 --auto-detect` detect this camera as canon eos 1500D

msmeissn, 2021-11-12:
i dont know why canon chose this, but i think it just that they use the same firmware for 1500D and 2000D (and Rebel equivalents).

This is nothing to worry about. 

We handle EOS generically anyway in libgphoto2.

closed 2021-11-12 by msmeissn