gphoto/libgphoto2 #624: Panasonic GH5 - stuck at end of capture-image-and-download

proven28, 2021-02-09:
**Describe the bug**

I recently got a Panasonic Lumix GH5 (DC-GH5) and saw that it was listed as compatible with gphoto2.  I carefully followed the guides for installation and DSLR setup, but I've run into a problem in Step 4 - Test your camera.
--capture-image seems to work, but when I test -capture-image-and-download it hangs at the end in a seemingly endless loop.  
[image]

[image]

It does successfully take the picture, download it from the camera to the Pi, and then delete the file from the camera, but after that it gets stuck.  The camera is completely functional, but the Pi doesn't go back to the command prompt.  I can't cancel out of it with CTRL-C, so I end up powering off the camera.  I have run it with --debug and created a log file below (I removed most of the JPG hex dump from the file to make it smaller, and noted the spot where I turned the camera off).

Any help would be appreciated--I'm trying to get this working for octolapse.

Thank you!

**Name the camera**
Panasonic DC-GH5     usb:002,007
Connected via PC(Tether) on the camera.  If I choose PTP, it comes up as the incorrect camera and capture-image says it isn't supported.

**libgphoto2 and gphoto2 version**
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.26.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.26.1       standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking

**To Reproduce**
This happens every time I try gphoto2 --capture-image-and-download, with or without --autodetect included.

Log File
[gh5log - Copy without JPG hex.log](https://github.com/gphoto/libgphoto2/files/5953488/gh5log.-.Copy.without.JPG.hex.log)

msmeissn, 2021-02-14:
I see you built libgphoto2 with the very old libusb0.

I strongly recommend to  build libgphoto2 with libusb1 development headers/libraries , not libusb0. This likely might solve this issue.
(libusb1-dev or similar named)

msmeissn, 2021-02-14:
I also added logic in gphoto2 to abort the wait at some point to avoid endless loops

proven28, 2021-02-16:
Thanks. I tried rebuilding it with the latest dev version of libusb and the rest of the libraries, but it's still giving me issues.  I even tried a fresh image just to be sure. Now it hangs after capture-image.  However, it is working on my old Rebel T3i, so I could use that one.  But I'm still trying to figure out the GH5 since it's my newer camera.

msmeissn, 2021-02-27:
can you check with gphoto2 --version that you use libusb1 and not libusb0 ?

proven28, 2021-02-27:
[image]

msmeissn, 2021-02-27:
hmm libghphoto2 version is back 2.5.22 :( but built with usb1 here, but its probably the system one :/

proven28, 2021-03-01:
I'm a bit of a noob when it comes to installing/compiling, so I'm not sure how I got an older version.  I used 'git clone https://github.com/gphoto/libgphoto2.git' before installing.  I'll try again and see if I get the latest version

proven28, 2021-03-07:
Okay.. so I reinstalled everything.  Now running gphoto2 and libgphoto 2.5.27.1 with libusb1.

[image]
[GH5hang.txt](https://github.com/gphoto/libgphoto2/files/6097706/GH5hang.txt)

It's still hanging when I do capture-image.  It says could not capture image, but the camera is taking the shot immediately.  I've attached the debug log here.

msmeissn, 2021-03-10:
mostly looking good. But  looking at the events i do not see the object added events. ... there are others emitted, but these seem not objectadded ones.

msmeissn, 2021-03-10:
(panasonic capture is all really underdocumented and reverse engineered... i rented a gc9 some weeks ago and that worked well after fixes. i rented a gh5 last year and that was working... not sure what changed :( )

qq297110281, 2021-12-13:
same issue here ,any news?

dustinkerstein, 2022-07-19:
Another quick update - I was able to get the capture working by switching to a USB2 port on the Jetson Nano. I'll do a little more testing later.

msmeissn, 2022-07-23:
considering solved, even if a bit unstable

s-ol, 2023-07-19:
I'm also having this problem and so far haven't been able to download any images.

```
Body Firmware: 2.8

gphoto2 --version
gphoto2 2.5.28

Copyright (c) 2000-2021 Marcus Meissner and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.28         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.30         standard camlibs, gcc, no ltdl, EXIF
libgphoto2_port 0.12.1         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, no ltdl, EXIF, USB, serial without locking
```

I'm running in Archlinux on windows WSL with USB forwarded (usbipd). `--capture-preview` and Entangle liveview work. Both with `--capture-image` and `--capture-image-and-download` it gets stuck waiting:

```
gphoto2 --debug --capture-image
0.000088 main                        (2): ALWAYS INCLUDE THE FOLLOWING LINES WHEN SENDING DEBUG MESSAGES TO THE MAILING LIST:
0.000150 main                        (2): gphoto2 2.5.28
0.000159 main                        (2): gphoto2 has been compiled with the following options:
0.000163 main                        (2):  + gcc (C compiler used)
0.000166 main                        (2):  + popt (mandatory, for handling command-line parameters)
0.000170 main                        (2):  + exif (for displaying EXIF information)
0.000173 main                        (2):  + no cdk (for accessing configuration options)
0.000176 main                        (2):  + no aa (for displaying live previews)
0.000179 main                        (2):  + jpeg (for displaying live previews in JPEG format)
0.000182 main                        (2):  + readline (for easy navigation in the shell)
0.000186 main                        (2): libgphoto2 2.5.30
0.000189 main                        (2): libgphoto2 has been compiled with the following options:
0.000193 main                        (2):  + standard camlib set (ax203 canon digigr8 dimagev directory docupen jl2005a jl2005c kodak_dc240 lumix mars pentax ptp2 ricoh_g3 sierra sonix sq905 st2205 topfield tp6801)
0.000199 main                        (2):  + gcc (C compiler used)
0.000202 main                        (2):  + no ltdl (for portable loading of camlibs)
0.000205 main                        (2):  + EXIF (for special handling of EXIF files)
0.000214 main                        (2): libgphoto2_port 0.12.1
0.000262 main                        (2): libgphoto2_port has been compiled with the following options:
0.000268 main                        (2):  + iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi
0.000272 main                        (2):  + gcc (C compiler used)
0.000280 main                        (2):  + no ltdl (for portable loading of iolibs)
0.000283 main                        (2):  + EXIF (for vusb)
0.000286 main                        (2):  + USB (libusb1, for USB cameras)
0.000289 main                        (2):  + serial (for serial cameras)
0.000292 main                        (2):  + no resmgr (serial port access and locking)
0.000295 main                        (2):  + no ttylock (serial port locking)
0.000300 main                        (2):  + no lockdev (serial port locking)
0.000303 main                        (2): CAMLIBS env var not set, using compile-time default instead
0.000344 main                        (2): IOLIBS env var not set, using compile-time default instead
0.000352 main                        (2): invoked with following arguments:
0.000356 main                        (2):   --debug
0.000361 main                        (2):   --capture-image
0.000380 load_settings               (2): Creating gphoto config directory ('/home/s-ol/.gphoto')
0.000499 load_settings               (2): Loading settings from file '/home/s-ol/.gphoto/settings'.
0.000536 main                        (2): The user has not specified both a model and a port. Try to figure them out.
0.000582 gp_port_init_localedir      (2): localedir has been set to /usr/share/locale (compile-time default)
0.000592 gp_port_info_list_load      (2): Using ltdl to load io-drivers from '/usr/lib/libgphoto2_port/0.12.1'...
0.000682 foreach_func                (2): Called for filename '/usr/lib/libgphoto2_port/0.12.1/disk'.
0.001048 gp_port_library_list        (2): found mtab fsname none
0.001113 gp_port_library_list        (2): found mtab fsname none
0.002158 gp_port_library_list        (2): found mtab fsname none
0.002242 gp_port_library_list        (2): found mtab fsname /dev/sdc
0.002317 gp_port_library_list        (2): found mtab fsname none
0.002414 gp_port_library_list        (2): found mtab fsname /dev/sdc
0.002534 gp_port_library_list        (2): found mtab fsname rootfs
0.002760 gp_port_library_list        (2): found mtab fsname none
[... 363 lines cut ...]
1.487200 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
1.649635 gp_port_check_int           (3): Reading 24 = 0x18 bytes from interrupt endpoint...
1.812940 gp_port_set_timeout         (2): Setting port timeout to 20000 milliseconds.
1.867629 camera_panasonic_capture    (2): **** GH5: checking for new object...
1.867798 gp_port_get_timeout         (2): Current port timeout is 20000 milliseconds.
[... cut ...]
3.623765 gp_context_error            (0): PTP General Error
```

Did the event maybe change? I don't know anything about this protocol, but is this "PTP_OC" perhaps suspicious?
```
0.445698 ptp_usb_getresp             (2): Reading PTP_OC 0x9404 (Unknown PTP_OC) response...
0.445711 gp_port_read                (3): Reading 1024 = 0x400 bytes from port...
0.450846 gp_port_read                (3): Read    12 = 0xc out of 1024 bytes from port: (hexdump of 12 bytes)
0000  0c 00 00 00 03 00 01 20-05 00 00 00              ....... ....
```

msmeissn, 2023-07-21:
9404 is "PANASONIC_InitiateCapture". it is just not yet decoded for debug printing.

A longer debug log perhaps?

s-ol, 2023-10-13:
@msmeissn: At some point it "randomly" started working sometimes, I then changed USB cable and it started working always (or most of the time?), albeit with rather long and variable delays. I've since switched to a Canon EOS model for this project since stability was key.