gphoto/libgphoto2 #629: Automate Trigger/Capture and Download Via USB Cable Without Saving To SD Card When Tethered on USB3?

vepicurean, 2021-03-02:
Hi:

I looked through the documentation, and I can't see how to trigger a "capture and download" command via gphoto2 without saving to the SD card, while also using tethered capture where the Linux computer is listening for trigger events via USB 3 from the camera? I hate using the camera's N3 port as a remote trigger (we've automated it using an Arduino MCU running a very complicated sketch via relays and some long custom made trigger cables which are a pain to design and build). 

We have be looking to figure out a way to just capture and download via the USB3 cable connection in rapid fashion (say every 2 to 3 seconds on average using either capture tethered or a bash script or Python script running on a Linux computer, but with an Arduino MCU acting as the virtual trigger to tell the Linux computer to execute the bash script (start the bash script and run a loop with the appropriate gphoto2 commands that listens to the serial connection between the Arduino and the Linux computer on a USB3 cable connection). In summary, the Arduino MCU runs a sketch that acts as virtual trigger via USB to the Linux computer. The Linux computer controls the capture and download to the cameras via differ USB3 ports and cable connections. This would eliminate the need for the external connection to the Canon N-3 connector to trigger the camera while it's tethered using gphoto2. 

Correct me if I'm wrong, but we can't enable tethered capture and also use use the standard gphoto2 Linux commands to capture and download (which we'd also like to automate)?

msmeissn, 2021-03-02:
I added some code to gphoto2 commandline tool, where you can now do

gphoto2 --wait-event-and-download=1000s 

and you can send a SIGUSR1 signal to this gphoto2, it will capture an image and download it.

and then run "killall -USR1 gphoto2" or  find the above PID of above gphoto2 and run "kill -USR1 PID"

bkapadia01, 2021-03-06:
Thanks for your response. 

Are you able to provide more details regarding the SIGUSR1 and how we can use it to trigger the image. Could you provide more details regarding how I can send that signal once we put gphoto2 into a tethered state.
`gphoto2 --wait-event-and-download=1000s` - what is the 1000s for?

bkapadia01, 2021-03-06:
We were not able to get `gphoto2 --wait-event-and-download=1000s`. we are currently using libgphoto2 2.5.27 which is the latest version.

Trying with `gphoto2 --capture-image-and-download --interval -1` and `kill -SIGUSR1 #PID#` works as expected.
We are currently using macOS for development plan to use the application on linux in production. 
We are wondering why killing the gphoto2 triggers a capture and doesn't disconnect the tether?

vepicurean, 2021-03-06:
We're not sure if the changes you made to gphoto2 --wait-event-and-download=1000s have been added via the package manager on MacOS and Linux yet?  Or would we need to download the latest repository from GitHub and compile from scratch to get the latest changes implemented on GitHub. I/we prefer to just let the package managers update the libgphoto2 app on MacOs and Linux. 

Also, separately, we're interested in comparing the stability and performance of just using a simple bash program calling the `gphoto2 --capture-image-and-download` command (calling this command repeatedly every say 2 seconds) using a bash script, vs using `gphoto2 --capture-image-and-download --interval -1` which seems to start a tether session and then using a script we issue the `kill -SIGUSR1 PID`, where the PID is the PID# for the gphoto2 process. The first method does not use a tether, while the second method uses a tether. Correct me if I'm wrong in my observation here. 

From what I've observed over time, my thinking is that the PTP (Picture Transfer Protocol) protocol may be hard on the processing capabilities of the Canon EOS cameras.  The slower we capture and download using the tether, the more stable the Canada EOS cameras seem to operate.   

Just wondering if you have a view on which method is best (stability is primary) to capture and download at around a 2 to 3 second rate (we'll never need to capture and download at more than a 2 second rate. The RAW image files on the EOS 5DS R range from 50Mbyte to close to 56Mbyte).

msmeissn, 2021-03-06:
the code is only in gphoto2 GIT master branch at this time and not yet in any release or a distribution.

So yes, currently you need to get gphoto2 GIT checkout and build yourself.

Stability of repeated calls of gphoto2 vs having a long running instance ... Hard to say. A long running instnace would be better on the camera I would say.

capturing and downloading at 2 second rate of RAW sized images seems challenging, not sure if this is achievable from USB bandwith and camera processing power capacity limits.

vepicurean, 2021-03-06:
Thanks for your reply. 

When you use the terminology "long running instance" is that the method of initiating a tether where I see the PTP protocol messages start and wait for an event? If yes, why is that better on the camera vs running a bash script repeatedly using `gphoto2 --capture-image-and-download` command (which does not use a tether)?

msmeissn, 2021-03-06:
the longrunning tether ... largely it saves initialization between computer and camera.

I think the impact on camera is probably quite low, but it might  saves bits of time when you plan to repeatedly call it as it saves several PTP command/reply roundtrips.

vepicurean, 2021-03-07:
Thanks for the clarification. That's what I thought you meant. I don't see the PTP messages when using the `gphoto2  --capture-image-and-download` command, but from your reply it seems that PTP messaging is occurring each time we call that command. 

When we use the `gphoto2 --capture-tethered` method, we notice a big difference in the stability based on the frequency of trigging a capture and download event.  Every 5 seconds capture frequency, after a while it just seems too much and the cameras can start acting up and even freeze. It you slow the capture and download event frequency down to every 10 seconds or more, then the stability increases significantly. Note that some of this has to do with Canon messing with Sigma (the Sigma art lens is vastly superior in producing sharper quality images than the Canon prime lens in this space), and Canon more or less did not deny that they are purposely messing with the operation of the camera if a non Canon lens is used. I've done enough experimentation that I can make a general observation that this situation between the Canon camera body and the non-Conon lens in our setup may be putting additional stress on the Canon EOS 5DS R cameras with the frequent capture and downloads of the RAW image files at the 5 second rate.... essentially it just brings out the instability. That's my observation over a very long time now with gphoto2 capture and download (using the tether method with automated external trigger until now) and the Canon 5DS R. 

We're just trying to find the most stable way to use gphoto2 and capture as fast as possible at the high end of our needs which is about one capture and download event every 2 seconds per camera. And we're also getting rid of the external relays and the external trigger cable using the N-3 connector (I could never figure out why we needed to trigger the camera like that, instead of using the USB3 connection between the Linux computer and the camera). 

We'll get things set up and work on some experiments with the `gphoto2 --wait-event-and-download` method and let you know how it goes. 

Thanks for your help and time. Appreciated.

vepicurean, 2021-03-18:
An update on the capture and download with the Canon EOS RDS R cameras. We got some good stability and performance with one camera in operation using gphoto2 via bash scripts. But now when multiple cameras operating independently for our target final application we run into some of the challenges we had before with the old code. 

We can establish separate tethers to each individual camera using `gphoto2 --capture-image-and-download --interval -1` and identifying the specific USB port for each camera. But when we try to use the kill -SIGUSR1 PID, thinking we can direct which camera to trigger from the PID, we're noticing that we having trouble identifying the PID associated with the respective camera. We don't want or need the cameras triggering at the same time, we have to be precise to a fraction of a second, so killing all gphoto2 related processes to trigger all cameras running on a tether is not an option with our application. 

Any suggestions anyone has to direct the trigger of the respective tethers to capture image and download on a per camera basis and at different times would be appreciated. Thanks in advance.

vepicurean, 2021-03-23:
Can seem to get a stable method to individually control two or more cameras of the same model via tether for capture and download. People have been talking about it in the forums for years using gphoto2, but number much detailed discussing about the code and method to achieve the end goal. 

This person here was asking about it on stack overflow years ago. We are running into the same issues with precise control with just two cameras. 

https://stackoverflow.com/questions/38530795/how-to-identify-2-of-the-same-type-camera-gphoto2

vepicurean, 2021-04-04:
We did get this figured out. Had to combine the gphoto2 functionality with some bash scripting and running multiple sessions of gphoto2 on separate threads (separate terminal sessions), but we got it working well with the two cameras.

websocketing, 2021-04-20:
> 我们确实弄清楚了这一点。必须将gphoto2功能与一些bash脚本结合起来，并在单独的线程（单独的终端会话）上运行gphoto2的多个会话，但是我们使它在两个相机上都可以很好地工作。

Hello, I'm gPhoto2 developer, too. Can you communicate with me? Do you have any contact information?

msmeissn, 2021-04-21:
@websocketing i do not know you ... do you need a gphoto2 developer or are you one? If you need a gphoto developer, either use gitzhub issues or my email address (which is in git logs)

msmeissn, 2021-04-21:
> We did get this figured out. Had to combine the gphoto2 functionality with some bash scripting and running multiple sessions of gphoto2 on separate threads (separate terminal sessions), but we got it working well with the two cameras.

good to hear you got it working well!

I was not aware of good guidance on this so far, most people have their own solutions as you noticed

closed 2021-04-21 by msmeissn