gphoto/libgphoto2 #630: SONY A7S imageQuality error

peppuccio, 2021-03-02:
**Describe the bug**
When using last gphoto driver, setting Imagequallity to RAW will download first a JPG then the RAW version in the next photo.

**To Reproduce**
Exact steps to reproduce the behavior.

**Run driver**
Take a photo assigning RAW (imagequality set value = 3)
Dowload a JPG
Took a second photo (change point of view) then you will see the first one (driver download raw version of previous photo)
**Expected behavior**
Everything seems to happen as Raw+JPG was selected but download take the first photo and second one is downloaded with the next request.
[image]

**Desktop** (please complete the following information):

OS: Rapsberian buster
last version
**Log Files**
[log_21-51-23.txt](https://github.com/gphoto/libgphoto2/files/6071766/log_21-51-23.txt)

msmeissn, 2021-03-28:
can you 
gphoto2 --get-config imagequality
and see what is set in the camera and if its reported correctly  , for RAW  and also for RAW+JPG?

Our mapping might not be correct for this camera

peppuccio, 2021-03-28:
That the result.

astroberry@astroberry:~ $ gphoto2 --get-config imagequality
Label: Image Quality                                                           
Readonly: 0
Type: RADIO
Current: RAW
Choice: 0 Standard
Choice: 1 Fine
Choice: 2 Extra Fine
Choice: 3 RAW
Choice: 4 RAW+JPEG (Fine)
END

peppuccio, 2021-03-28:
I don't know if this detail can help but when connected via gphoto2
My camera shows this 
[image]

And when I connect thru indi driver I get this
[image]

peppuccio, 2021-03-28:
Have a look to my description over indi criver
https://github.com/indilib/indi-3rdparty/issues/339

msmeissn, 2021-04-25:
In this mode it really should generate just 1 picture... weird :( The cameas also says it should just be RAW.

We are not doing anythiong special that could cause a 2nd image to be downloaded.

peppuccio, 2021-04-25:
I'm quite sure that the problem comes from Kstars/Indi driver... 
Thanks for your attention.