gphoto/libgphoto2 #633: error: implicitly declaring library function 'snprintf'

ryandesign, 2021-03-09:
**Describe the bug**

libgphoto2 2.5.27 doesn't build on macOS when implicit function declarations aren't allowed (i.e. with Xcode 12 or later or by requesting this behavior in CFLAGS):

```
libusb1/libusb1.c:319:3: error: implicitly declaring library function 'snprintf' with type 'int (char *, unsigned long, const char *, ...)' [-Werror,-Wimplicit-function-declaration]
                snprintf (path,sizeof(path), "usb:%03d,%03d",
                ^
libusb1/libusb1.c:319:3: note: include the header <stdio.h> or explicitly provide a declaration for 'snprintf'
1 error generated.
```

See previous similar bug reports #552 and #553.

libusb1.c already includes stdio.h but its declaration of snprintf is guarded like this:

```
#if __DARWIN_C_LEVEL >= 200112L || defined(_C99_SOURCE) || defined(__cplusplus)
__BEGIN_DECLS
int	 snprintf(char * __restrict __str, size_t __size, const char * __restrict __format, ...) __printflike(3, 4);
int	 vfscanf(FILE * __restrict __stream, const char * __restrict __format, va_list) __scanflike(2, 0);
int	 vscanf(const char * __restrict __format, va_list) __scanflike(1, 0);
int	 vsnprintf(char * __restrict __str, size_t __size, const char * __restrict __format, va_list) __printflike(3, 0);
int	 vsscanf(const char * __restrict __str, const char * __restrict __format, va_list) __scanflike(2, 0);
__END_DECLS
#endif /* __DARWIN_C_LEVEL >= 200112L || defined(_C99_SOURCE) || defined(__cplusplus) */
```

Adding `-D_DARWIN_C_SOURCE` to CFLAGS does allow compilation to succeed but I don't know if that's the right solution.

**Name the camera**
n/a

**libgphoto2 and gphoto2 version**
1.5.27

**To Reproduce**
Steps to reproduce the behavior:

```
./configure
make
```

ndim, 2021-03-09:
That sounds like adding `--std=c99` or `--std=c11` to `CFLAGS` instead might be worth a try.

Writes someone whose closest contact with MacOS development has been committing Travis CI scripts and waiting for Travis CI OSX builders to finish.

msmeissn, 2021-03-09:
i added a define _ISO_C99_SOURCE 

but erhaps we can try this global option instead of adding random #define _xxx_source

ndim, 2021-03-09:
Maybe `--std=gnu99` or `--std=gnu11` are better suited for compiling libgphoto2 than `--std=c99` or `--std=c11`.

ryandesign, 2021-03-09:
Adding `#define _DARWIN_C_SOURCE` to libusb1.c as you did in 9f76a476874465cdf861ab68ccf542c2fc8d4900 allows the build to succeed.

You then changed it to `#define _ISOC99_SOURCE` in 9f76a476874465cdf861ab68ccf542c2fc8d4900. This fails the same way as before. There are no occurrences of `_ISOC99_SOURCE` within the system headers (at least not on my macOS 10.13.6 system) so that does not appear to be a define that is of any significance to the toolchain on macOS.

msmeissn, 2021-03-10:
i switched it back for now.  any more places?

ryandesign, 2021-03-13:
That one change was all I needed to build libgphoto2 2.5.27.

closed 2021-03-14 by msmeissn