gphoto/libgphoto2 #634: Sony RX100VII - Random PTP Access Denied (0x200f)

ricardodeazambuja, 2021-03-09:
Using the ```gphoto2 --shell```, I'm able to use capture-preview, but directly using the command line ```gphoto2 --stdout --capture-preview``` I'm not.

**Name the camera**
Sony RX100VII, shows as USB PTP Class Camera.

**libgphoto2 and gphoto2 version**
```
gphoto2         2.5.23         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.25         standard camlibs (SKIPPING lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

**To Reproduce**
```$ gphoto2 --stdout --capture-preview```
Returns :

```
*** Error ***              
Sorry, your Sony camera does not seem to return a JPEG image in LiveView mode
*** Error (-1: 'Unspecified error') ***

```

```
$ gphoto2 --shell
gphoto2> capture-preview
```
The first or second time it returns the same error above, but after that it saves the preview to ```capture_preview.jpg```.

The debug output for ```gphoto2 --stdout --capture-preview``` shows a lot of  ```ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x1009 receiving resp failed: PTP Access Denied (0x200f)```, but the other parts of the code seem robust to this randomly occurring while ```gp_camera_capture_preview``` crashes the first time it receives the error:

```
0.536019 ptp_usb_getresp [usb.c:514] (0): PTP_OC 0x1009 receiving resp failed: PTP Access Denied (0x200f)
0.546146 gp_context_error            (0): Sorry, your Sony camera does not seem to return a JPEG image in LiveView mode
0.546203 gp_camera_capture_preview [gphoto2-camera.c:1403](0): 'camera->functions->capture_preview ( camera, file, context)' failed: -1
```

msmeissn, 2021-03-09:
can you try out a newer libgphoto2 version? both 2.5.27 and 2.5.26 had improvements for Sony liveview and other functionality.

ricardodeazambuja, 2021-03-09:
Thanks for your fast reply!
So, I removed gphoto2 and libgphoto2* from my system and installed from source gphoto2 and libgphoto2 using their latest releases. I had to add libgphoto2 to ldconfig:
`$ sudo vi /etc/ld.so.conf.d/libgphoto2.conf`
`/usr/local/lib`
and finally
`$ sudo ldconfig`
so gphoto2 was happy, but now my camera gets stuck in the "Connecting... USB" screen every time I connect it to my computer.

```
$ ldd /usr/local/bin/gphoto2
	linux-vdso.so.1 (0x00007ffd7cfa6000)
	libgphoto2.so.6 => /usr/local/lib/libgphoto2.so.6 (0x00007f6fbfbcd000)
	libgphoto2_port.so.12 => /usr/local/lib/libgphoto2_port.so.12 (0x00007f6fbfbc0000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6fbfb9d000)
	libpopt.so.0 => /lib/x86_64-linux-gnu/libpopt.so.0 (0x00007f6fbfb8f000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6fbf99d000)
	libltdl.so.7 => /lib/x86_64-linux-gnu/libltdl.so.7 (0x00007f6fbf992000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6fbf841000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f6fbfc96000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6fbf83b000)
```

```
$ gphoto2 --version
...
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.27         gcc, popt(m), no exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.27         standard camlibs (SKIPPING docupen lumix), gcc, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial, gcc, ltdl, no EXIF, no USB, serial without locking
```

Am I missing any step?

msmeissn, 2021-03-09:
no... what is gphoto2 --auto-detect and --summary   repiorting?

msmeissn, 2021-03-09:
you need the libusb1 development headers and libs to build. libusb1-dev or libusb1-devel or similar

ricardodeazambuja, 2021-03-09:
After installing the headers above `$ sudo apt-get install libusb-1.0-0-dev` and recompiling/installing libgphoto2... now capture-preview works! The camera starts in the "Connecting... USB" screen and after running --summary it "unlocks" and shows the image on its screen.

However, now --show-preview saves to a file instead of opening a window. I would bet this means I'm missing something in the compilation process of gphoto2. So I will investigate that later.

Thanks for your help!!!

closed 2021-03-09 by ricardodeazambuja

msmeissn, 2021-03-09:
if --show-preview showed ascii art preview, you need to install aalib-devel or aalib-dev and build gphoto2  with ./configure --with-aalib