gphoto/libgphoto2 #636: Canon 60D does not attach

wimjongman, 2021-03-14:
I try to attach Canon 60D which Is listed as supported. It has worked one time, but I can't get it to work anymore. Any hint would be appreciated.

```
gphoto2         2.5.20         gcc, popt(m), exif, cdk, aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

Looking in the log, the camera has issues connecting as USB. It has no settings to change the mode

```
usb 1-1.3: USB disconnect, device number 13
usb 1-1.3: new high-speed USB device number 14 using dwc_otg
usb 1-1.3: New USB device found, idVendor=04a9, idProduct=3215, bcdDevice= 0.02
usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
usb 1-1.3: Product: Canon Digital Camera
usb 1-1.3: Manufacturer: Canon Inc.
mtp-probe: checking bus 1, device 14: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
mtp-probe: bus: 1, device: 14 was not an MTP device
mtp-probe: checking bus 1, device 14: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3"
mtp-probe: bus: 1, device: 14 was not an MTP device
usb 1-1.3: USB disconnect, device number 14
```

`gphoto2 --auto-detect` yields nothing
`lsusb` has no entry for the device

I'm running R-Pi 
` Linux version 5.10.17-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP Mon Feb 22 11:29:51 GMT 2021`

I played around with the udev rules to exclude the device from mtp but it did not work.

msmeissn, 2021-03-15:
can you check if the camera battery is full enough?

does it disconnect immediately?

wimjongman, 2021-03-15:
Yes, the battery is now 75% full and the log above is in the same step. So it immediately disconnects.

Weirdly, it worked one time. Shut down the camera, connect the USB, Switch on the camera. At this point, I could see the device with `lsusb` and I could operate it with `gphoto2` for about one hour. Then I could never get it working again.

Maybe it is a faulty USB cable, so I will try that next. In the meantime, shoot any suggestion you might have my way and I will try it.

Sija, 2021-03-15:
Faulty cables are often to blame.

hfiguiere, 2021-03-15:
also which raspberry pi? and is it RaspberryPi OS or some other distro?

The USB stack is known to have flaws.

wimjongman, 2021-03-15:
> also which raspberry pi? and is it RaspberryPi OS or some other distro?

I installed Astroberry and updated the system with the latest patches. It is based on the official RPI release.

https://www.astroberry.io/docs/index.php?title=Astroberry_Server#Features

`Linux version 5.10.17-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP Mon Feb 22 11:29:51 GMT 2021`

> The USB stack is known to have flaws.

Do you have more information about that?

hfiguiere, 2021-03-16:
still didn't say which Raspberry Pi

wimjongman, 2021-03-16:
Sorry. RPI3

wimjongman, 2021-03-17:
Ok, I pinched down the USB connector that goes in the camera to tighten it a bit (I did it with my teeth), and that solved the problem. 

It turns out that the message 

`mtp-probe: bus: 1, device: 14 was not an MTP device`

had nothing to do with. 

`usb 1-1.3: USB disconnect, device number 14`

The former is just a diagnostic message and not an error. The latter is also just a diagnostic message, the USB really disconnected because of a faulty cable ^^.

Also when @msmeissn said: "does it disconnect immediately?", he was right that it did not disconnect immediately. There was around a second between the two last messages. That will teach me to strip timestamps from log messages.

Anyway, all is now explained and I apologize for wasting your time.

closed 2021-03-17 by wimjongman

msmeissn, 2021-04-30:
no problem, thanks for feedback! :)