gphoto/libgphoto2 #638: Sony RX100M4 Long Delay after Capture

jbienz, 2021-03-19:
**Describe the bug**

When using gphoto CLI with Sony DSC-RX100M4, photo is taken quickly but gphoto takes 3 seconds to return to command line.

**Name the camera**
Sony DSC-RX100M4

**libgphoto2 and gphoto2 version**
2.5.17

**To Reproduce**
Steps to reproduce the behavior:

gphoto2 --auto-detect --capture-image-and-download --keep

or

gphoto2 --auto-detect --capture-image-and-download

Photo is taken in less than 1 second and downloaded in about 1 second. But then whether deleting or keeping the file, it takes another 3 seconds to return to the command line. I also have an EOS Rebel XSI and it does not exhibit this behavior. It appears it may be an issue with the Sony driver?

I'm using gphoto with [Octolapse](https://github.com/FormerLurker/Octolapse/) and the longer it takes to return to the command line the longer the print is paused (which leaves artifacts in the print). Anything you can do to reduce the delay would be greatly appreciated.

P.S. I prefer to use the 100M4 over the Rebel XSI because the M4 has HDMI out and I can set my focal point much more accurately on a large display. Thank you!

msmeissn, 2021-03-19:
the 2.5.27 release will be better, but we now have a initial 2.5 second delay as otherwise other rx100m4 would not capture/autpofocus... 
do you use manual or autofocus?

jbienz, 2021-03-19:
When doing Octolapse prints I always use manual focus. Don't want the focal point changing or accidentally focusing on the print head.

Is the 2.5 second delay at the beginning or the end? I looked at the code and I see the timing in there for half-press, full-press. But the delay I'm talking about is actually after the image is captured and downloaded. Capture and download takes 2.6 seconds. After that, there's anther 3+ seconds before the process ends and returns to the command line.

msmeissn, 2021-03-28:
One thing I see in capture.log is that libusb0 is in use, this causes also unnecessary delays :( 
libgphoto2 should be compiled with libusb1 , especially for Sony cameras.

hfiguiere, 2021-03-28:
Would it make sense to remove libusb0 support now?

msmeissn, 2021-03-28:
i am thinking about bigger warnings... and off unless specific on configure line or so

jbienz, 2021-03-29:
I didn't select libusb0 myself. I installed libgphoto2 by following this official guide from the Octolapse Wiki:

https://github.com/FormerLurker/Octolapse/wiki/V0.4---GPhoto2-Installation

He recommends a copy-and-paste script rather than using app-get. Do you see any potential issues with this script or recommended approach?

jbienz, 2021-04-26:
Just wanted to follow up on this.

Two questions:

1. How can I update libgphoto2 to use libusb1 instead of libusb0? Is there a how-to somewhere I can follow?
2. Should I request the author of Octolabse to change instructions for installing gphoto2? Every millisecond is precious when a print is paused waiting for a photo, so I'm assuming these steps should be updated to somehow include libusb1.

Thank you!

msmeissn, 2021-04-26:
you basicaly install libusb 1 devlopment like libusb-1-dev or similar package before running configure of libgphoto2

I filed a github issue with octolapse to adjust the howto

jbienz, 2021-04-26:
Thank you @msmeissn. I didn't see that issue previously but I'll jump in on it over at Octolapse.