gphoto/libgphoto2 #640: does not mount camera, does not start gthumb

udovdh, 2021-03-19:
Description of problem:
On Fedora 33, a Canon S110 is no longer mounted after being connected via USB;
also gthumb is not started 

Version-Release number of selected component (if applicable):
libgphoto2-2.5.27-1.fc33.x86_64

How reproducible:
Have Canon S110. Have Fedora 33. Connect camera to Fedora using mini-USB to
USB cable. See `dmesg`. See nautilus.

Actual results:
No camera visible in nautilus. Camera is connected, as can be seen in `dmesg`.
gthumb is not started as it was on older Fedora versions.

Expected results:
Camera mounted, gthumb started when camera is connected via USB.

Additional info:
idVendor=04a9, idProduct=325b
camera is no longer in libgphoto2 udev rules.

msmeissn, 2021-03-28:
these days usually "hwdb" is in use, which should specify the correct tags.
If this worked in older fedora, but not in newer one, it is probably more of a Fedora issue, can you report it there too with this link as cross reference?

udovdh, 2021-03-28:
You mean https://bugzilla.redhat.com/show_bug.cgi?id=1940904 ?
How do I verify if Canon S110 is in hwdb?
How do I get Canon S110 in hwdb if not there?
How does hwdb control mounting and gthumb startup?
Please let me know.

msmeissn, 2021-03-28:
rpm -ql  libgphoto2-2.5.27-1.fc33.x86_64
should give a file called ...libgphoto2....hwdb 
e.g. /usr/lib/udev/hwdb.d/20-gphoto.hwdb  would it be on SUSE, fedora location might vary.

in this textfile you should find these snippets:

# Canon PowerShot S110 (2001)
usb:v04A9p3051*
 GPHOTO2_DRIVER=proprietary
 ID_GPHOTO2=1

# Canon PowerShot S110 (PTP Mode)
usb:v04A9p325B*
 GPHOTO2_DRIVER=PTP
 ID_GPHOTO2=1

The handling of gthumb is magic done by udev -> gvfs in gnome, outside of libgphoto2 scope.

udovdh, 2021-03-29:
Thanks for clarifying a bit.
Where can I influence the starting of gthumb?
gthumb should be run with --import-photos ....

ndim, 2021-03-29:
The Fedora location for the hwdb file is `/usr/lib/udev/hwdb.d/20-gphoto2.hwdb` as well.