gphoto/libgphoto2 #642: Switching Movie Mode On/Off is not working for Canon EOS m50 mark II

esarthou, 2021-03-29:
## **Description:**

We're trying to build a camera controller software for a Photo Booth type product, running on linux (raspbian).

It seems like the functionality implemented on this [commit](https://github.com/gphoto/libgphoto2/commit/5734e314d713a4a2028f4524d736e50a7cf5) is not currently working as expected:

```
static int
_put_Canon_EOS_MovieModeSw(CONFIG_PUT_ARGS) {
	int		val;
	PTPParams	*params = &(camera->pl->params);
	CR (gp_widget_get_value(widget, &val));

	if(val)
		C_PTP_MSG (ptp_generic_no_data(params, PTP_OC_CANON_EOS_MovieSelectSWOn, 0), "Failed to set MovieSetSelectSWOn");
	else
		C_PTP_MSG (ptp_generic_no_data(params, PTP_OC_CANON_EOS_MovieSelectSWOff, 0), "Failed to set MovieSetSelectSWOff");
	return GP_OK;
}
```

Of course, I can be doing something wrong in the configuration of the camera or the library and I'm also willing to get my hands dirty if given some guidance, but I've been trying to make this specific functionality work for a couple days now and had no success. 

On all my search results on Stack Overflow people seem to assume this isn't even possible, even though the camera seems to have the native support for changing the Movie switch stat remotely (EDSDK supports it).

From EDSDK's docs:
[image]

When I saw that it was in fact supported on EDSDK's docs, I decided to look intop the libgphoto's codebase and was happy to find that someone had already implemented it, but for some reason I'm still not able to make it work.

Goes without saying but any help or guidance regarding this issue will be greatly appreciated.

## **Camera:**

`gphoto2 --auto-detect` :

```
Model                          Port
----------------------------------------------------------
USB PTP Class Camera           usb:001,001
```

`gphoto2 --get-config cameramodel`:

```
Label: Camera Model
Readonly: 0
Type: TEXT
Current: Canon EOS M50m2
END
```

## **Software Version**:

`gphoto2 --version`:
```
gphoto2         2.5.27         clang, popt(m), no exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs, clang, ltdl, no EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1, clang, ltdl, no EXIF, USB, serial without locking
```

## **Steps to Reproduce**

1. Output for the following command `gphoto2 --get-config eosmovieswitch`  when camera mode dial is on M position:

Label: Movie Switch
Readonly: 0
Type: TEXT
Current: 0
END
 
2. After moving the camera mode dial to the movie position:

Label: Movie Switch
Readonly: 0
Type: TEXT
Current: 1
END

3. Then let's try for instance switching it from 0 to 1 (with the camera mode dial on the M position):
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/6224945/my-logfile.txt)

`gphoto2 --set-config eosmovieswitch=1`

**Result**:
```
*** Error ***
Parsing the value of widget 'Movie Switch' / 0xd1c2 failed with -6.

*** Error ***
Failed to set new configuration value 1 for configuration e
ntry eosmovieswitch.
*** Error (-6: 'Unsupported operation') ***
```

debug log file:
[my-logfile.txt](https://github.com/gphoto/libgphoto2/files/6224947/my-logfile.txt)

trilader, 2021-03-30:
Hi.

Maybe I should have added (better) documentation to that or at least an explanation in the commit message. `eosmovieswitch`is only for getting the state of the physical switch on the camera and `eosmoviemode` is used to change it by software. This second function (and also the `MovieSelectSwOn`/`MovieSelectSwOff` ones in the SDK) is for switching the camera to movie mode without moving any physical switch/wheel on the camera. But from my observations it only works if the camera isn't already in that state. 

Here is a (simplified) example how I use it in my application:
```cpp
const bool eos_software_video_mode = config_get_int_str_value("eosmovieswitch") == 0;
if(eos_software_video_mode) {
    config_set_int_value("eosmoviemode", GP_WIDGET_TOGGLE, 1);
}

/* Do your stuff in movie mode here */

if(eos_software_video_mode) {
    config_set_int_value("eosmoviemode", GP_WIDGET_TOGGLE, 0);
}
```

I hope this works for you.

esarthou, 2021-03-30:
It works!

you're a legend, thanks!!!!

msmeissn, 2021-04-19:
should be solved?

closed 2021-04-19 by msmeissn