gphoto/libgphoto2 #645: Canon EOS 90D: "PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)" during movie capturing

szotsaki, 2021-04-02:
**Describe the bug**

When I turn on movie recording with Canon EOS 90D, in the debug log the following errors are reported:

```
0.137700 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
0.139935 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.145555 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.156283 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.171969 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.192559 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.218271 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.248847 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.285126 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.521874 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.568445 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.619315 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.675675 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.807153 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.873210 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
0.948034 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.049355 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.080528 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.107635 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.135239 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.155060 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.172862 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.191981 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.216009 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.232261 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.248199 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.268057 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.288837 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.305393 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.319941 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.338772 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.352406 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.368015 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.386041 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.386908 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.410063 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.425234 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.426037 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.448986 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.464995 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.465847 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.488270 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.506709 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.507636 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.530100 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.548570 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.569888 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.586612 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.587461 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.608867 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.625699 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.626605 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.647711 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.659763 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.660474 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.666262 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)
1.699190 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
1.712333 ptp_usb_getresp [usb.c:515] (0): PTP_OC 0x9110 receiving resp failed: PTP Device Busy (0x2019)
[...]
```

For some reason there is a big delay in showing the output from the device, that's what I try to debug in the end. I tried to patch the library with `PTP_DONT_CLOSE_SESSION` similar to 80D (see #60) but this didn't help.

**Name the camera**

```
Modell                         Port                                            
----------------------------------------------------------
Canon EOS 90D                  usb:005,027 
```

**libgphoto2 and gphoto2 version**

```
gphoto2         2.5.27.1       gcc, popt(m), exif, cdk, no aa, jpeg, readline
libgphoto2      2.5.27.1       standard camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial lockdev locking
```

**To Reproduce**

```
LC_ALL=C ./gphoto2 --stdout --capture-movie --debug --debug-loglevel=error > /dev/null
```

[eos-90d-debug.log](https://github.com/gphoto/libgphoto2/files/6250713/eos-90d-debug.log)

msmeissn, 2021-04-19:
how many frames/time do you get with e.g.
gphoto2 --capture-movie=10s 

if the camera is busy it will report that, but in your log there is a steady stream of liveview coming

szotsaki, 2021-04-19:
I turned off autofocus and image stabiliser and I got the following results from multiple runs:

```
$> LC_ALL=C  gphoto2 --capture-movie=10s

Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.              
Movie capture finished (461 frames)

Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.              
Movie capture finished (451 frames)

Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.              
Movie capture finished (450 frames)

Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.              
Movie capture finished (450 frames)

Capturing preview frames as movie to 'movie.mjpg' for 10 seconds.              
Movie capture finished (457 frames)
```

It seems there are no error messages but the number of captured frames varied. I don't know if it's because of the camera internals or the time measurement inaccuracies.

msmeissn, 2021-04-21:
that would mean you get around 45fps, which is very very good (usually I saw 25fps or so).

does the fps rate drop significantly when you reenable the autofocus and stabilizer?

szotsaki, 2021-04-21:
Not really as I saw. With turned on IS and AF from four tests I got 450, 449, 450, and 449 frames again. With a different lens I got 455, 457, 454, and 458 frames.

For scientific purposes :) I forced the latter lens to continuously refocus for 10 seconds and got 446, 455, 457, and 454 frames.

msmeissn, 2021-04-22:
so i think its working nicely. it will occasionaly report the above errors if we are too fast, but these are ignored by the internal code already, and just posted to the debuglog.

szotsaki, 2021-04-30:
Without actually knowing the code my internal pedantic me says that it'd be better to fix those warnings (since something is off when we get error messages like those) but I can imagine that the PTP protocol is not that well defined to have a simple mechanism to protect against these errors. Since you said that they're handled at a later stage I believe those are not serious errors and therefore this ticket is not urgent to fix.

Therefore, feel free to choose between closing this report or maybe fixing it at a later time.