gphoto/libgphoto2 #646: New application

akashmon598, 2021-04-04:
Dear Sir,
Firstly, Thank you so much for creating such an incredible git hub repo. I'm sure a lot of people have benefited from your excellent work. Sir, the reason behind this mail is that I am using a Sony hx400v camera. This camera does not support your repo. Due to the same I had used another repo which I am attaching the link below. This repo allows me to control the camera. But the problem here is that I cannot send the photos from my camera to my Rasberrie pie wirelessly. As to the same I had also found another repo which allows me to send photos, link attached below. But the problem here is that the method used is Sony play memories which as you know is only supported in windows and mac OS. When going through the method I understood that it is done to establish link between camera and Rasbespierre, which is to be done only once. Sir, could you please guide me through the same and help me to establish a connection without Sony play memories. Your consideration and support on the same will be highly appreciated. Awaiting your positive reply.
link for sony camera api: https://github.com/Bloodevil/sony_camera_api
link for sending photos from camera to pc: https://github.com/ma1co/Sony-PMCA-RE 
Yours sincerely
Akash Sureshbabu

msmeissn, 2021-04-05:
I will take a look.
the first URL is for Wireless transfer only, not USB. (Something we can implement, but if there is a standalone app, it can be used too)

the second one seems to do USB wioth some knowledge I do not yet have, i will take a look there.

akashmon598, 2021-04-05:
Sir, I have created an application that connects the camera via wifi and
captures image through the sony remote option. I could send that to you  if
you want it for reference.