gphoto/libgphoto2 #647: Canon EOS USB disconnect and reconnect

koertvdp, 2021-04-06:
**Describe the bug**
Running 3 raspberry pi 4 systems with 2x Canon EOS 2000D and 1x 750D attached. Working as a timelapse system, there is also a SandDisk USB Drive attached on USB that keeps working properly. 

But after a week or so the camera loses USB connection and gphoto2 --auto-detect only reports PTP instead of camera model. After reboot of the Pi there is no mention of the USB device in the system at all (lsusb). After turning the camera on-off all seem to work for a short amount of time. 

Camera does have a separate power supplie, it only uses usb from the pi for data connection. 

When the system is running fine it seems to connect and disconnect all the time. 
When using journalctl -k or dmesg I see that for every minute it creates this: 
kernel: usb 1-1.4: USB disconnect, device number 42
kernel: usb 1-1.4: new high-speed USB device number 43 using xhci_hcd
kernel: usb 1-1.4: New USB device found, idVendor=04a9, idProduct=32a1, 
kernel: usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumbe
kernel: usb 1-1.4: Product: Canon Digital Camera
kernel: usb 1-1.4: Manufacturer: Canon Inc.

The Pi that allready lost the connection to the camera I see these errors: 
usb 1-1.3: device descriptor read/64, error -110
usb 1-1-port3: unable to enumerate USB device
v3d fec00000.v3d: MMU error from client L2T (0) at 0x801000, pte invalid

The script I run with a cron. The cron starts every day in the morning, and after reboot.
#!/bin/bash
sleep 15;
pkill -f gphoto2 
cd /mnt/SSSD1TB-3
mkdir -p $(date +%Y-%m-%d) && cd $(date +%Y-%m-%d)
while [ $(date "+%H") -lt 23 ]; do
	gphoto2 --set-config eosremoterelease="Immediate" --set-config eosremoterelease="Release Full" --wait-event-and-download=ObjectRemoved --filename SSS3_TIME_%Y-%m-%d_%H-%M-%S.%C 
	sleep 60 ;
done
cd ..

**hypothesis** 
My guess is that the camera loses connection because of the continuing connecting and disconnecting of the camera trough gphoto2. 

**Name the camera**
Camera name as shown by gphoto2 --auto-detect or USB IDs
Bus 001 Device 076: ID 04a9:32a1 Canon, Inc. 
Canon EOS 750D & Canon EOS 2000D (sees it as1500D) 

**libgphoto2 and gphoto2 version**
output of: gphoto2 --version
gphoto2 2.5.23.1
This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.23.1       gcc, popt(m), exif, no cdk, no aa, no jpeg, no readline
libgphoto2      2.5.22         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, USB, serial without locking

Also using gphoto server for remote control, focus and live view. But this does not seem to intervere as the error occurs without accessing the service.

**To Reproduce**
Trying to replicate the error on one system without succes so far, as this seems like a random error.

I can reproduce a part of the issue when switching the power to the camera with a relay. It does not turn off nor on correctly and after that i'm not able to take picture with the camera trough gphoto2. Bellow a small part of the log file that occurs from that point. 
0.797526 gp_camera_free              (2): Freeing camera...
0.797553 gp_camera_exit              (2): Exiting camera ('Canon EOS 750D')...
0.797576 ptp_usb_sendreq             (2): Sending PTP_OC 0x9116 (PTP_OC_CANON_EOS_GetEvent) request...
0.797592 gp_port_write               (3): Writing 12 = 0xc bytes to port...
0.797787 gp_libusb1_write [libusb1.c:596](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Input/Output error (-1)
0.797810 gp_port_write [gphoto2-port.c:406](0): Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
0.797829 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x9116 sending req failed: Error writing to the port (-35)
0.797842 ptp_usb_sendreq             (2): Clearing halt on OUT EP and retrying once.
0.797856 gp_port_usb_clear_halt      (2): Clear USB halt...
0.798004 gp_libusb1_clear_halt_lib [libusb1.c:582](0): 'libusb_clear_halt(port->pl->dh, internal_ep)' failed: Other error (-99)
0.798020 gp_port_write               (3): Writing 12 = 0xc bytes to port...
0.798260 gp_libusb1_write [libusb1.c:596](0): 'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Input/Output error (-1)
0.798281 gp_port_write [gphoto2-port.c:406](0): Writing 12 = 0xc bytes to port failed: Error writing to the port (-35)
0.798299 ptp_usb_sendreq [usb.c:109] (0): PTP_OC 0x9116 sending req failed: Error writing to the port (-35)

msmeissn, 2021-04-16:
Does this happen for all the cameras and all the pis?

that it runs for a week is already very good.

" v3d fec00000.v3d: MMU error from client L2T (0) at 0x801000, pte invalid" looks like a lowlevel rasperry pi problem.

also that the device does not enumerate is more at kernel / pi / hardware level :( 
Not much gphoto2 can do.

websocketing, 2021-04-16:
> 所有相机和所有pi都会发生这种情况吗？
> 
> 它运行了一周已经非常好了。
> 
> “ v3d fec00000.v3d：来自客户端L2T（0）的MMU错误位于0x801000，pte无效”看起来像是一个低级树莓派问题。
> 
> 此外，设备不枚举更多的是在内核/ pi /硬件级别:(
> gphoto2不能做很多。

First of all, I feel your selfless dedication. Can gPhoto2 run on Android system? I tried to compile, but unfortunately failed. Please let me know if you have any information.

msmeissn, 2022-07-23:
this sounds a bit more like a hardware issue, sorry

STR4NG3RdotSH, 2024-11-30:
Same issue with Pi4 & Canon EOS 20D
Camera's power save setting is disabled (Never turn self off).
Its gone for a few days without issue before, but normally it seems to kill the camera a few times a day. The only way to fix it is to do one of the following:
 - Re-insert power for camera
 - Turn camera to OFF then back to ON via power dial
 
Rebooting the Pi seems like a pointless step since rebooting the camera fixes it. Should also note, re-inserting USB/Data line made no difference. Its like the camera is receiving some kind of power-off command and yes I know that doesn't exist in gphoto2 (Or powering off because it's erroring out from something?). 
I've also tried different power sources.