gphoto/libgphoto2 #648: nikon z5&&z6&&z50 record  error  0x200f

stronges, 2021-04-16:
i need nikon record operate , but have error_code 0x200f ,
this is detail 
[image]

msmeissn, 2021-04-16:
can you unfold the last OUT transaction, so we can see what property it tries to set there?
also unfold unknown command (last line)

stronges, 2021-04-17:
[image]
this  compelete picutre

stronges, 2021-04-17:
here is  another screenshot of  a successful recording of the product
[image]

msmeissn, 2021-04-18:
what exact gphoto2 commands do you send?
what version is in use?

output of:
gphoto2 --version

if libgphoto2 is not 2.5.27 , please update libgphoto2 to 2.5.27 as I fixed this movie capture stuff.

stronges, 2021-04-19:
i want  record video, and want use command  PTP_OC_NIKON_StartMovieRecInCard,  but no implementation ,
I found that I could not enter the application mode during initialization, so not implementation  recording

stronges, 2021-04-19:
I developed on other platforms without using libphoto library. I just refer to libphoto. May I ask what you have fixed in v2.5.27 and what I need to do on my platform

msmeissn, 2021-04-19:
The Z series has a new method to switch to application, the PTP_OC_NIKON_ChangeApplicationMode opcode 0x9435 , 1 argument (on = 1, off = 0), no data,   that needs to be used . See the SDK.

closed 2021-04-19 by msmeissn

stronges, 2021-04-20:
thanks, i find one of my interrupt  endpoints  is not in runing due to driver reason,  after the repair, recording can proceed normally