gphoto/libgphoto2 #650: Canon EOS: whitebalanceadjusta and whitebalanceadjustb always report 0

jktjkt, 2021-04-19:
I'm always getting `0` for both `whitebalanceadjusta` and `whitebalanceadjustb` on all of my Canon EOS cameras (EOS 5D IV, EOS RP mirrorless, 6D with Magic Lantern, and 5D II with Magic Lantern). I'm using this Python script, and I'm hoping that it reports back the WB shift I select in my camera. However, I always get zeros. The camera are set to either the movie mode (RP and 5Div), or live view (6D, 5Dii).

```python
import gphoto2 as gp
cam = gp.Camera()
cam.init()
cfg = cam.get_config()
viewfinder = cfg.get_child_by_name('viewfinder')
viewfinder.set_value(1)
cam.set_single_config('viewfinder', viewfinder)
while True:
    event_type, event_data = cam.wait_for_event(5000)
    if event_type == gp.GP_EVENT_UNKNOWN:
        cfg = cam.get_config()
        wb_shift_a = cfg.get_child_by_name('whitebalanceadjusta')
        wb_shift_b = cfg.get_child_by_name('whitebalanceadjustb')
        print(wb_shift_a.get_value(), wb_shift_b.get_value())
```

### Camera name as shown by gphoto2 --auto-detect or USB IDs

- `04a9:3281`
- `04a9:32e2`
- `04a9:3250`
- `04a9:3199`

### libgphoto2 and gphoto2 version
```
gphoto2         2.5.27         gcc, popt(m), exif, no cdk, no aa, jpeg, readline
libgphoto2      2.5.27         standard camlibs (SKIPPING docupen lumix), gcc, ltdl, EXIF
libgphoto2_port 0.12.0         iolibs: disk ptpip serial usb1 usbdiskdirect usbscsi, gcc, ltdl, EXIF, USB, serial without locking
```

When I print out the `event_data`, I'm getting a `PTP Property d10b changed` each time I change the WB shift in-camera (that's `PTP_DPC_CANON_EOS_WhiteBalanceAdjustA`), but the value printed is always `0` for both axes.

To rule out a possible error in the Python bindings, I've also tried running `gphoto2 --list-all-config` when connected to one of these cameras (EOS RP) in both the movie mode and the manual photo shooting modes, with the AWB plus a manual WB shift already preconfigured. These report `Current: 0` for both `/main/imgsettings/whitebalanceadjusta` and `/main/imgsettings/whitebalanceadjustb`.

trilader, 2021-04-20:
Hi. I noticed that yesterday myself as well and submitted a fix in ff6013b. You either need to wait for the next release or build a version with that commit included yourself.

jktjkt, 2021-04-20:
Works like a charm, @trilader, thanks.

closed 2021-04-20 by jktjkt