gphoto/libgphoto2 #651: Canon EOS | Setting the Movie Recording Image Quality

esarthou, 2021-04-20:
**Is your feature request related to a problem? Please describe.**
We would like to have the ability of controlling the video recording quality remotely from our application using libgphoto2. Out camera is a Canon EOS M50 mark II.

I've noticed reading Canon's EDSDK docs that this config should be available remotely and I've also noticed a mention to the prop name on one of @msmeissn 's latest commits https://github.com/gphoto/libgphoto2/commit/dd51cc9d5b149053357c2ea20ca201e45a726501

So I guess this could be a question, is it possible to access this prop already using the latest version of the master branch? 
I've downloaded it and installed it from the repo on my rPi 4 and had no success. it doesn't seem to be available doing

`gphoto2 --list-config`

**Describe the solution you'd like**
If in fact it is not yet possible to access this property Ihave 2 questions.

- Is it currently in progress?

- If not, how can I help?

**Additional context**

EDSDK.pdf pag 133
[image]

Example mention on the above linked commit:
[image]

msmeissn, 2021-04-21:
hmm,. d1cd movieparam4 might be it, but it seems its a compound blob, not a 32bit/16bit or 8bit value.

its not hooked up yet, so not reachable.

I will see what I can do here

esarthou, 2021-06-28:
Hi team,

Is there any approximate timeline in which this parameter could be hooked up?

Thanks!!